ghdl/ghdl #460: Should GHDL ship OSVVM as a pre-compiled library?

Paebbels, 2017-11-15:
I was updating my commercial VHDL simulator installations, as I noticed that OSVVM is shipped in source and pre-compiled in Riviera-PRO, Active-HDL, ModelSim and QuestaSim.

Should GHDL do the same?

OSVVM is using an accepted free license (Perl Artistic License) and should be no problem for Debian packaging.

---------------
Notification list: @JimLewis

tgingold, 2017-11-17:
I would say no.  I prefer to have a different package for OSVVM; it simplifies the bug report process.

Paebbels, 2017-11-20:
You mean a separate e.g. Debian package?

My ideas was to integrate OSVVM as a Git submodule into `libraries` to it can be pre-compiled and installed with `make`. If a package maintainer creates a GHDL package, a separate OSVVM package could be created.

I see a usage advantage: Users could refer to OSVVM library without needing the `-P` parameter to use it. More over it gets installed into a global directory and is usable for all users on a system.

JimLewis, 2017-11-20:
It would be my preference that the latest version is available in the simulator download.   As Patrick pointed out, this would be consistent with what some of the simulator vendors are doing.

Brian-Drummond, 2017-11-27:
There's a danger that a pre-installed OSVVM adds possible confusion if you're also installing VUnit - do you need to install both, or does VUnit install OSVVM for you (yes it does), then if you install both, which one are you using?
If we do, however, VUnit does it via github as per @Paebbels suggestion, we can presumably use the same approach.

olofk, 2017-12-05:
If you're using FuseSoC, you can also just add a dependency on osvvm

Paebbels, 2017-12-05:
What if we use a global (and/or per user) configuration file of library mappings for GHDL.

```
/etc/ghdl.config
osvvm    /opt/ghdl/0.35/libraries/osvvm/
```

This allows us to:
- ship OSVVM with the latest version without disturbing user that need older versions (they can remove that line)
- other libraries can also be pre-compiled and installed system wide
- if also /home/<user>/ghdl.config would be read, we could merge such settings
- user can overwrite libraries by their own modified versions by changing the libraries path

tgingold, 2017-12-10:
If you install OSVVM in the ghdl install dir (like the ieee packages or vital packages), it will be available to all users.  So I would say no need for a global configuration file.
Per user can make sense.

Paebbels, 2017-12-10:
Some users have requirements to use:
- a newer version than installed with GHDL
- a patched version due to a bug in a library
- an older version to maintain compatibility

That's why I think it should be configurable where GHDL loads a pre-compiled library from.
If no entry exists or a default entry exists, it should point to the pre-compiled library in the GHDL installation directory. But it should also be possible to redirect that search paths.

ModelSim has a simple syntax based on INI files. It has this search oder for INI files:

0. check if option `-modelsim_ini` was supplied to a program
1. installation directory
2. current working directory

A special entry allows to include other INI files.

The Python infrastructure of PoC makes heavy use of these INI files to create a very flexible ModelSim configuration setup. E.g. it's very easy to switch between Xilinx ISE and Xilinx Vivado libraries.

I propose:
1. A system wide INI file to define the mappings of library logical names and library locations.
2. Check if a mapping file exists in the current working directory.
3. A command line option to read a different mapping file.

File Format:
```ini
[VHDL:1993]
std = /opt/ghdl/0.35-dev/lib/std
ieee = /opt/ghdl/0.35-dev/lib/ieee

[VHDL:2008]
std = /opt/ghdl/0.35-dev/lib/std
ieee = /opt/ghdl/0.35-dev/lib/ieee2008
osvvm = /opt/ghdl/0.35-dev/lib/osvvm
```

We could also think about integrating the file for long command line options into a settings section of such a file.

tgingold, 2017-12-12:
So this is the same idea as #55.

closed 2017-12-12 by tgingold