ghdl/ghdl #461: 'with select' with undefined selector gives return value 255 on execution with '--std=08'

poinu, 2017-11-16:
Hi!

The following fails to execute with a return code '255' when compiled with `ghdl -m --std=08 test`:

```
library ieee;                                                                   
use ieee.std_logic_1164.all;                                                    
use ieee.numeric_std.all;                                                       
                                                                                
entity test is                                                                  
end test;                                                                       
                                                                                
architecture behaviour of test is                                               
    signal selector : std_logic_vector(1 downto 0) := "UU";                     
    signal result   : std_logic_vector(7 downto 0);                             
                                                                                
    signal op_1     : std_logic_vector(7 downto 0);                             
    signal op_2     : std_logic_vector(7 downto 0);                             
begin                                                                           
    with selector select                                                        
        result <= (std_logic_vector(signed(op_1) + signed(op_2))) when "00",    
                  (others => '-') when others;                                  
end behaviour;                                                                  
```
The code, however, runs fine if no standard is specified.

Is this code real? A similar code using a process works fine when using 08 standard.

Thanks!

Paebbels, 2017-11-16:
What GHDL version did you use? Please add the printout of `ghdl.exe -v` to your question.

tgingold, 2017-11-17:
I can reproduce the issue and I can confirm it works correctly for when --std is not set to 08.

poinu, 2017-11-17:
Just for completeness:

version: `GHDL 0.34 (2017-03-01-203-g06a78d22) [Dunoon edition]`
command: `ghdl -r test --wave=wave.ghw --stop-time=1us`

commit 031e1aa1 (2017-11-19): Fix missing open_temp/close_temp for selected sequential signal assignment.

Fix #461
closed 2017-11-19 by tgingold