ghdl/ghdl #462: Is 'sra' operator supported with standards previous than VHDL-08?

poinu, 2017-11-16:
Hi!

In the GHDL [documentation](http://ghdl.readthedocs.io/en/latest/GHDL_implementation_of_VHDL.html?highlight=sra) it is stated that the operator `sra` was introduced in VHDL-93. However, I fail to get it working in GHDL with any standard previous to VHDL-08.

Consider the following snippet:

```
library ieee;                                                                   
use ieee.std_logic_1164.all;                                                    
use ieee.numeric_std.all;                                                       
                                                                                
entity test is                                                                  
end test;                                                                       
                                                                                
architecture behaviour of test is                                               
    signal result   : std_logic_vector(7 downto 0);                             
    signal tmp      : std_logic_vector(7 downto 0);                             
begin                                                                           
    result <= (std_logic_vector(signed(tmp) sra 2));                            
end behaviour;                                                                  
```
When analysing the code with `ghdl -a --std=93 test.vhd` the analysis fails with this error message: 

```
test.vhd:12:45:error: no function declarations for operator "sra"
ghdl:error: compilation error
```
The code also fails to compile with `--std=00` and `--std=02`, but works fine with `--std=08`.

Am I missing something?
Thank you!

Paebbels, 2017-11-17:
I can find `sra` only for `bit_vector`, but not for `std_logic_vector`.

tgingold, 2017-11-17:
You should indeed use srl instead for signed or unsigned types.
SRA is not defined in numeric_std.

closed 2017-11-17 by tgingold