ghdl/ghdl #463: vendors compilation script fails on OSX

bergnoli, 2017-11-17:
in the shared.sh file there is no OSX workaround for using "greadlink" instead "readlink"

tgingold, 2017-11-25:
If you can test a patch, I can merge it.

bergnoli, 2017-11-27:
`diff --git a/libraries/vendors/shared.sh b/libraries/vendors/shared.sh
index 6b77b2f5..1c672fb1 100644
--- a/libraries/vendors/shared.sh
+++ b/libraries/vendors/shared.sh
@@ -76,7 +76,8 @@ SetupDirectories() {
        fi

        # Resolve paths to an absolute paths
-       SourceDirectory=$(readlink -f $SourceDirectory)
+       READLINK=readlink; if [[ $(uname) == "Darwin" ]]; then READLINK=greadlink; fi
+       SourceDirectory=$($READLINK -f $SourceDirectory)
        if [[ ! "$DestinationDirectory" = /* ]]; then
                DestinationDirectory=$WorkingDir/$DestinationDirectory
        fi`

commit 7d95eb2a (2017-11-28): Use greadlink on Darwin.

Fix #463
closed 2017-11-28 by tgingold