ghdl/ghdl #466: Using a parametrized-length port as selector: object subtype is not locally static

poinu, 2017-11-24:
Hi,

I have a code with a generic entity where a port whose length is parametrized is used to select a given value for another port. In particular, I've tested four different versions to achieve a similar behaviour.

```
library ieee;                                                                   
use ieee.std_logic_1164.all;                                                    
                                                                                
entity test is                                                                  
    generic (                                                                   
        bits : integer := 2                                                     
    );                                                                          
    port (                                                                      
        selector : in  std_logic_vector(bits - 1 downto 0);                     
        input    : in  std_logic;                                               
        output   : out std_logic                                                
    );                                                                          
end test;                                                                       
                                                                                
architecture behaviour of test is                                               
    constant clr : std_logic_vector(bits - 1 downto 0) := (others => '0');      
begin                                                                           
    -- A) This works --                                                         
    output <= '0' when selector = clr else input;                               
                                                                                
    -- B) This fails: object subtype is not locally static --                   
    with selector select                                                        
        output <= '0' when clr,                                                 
                  input when others;                                            
                                                                                
    process (selector)                                                          
    begin                                                                       
        -- C) This fails: object subtype is not locally static --               
        case selector is                                                        
            when clr => output <= '0';                                          
            when others => output <= input;                                     
        end case;                                                               
                                                                                
        -- D) This works --                                                     
        if selector = clr then                                                  
            output <= '0';                                                      
        else                                                                    
            output <= input;                                                    
        end if;                                                                 
    end process;                                                                
end behaviour;                                                                  
```
I have experienced that versions A and D are analized without any problem in GHDL while versions B and C fail with this compilation error: `error: object subtype is not locally static`.

I've also tried using the VHDL-08 standard but the result is the same.

I was wondering whether there is a problem on GHDL with versions B and D or if, in the other hand, they are just illegal in VHDL-08 (and 93c).

I'm currently on GHDL version:
```
GHDL 0.34 (2017-03-01-203-g06a78d22) [Dunoon edition]code is legal 
 Compiled with GNAT Version: 7.2.0
 GCC back-end code generator
```

Thanks

andreygursky, 2017-11-24:
This reminds me [#40 relax "choice must be locally static expression"](https://sourceforge.net/p/ghdl-updates/tickets/40/).

poinu, 2017-11-24:
I've now tried with a more recent version of GHDL:

```
GHDL 0.35-dev (2017-03-01-427-g0872dc63) [Dunoon edition]
 Compiled with GNAT Version: 7.2.0
 GCC back-end code generator
```

And those codes still fail but the error message has changed: `error: choice must be locally static expression`.

I see this is the same error message as the one given in the ticket @andreygursky referred (which is, in fact, very similar), so I guess B and C versions are just non-standard.

andreygursky, 2017-11-24:
I still believe, things that can be resolved at compile/elaboration stage instead of runtime can be treated as locally static (at least since VHDL 2008).

tgingold, 2017-11-25:
B and C are incorrect even with vhdl-08.

closed 2017-11-25 by tgingold

poinu, 2017-11-25:
Thanks for the answer @tgingold. However, if possible, I'd like to know why those versions are ambiguous (and incorrect).

For the ticket referred by @andreygursky I understand that, at analysis time, the generic parameter's value is unknown and this makes the comparison of such generic parameter with another signal unevaluable in an unpriorized conditional statement (case or select), as multiple 'options' could 'match' at the same time.

However, in my code I fail to see this ambiguity, as the generic parameter is only used for the length of those signals, but not the value. That is, the constant `clr` is always intitialized to zero no matter if its 3 bits long or 32. In the other hand, the `selector` is to be compared some constants of its same length, but that doesn't allow that more than one 'options' can 'match' at the same time.

What is the 'ambiguity' that makes those statements incorrect?

andreygursky, 2017-11-26:
@poinu, back to the time of my issue with "locally static", I've also reported this for an alternative VHDL simulator [nvc](https://github.com/nickg/nvc) and thankfully this was implemented there. Since then you can use either --relax=locally-static or --std=2008. I'm not sure about correctness of your example but I've quickly tested it, and it has been at least successfully analyzed.

tgingold, 2017-11-26:
> On 25 Nov 2017, at 12:23, Ferran Pallarès <notifications@github.com> wrote:
> 
> Thanks for the answer @tgingold. However, if possible, I'd like to know why those versions are ambiguous (and incorrect).
> 
> For the ticket referred by @andreygursky I understand that, at analysis time, the generic parameter's value is unknown and this makes the comparison of such generic parameter with another signal unevaluable in an unpriorized conditional statement (case or select), as multiple 'options' could 'match' at the same time.
> 
> However, in my code I fail to see this ambiguity, as the generic parameter is only used for the length of those signals, but not the value. That is, the constant clr is always intitialized to zero no matter if its 3 bits long or 32. In the other hand, the selector is to be compared some constants of its same length, but that doesn't allow that more than one 'options' can 'match' at the same time.
> 
> What is the 'ambiguity' that makes those statements incorrect?

The VHDL standard requires that all choices are 'locally' static (which means known during analysis).
In particular, that implies that the length of the choices are also locally static.

andreygursky, 2017-11-26:
@poinu, would such change still fit your needs?
```diff
entity test is
    generic (
-         bits : integer := 2
+         constant bits : integer := 2
    );
```
Generic constant is considered locally static in VHDL 2008 (9.4.2, c), right Tristan?

Paebbels, 2017-11-26:
@andreygursky 
There is no difference in your declaration of `bits`. The object class `constant` is the default if no class is given for an item in the generic class. `bits` is still not locally static, because any generic map can overwrite it after analysis.

andreygursky, 2017-11-26:
Patrick, thanks for your clarification.

> ```bits``` is still not locally static, because any generic map can overwrite it after analysis.

I believe, whether it is locally static or not, is not defined in terms of analysis stage, since some processing may be deferred until elaboration.

poinu, 2017-11-26:
Thank you all for the clarifications and hints. It's a pity such things are not considered in the standards. In my case, I'm not trying to run an existing code but instead I'm developing a project from scratch. For this reason I still prefer to write standard code, so that it is portable to other platforms :)

andreygursky, 2017-11-26:
@poinu, for now GHDL cannot be used for every legal VHDL-2008 code:
> GHDL fully supports IEEE 1076-1987, IEEE 1076-1993, IEEE 1076-2002 **and partially the 1076-2008** version of VHDL.

Regarding "not-so-standard". If proprietary tools work with your code perhaps with warnings and those warnings go away once you enable VHDL-2008 switch, then it is very likely the code is 2008 compliant. So here we can only kindly ask Tristan to reconsider his decision and leave such bugs open marking them with something like "VHDL-2008 hard-to-implement" (due to how analyze/elaborate stages are separated now in GHDL).

IMHO "not-so-standard" code is such code, warnings for which don't vanish even after you've enabled the support for the latest available VHDL Standard (or draft for a coming one?), though proprietary tools still continue to work. An issue with this and really non-standard but working code is when you write VHDL that will be synthesized. Actually your primary goal in this case is to get the best for the hardware, while simulation plays the same important but supporting role. So if non-standard code gives you better synthesis results, why shouldn't you use it? And simulators have to play with, which is true for proprietary ones. but not so easy for free ones.

Brian-Drummond, 2017-11-27:
@andreygursky You can certainly find counterexamples, where proprietary tools "work" in terms of compiling and appearing to run clearly incorrect VHDL (both 2008 and earlier standards) where ghdl correctly rejects the code.

While I am not absolutely certain this is the case here, it seems highly likely. There are precedents for relaxing some rules (with the -frelaxed-rules option) - would it be possible to add this one to the list?