ghdl/ghdl #467: Elaboration crashes on unconstrained record

NJDFan, 2017-11-25:
I can get the following code (with the attached package) to analyze correctly, but GHDL crashes on elaboration.  This was built with the mcode generator.  Apologies for turning AbstractMmPkg.vhd into a .txt file, it wouldn't take the upload otherwise and it's a lot of code to inline. 

```
******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.35-dev (2017-03-01-427-g0872dc63) [Dunoon edition]
Compiled with GNAT Version: 6.3.0
Target: x86_64-linux-gnu
In directory: /home/rgaddi/work/amr/
Command line:
ghdl -c --std=08 -g -e testbench
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : trans-chap7.adb:3932
Call stack traceback locations:
0x7f5363d93a11 0x564e40f0b77c 0x564e40f520ea 0x564e40f642ca 0x564e40f644ab 0x564e40f88dbd 0x564e40f704f7 0x564e40f8a0aa 0x564e40f352b6 0x564e40f930be 0x564e40f403a4 0x564e40f97c29 0x564e40fa11cd 0x564e40f9ee50 0x564e40dfe0be 0x564e40d0af08 0x7f53633733ef 0x564e40d09d08 0xfffffffffffffffe
******************************************************************
```

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

use work.AbstractMmPkg.all;

entity testbench is
end entity testbench;

architecture TB of testbench is

	signal rec : AbstractMmRecType(
		writedata(31 downto 0),
		readdata(31 downto 0),
		address(4 downto 0),
		byteen(3 downto 0)
	);

begin

end architecture TB;
```

[AbstractMmPkg.vhd](https://github.com/tgingold/ghdl/files/1502790/AbstractMmPkg.vhd.txt)

tgingold, 2017-11-25:
The aggregate in INIT_AMR (line 260) is not yet supported.

NJDFan, 2017-11-25:
Thanks @tgingold.  I think there's something deeper going on here.  I commented out the INIT_AMR code in AbstractMmPkg and the above elaborated successfully, but when I went to the next step of adding in my BFM I started getting 

```
******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.35-dev (2017-03-01-427-g0872dc63) [Dunoon edition]
Compiled with GNAT Version: 6.3.0
Target: x86_64-linux-gnu
In directory: /home/rgaddi/work/amr/
Command line:
ghdl -c --std=08 -g -e testbench
Exception ORTHO_CODE.SYNTAX_ERROR raised
Exception information:
raised ORTHO_CODE.SYNTAX_ERROR : ortho_code-exprs.adb:459
Call stack traceback locations:
0x55ba2ff1afc4 0x55ba2ff1cbb1 0x55ba2fe87aa4 0x55ba2ffa5686 0x55ba2ffa6c98 0x55ba2ffa7649 0x55ba2ffd5ea7 0x55ba2ffd1b25 0x55ba2ffd2388 0x55ba3001ecaf 0x55ba2ffbf85c 0x55ba3001d08c 0x55ba2ffca3a4 0x55ba30021c29 0x55ba3002b1cd 0x55ba30028e50 0x55ba2fe880be 0x55ba2fd94f08 0x7fe8b3e333ef 0x55ba2fd93d08 0xfffffffffffffffe
******************************************************************
```

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

use work.AbstractMmPkg.all;

entity testbench is
end entity testbench;

architecture TB of testbench is

	signal AWADDR	: std_logic_vector(6 downto 0);
	signal AWPROT	: std_logic_vector(2 downto 0);
	signal AWVALID	: std_logic;
	signal AWREADY	: std_logic;
	signal WDATA	: std_logic_vector(31 downto 0);
	signal WSTRB	: std_logic_vector(3 downto 0);
	signal WVALID	: std_logic;
	signal WREADY	: std_logic;
	signal BRESP	: std_logic_vector(1 downto 0);
	signal BVALID	: std_logic;
	signal BREADY	: std_logic;
	signal ARADDR	: std_logic_vector(6 downto 0);
	signal ARPROT	: std_logic_vector(2 downto 0);
	signal ARVALID	: std_logic;
	signal ARREADY	: std_logic;
	signal RDATA	: std_logic_vector(31 downto 0);
	signal RRESP	: std_logic_vector(1 downto 0);
	signal RVALID	: std_logic;
	signal RREADY	: std_logic;
	signal ACLK	    : std_logic;
	signal ARESETn	: std_logic;
	
	signal rec : AbstractMmRecType(
		writedata(31 downto 0),
		readdata(31 downto 0),
		address(4 downto 0),
		byteen(3 downto 0)
	);

begin

	BFM: entity work.axi_master
		generic map (
			DATAWIDTH => 32,
			ADDRWIDTH => AWADDR'length
		) port map(
			-- AXI interface,
			AWADDR	=> AWADDR,
			AWPROT	=> AWPROT,
			AWVALID	=> AWVALID,
			AWREADY	=> AWREADY,
			
			WDATA	=> WDATA,
			WSTRB	=> WSTRB,
			WVALID	=> WVALID,
			WREADY	=> WREADY,
			
			BRESP	=> BRESP,
			BVALID	=> BVALID,
			BREADY	=> BREADY,
			ARADDR	=> ARADDR,
			ARPROT	=> ARPROT,
			ARVALID	=> ARVALID,
			ARREADY	=> ARREADY,
			
			RDATA	=> RDATA,
			RRESP	=> RRESP,
			RVALID	=> RVALID,
			RREADY	=> RREADY,
			
			ACLK	=> ACLK,
			ARESETn	=> ARESETn,

			-- AMR interface
			amr => rec
		);
end architecture TB;
```
[axi_master.vhd](https://github.com/tgingold/ghdl/files/1502854/axi_master.vhd.txt)

diogratia, 2017-11-26:
I got your testbench to -e elaborate after analyzing osvvm 2015_05 and your commented out AbstractMmPkg.vhd with:

```
Commit:  5e534966512e77a05b0ebc944568c64c0c433a6b [5e53496]
Date:    November 2, 2017 at 5:50:34 PM GMT+13 (New Zealand)
```
```
ghdl --version
GHDL 0.35-dev (tarball) [Dunoon edition]
 Compiled with GNAT Version: GPL 2015 (20150428-49)
 llvm code generator
```
On OSX 10.10.5

HOWEVER, It failed to  -r run

I had trouble building osvvm_2017_05
```
ghdl -a --std=08 --work=osvvm --workdir=osvvm  CoveragePkg.vhd
CoveragePkg.vhd:1477:33:error: operator ""&"" is overloaded
CoveragePkg.vhd:1477:33:error: possible interpretations are:
*std_standard*:1:1:error: function "&" [string, character return string]
*std_standard*:1:1:error: function "&" [string, string return string]
ghdl:error: compilation error
```
That looks like a possible ghdl bug, there are two impure functions GetAlertLogName declared in AlertLogPkg.vhd differentiated by parameter type. Both return a string. AlertLogIDVar is a variable (this all appears to be in a protected type body):
```
grep -n "impure function GetAlertLogName" *.vhd
AlertLogPkg.vhd:281:  impure function GetAlertLogName(AlertLogID : AlertLogIDType := ALERTLOG_BASE_ID) return string ;
AlertLogPkg.vhd:474:    impure function GetAlertLogName(AlertLogID : AlertLogIDType) return string ;
AlertLogPkg.vhd:1016:    impure function GetAlertLogName(AlertLogID : AlertLogIDType) return string is
AlertLogPkg.vhd:2787:  impure function GetAlertLogName(AlertLogID : AlertLogIDType := ALERTLOG_BASE_ID) return string is
```
The first two are declarations for the second two function bodies.

Using a qualified expression:
```vhdl
        write(buf, Prefix & S & string'(GetAlertLogName(AlertLogIDVar))) ;
```
allowed the library analysis to complete. You could speculate this has something to do with protected types.

Anyway once the osvvm library was built:
```
ghdl -a --std=08 -P/Users/diogratia/Desktop/osvvm1/osvvm AbstractMmPkg.vhd
ghdl -a --std=08 -P/Users/diogratia/Desktop/osvvm1/osvvm axi_master.vhd
ghdl -a --std=08 -P/Users/diogratia/Desktop/osvvm1/osvvm testbench.vhdl
ghdl -e --std=08 -P/Users/diogratia/Desktop/osvvm1/osvvm testbench
```
Followed by the -r elaboration failure.
```
ghdl -r testbench
./testbench:error: NULL access dereferenced
./testbench:error: error during elaboration
```
(This is not an mcode version and -r doesn't require --std)

The second error above shows you used an mcode version and ortho_code-exprs.adb seems unchanged from your build:
```ada
   procedure Check_Value_Type (Val : O_Enode; Vtype : O_Tnode) is
   begin
      if Get_Enode_Type (Val) /= Vtype then
         raise Syntax_Error;
      end if;
   end Check_Value_Type;
```
Line 459 is the line with  raise Syntax_Error.

There are three calls to Check_Value_Type all in ortho_code-exprs.adb:
```
adafind -n Check_Value_Type
[... 48 lines cut ...]
    signal act_sig:     UnbRecType (writedata (7 downto 0));
begin
UUT:
    entity work.eut
        port map (
            form_sig => act_sig
        );
end architecture;
```
Analyze, elaborate and simulate:
```
ghdl -a --std=08 eu_tb.vhdl
ghdl -e --std=08 eu_tb
ghdl -r eu_tb
```
Or for an mcode version:
```
ghdl -a --std=08 eu_tb.vhdl
ghdl -r --std=08 eu_tb
```
And this gets the error. Experimenting around shows that changing the port mode from inout to in will stop the NULL access dereferenced error.  Switching the port mode to out and the error will show up.

So is it unique to unbound array elements? YES, providing an element subtype indication containing a constraint in the record declaration and removing the record constraint from the signal declaration will stop the error. We've definitely identified an MCVe.

Peculiar it only fails with a port mode of out or inout. That would seem finding the drivers for the port is not succeeding.

closed 2017-11-28 by tgingold

Paebbels, 2017-11-28:
Does this commit fix #453 too?

commit 93e1e6fa (2017-11-28): trans-chap3: unbox recod in Get_Composite_Base.

Fix #467