ghdl/ghdl #468: Spurious files after building ghdl gcc favor

andreygursky, 2017-11-26:
```
$ git status
On branch master
Your branch is up-to-date with 'origin/master'.

Untracked files:
  (use "git add <file>..." to include in what will be committed)

	Makefile
	libghdlvpi.so
	version.ads
	version.tmp

nothing added to commit but untracked files present (use "git add" to track)
```
Are instructions for gcc in BUILD.txt up-to-date? I've successfully built ghdl but it seems these files are not supposed to be in the source directory and should generated in the build one instead?

tgingold, 2017-11-27:
Can you post all the commands you have executed from untar point ?

Although it is possible that some files are generated in the source dir, I am pretty sure that Makefile is not.  Hence some doubts about this issue.

andreygursky, 2017-11-29:
```
cd PATH_TO/ghdl/ghdl.git
CC=gcc-7 ./configure --prefix=PATH_TO/ghdl/gcc/install --with-gcc=PATH_TO/ghdl/gcc/gcc-7/src
make copy-sources
cd ..
cd gcc-7
mkdir build
cd build
CC=gcc-7 CXX=g++-7 ../src/configure --prefix=PATH_TO/ghdl/gcc/install --enable-languages=c,vhdl --enable-default-pie --with-system-zlib --disable-bootstrap --disable-lto --disable-multilib --disable-libssp --disable-libgomp --disable-libquadmath --disable-libstdcxx
make
make install
cd ../..
cd ghdl.git
make ghdllib
make install
```
I've named the directory where I build gcc "build" instead of "gcc-objs".

In .gitignore there is already "Makefile", but in 68e9c3dbb0f74bbaa6e285960ac38428d4d137e0 you've renamed it to "./Makefile".

@Paebbels has included versions.ads in 97f4808fb4337065ad2a21d63e3d5bad3e03500c: "src/version.ads". I guess it should be generated in the same directory regardless OS.

> Although it is possible that some files are generated in the source dir, I am pretty sure that Makefile is not.

From what I see in BUILD.txt ghdl should be configured in it's top source directory (because of "./"):
```
First configure ghdl, specify gcc source dir and prefix.
(replace /usr/local by your prefix directory):
$ ./configure --with-gcc=/path/to/gcc/source/dir --prefix=/usr/local
```
Or should it be ```/path/to/ghdl/configure```? But then it is not clear what directory is recommended to invoke configuration. Is it simply:
```
mkdir ghdl-objs
cd ghdl-objs
../ghdl.git/configure ...
```
?

tgingold, 2017-11-30:
If you configure in the source directory, ghdl creates some files in the source directory.
But you can also configure in a different directory.

The configure supports both, and there is no preference.

I will explain that in the file.

closed 2017-11-30 by tgingold

andreygursky, 2017-11-30:
That's good, thanks for clarification.

Since configuration in the build directory is supported, shouldn't this bug be closed by a commit fixing .gitignore instead of manual closing?