ghdl/ghdl #471: "GHDL Bug Occurred"

gardners, 2017-11-28:
As described: GHDL reports a bug has occurred and provides the following error output:

Pauls-MacBook-Pro-3:newcpu gardners$ make
/usr/local/bin/ghdl -i cpu_test.vhdl debugtools.vhdl alu.vhdl instructions.vhdl instruction_lengths.vhdl addressing_modes.vhdl instruction_equations.vhdl extra_instruction_equations.vhdl ram0.vhdl ram1.vhdl ram2.vhdl ram3.vhdl address_translator.vhdl gs4502b.vhdl memory_controller.vhdl gs4502b_core.vhdl gs4502b_instruction_prefetch.vhdl gs4502b_stage_decode.vhdl gs4502b_stage_validate.vhdl gs4502b_stage_execute.vhdl disassemble.vhdl visualise.vhdl
/usr/local/bin/ghdl -m cpu_test

******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.35-dev (tarball) [Dunoon edition]
Compiled with GNAT Version: GPL 2016 (20160515-49)
Target: x86_64-apple-darwin14.5.0
In directory: /Users/gardners/Projects/c64accel/git/newcpu/
Command line:
/usr/local/bin/ghdl -m cpu_test
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : iirs.adb:138
Load address: 0x10c52b000
Call stack traceback locations:
0x10c841561 0x10c5f589e 0x10c697df3 0x10c697e30 0x10c7d7583 0x10c7dd6bf 0x10c7d9415 0x10c7dd6e2 0x10c7f42e4 0x10c7f5033 0x10c7c76f9 0x10c7c9332 0x10c7c9198 0x10c7c94c9 0x10c7ca11e 0x10c7ca182 0x10c7ca76f 0x10c7caab5 0x10c7fe27e 0x10c805484 0x10c6942f4 0x10c694b1f 0x10c6dcd74 0x10c6dc9c6 0x10c6dc59c 0x10c6dc393 0x10c6dcdf6 0x10c6dc9c6 0x10c6dc59c 0x10c6dc393 0x10c6dcdf6 0x10c6dc9c6 0x10c6dc59c 0x10c6dc393 0x10c6ddeda 0x10c6e88d3 0x10c7ba267 0x10c7b651d 0x10c61e65d 0x10c52d7dc
******************************************************************
make: *** [cpu_test] Error 2

This is with ghdl freshly built from the head of master at commit 0872dc6307d0f1ead0f2e109ac8fd9293a0792dc

gardners, 2017-11-28:
To reproduce error, checkout the following, and type: make

https://github.com/gardners/gs4502b/tree/ghdl-issue-471

diogratia, 2017-11-28:
With a slightly older ghdl
Commit:  5e534966512e77a05b0ebc944568c64c0c433a6b 
```
...
cat mega65ram.bin c65rom.bin > mega65mem.bin
./makeram mega65mem.bin
/usr/local/bin/ghdl -i cpu_test.vhdl debugtools.vhdl alu.vhdl instructions.vhdl instruction_lengths.vhdl addressing_modes.vhdl instruction_equations.vhdl extra_instruction_equations.vhdl ram0.vhdl ram1.vhdl ram2.vhdl ram3.vhdl address_translator.vhdl gs4502b.vhdl memory_controller.vhdl gs4502b_core.vhdl gs4502b_instruction_prefetch.vhdl gs4502b_stage_decode.vhdl gs4502b_stage_validate.vhdl gs4502b_stage_execute.vhdl disassemble.vhdl visualise.vhdl
/usr/local/bin/ghdl -m cpu_test
alu.vhdl:25:13:error: no declaration for "bytes4"
make: *** [cpu_test] Error 1
```
(llvm backend, OS X, built 4 Nov 2017 NZDST)

```
Macbook: ghdl -a alu.vhdl
alu.vhdl:9:10:error: package "instructions" is obsoleted by package "addressing_modes"
ghdl:error: compilation error
Macbook: ghdl -a instructions.vhdl
Macbook: ghdl -a alu.vhdl
alu.vhdl:25:13:error: no declaration for "bytes4"
ghdl:error: compilation error
```
A `grep -i bytes4 *.vhdl` search shows no declaration for subtype/type bytes4, implying a workaround to avoid the crash.

gardners, 2017-11-28:
The problem here is that that bytes4 is defined, in instruction_lengths.vhdl, which is pulled into visualise.vhdl.  Is the problem that GHDL is not seeing the definition for some reason (which may be my VHDL incompetence)?

Pauls-MacBook-Pro-3:newcpu gardners$ grep bytes4 *.vhdl
alu.vhdl:    bytes : bytes4;
gs4502b_core.vhdl:  signal vector_fetch_out_bytes : bytes4;
gs4502b_instruction_prefetch.vhdl:    vector_fetch_out_bytes : out bytes4;
gs4502b_stage_validate.vhdl:    vector_fetch_vector : in bytes4;
**instruction_lengths.vhdl:  type bytes4 is array (0 to 3) of std_logic_vector(8 downto 0);**
memory_controller.vhdl:  signal bram_bytes_out : bytes4;
memory_controller.vhdl:  function bytes_reorder(v : bytes4; rotate : unsigned(1 downto 0)) return bytes4 is
memory_controller.vhdl:      variable vout : bytes4;
visualise.vhdl:  function visualise(e : string; s : string; v : bytes4) return boolean;
visualise.vhdl:  function visualise(e : string; s : string; v : bytes4) return boolean is
visualise.vhdl:    report "VISUALISE:" & e & ":" & s & ":bytes4:"

diogratia, 2017-11-28:
```
alias vhdlfind 'find . -iname \*.vhd\* -exec grep -H !* \{\} \;'
```
and  
```
Macbook: vhdlfind -i bytes4
./alu.vhdl:    bytes : bytes4;
./gs4502b_core.vhdl:  signal vector_fetch_out_bytes : bytes4;
./gs4502b_instruction_prefetch.vhdl:    vector_fetch_out_bytes : out bytes4;
./gs4502b_stage_validate.vhdl:    vector_fetch_vector : in bytes4;
./memory_controller.vhdl:  signal bram_bytes_out : bytes4;
./memory_controller.vhdl:  function bytes_reorder(v : bytes4; rotate : unsigned(1 downto 0)) return bytes4 is
./memory_controller.vhdl:      variable vout : bytes4;
./visualise.vhdl:  function visualise(e : string; s : string; v : bytes4) return boolean;
./visualise.vhdl:  function visualise(e : string; s : string; v : bytes4) return boolean is
./visualise.vhdl:    report "VISUALISE:" & e & ":" & s & ":bytes4:"
```
no got in instruction_lengths.vhdl

This from downloading your https://github.com/gardners/gs4502b/tree/ghdl-issue-471 and downloading Ophis
https://github.com/gardners/Ophis/tree/98ffd2af95d1a4a1b9dac180c0bdf681f815d47a.

```
Macbook: rm instruction_lengths.vhdl
Macbook: make instruction_lengths.vhdl
Rule #0: OPCODE & $00b = $001 -> length is 2 (covers 64/276 remaining instructions).
Rule #1: OPCODE & $00d = $008 -> length is 1 (covers 64/212 remaining instructions).
Rule #2: OPCODE & $00e = $006 -> length is 2 (covers 64/148 remaining instructions).
Rule #3: OPCODE & $099 = $080 -> length is 2 (covers 24/124 remaining instructions).
Rule #4: OPCODE & $115 = $101 -> length is 2 (covers 24/100 remaining instructions).
Rule #5: OPCODE & $01e = $010 -> length is 2 (covers 16/84 remaining instructions).
Rule #6: OPCODE & $08c = $004 -> length is 2 (covers 16/68 remaining instructions).
Rule #7: OPCODE & $017 = $001 -> length is 2 (covers 8/60 remaining instructions).
Rule #8: OPCODE & $109 = $101 -> length is 2 (covers 8/52 remaining instructions).
Rule #9: OPCODE & $117 = $112 -> length is 1 (covers 8/44 remaining instructions).
Rule #10: OPCODE & $11b = $012 -> length is 2 (covers 8/36 remaining instructions).
Rule #11: OPCODE & $18e = $00a -> length is 1 (covers 8/28 remaining instructions).
Rule #12: OPCODE & $02c = $004 -> length is 2 (covers 4/24 remaining instructions).
Rule #13: OPCODE & $0b7 = $002 -> length is 1 (covers 4/20 remaining instructions).
Rule #14: OPCODE & $0d7 = $040 -> length is 1 (covers 4/16 remaining instructions).
Rule #15: OPCODE & $04c = $004 -> length is 2 (covers 2/14 remaining instructions).
Rule #16: OPCODE & $06a = $000 -> length is 2 (covers 2/12 remaining instructions).
Rule #17: OPCODE & $0b8 = $0a0 -> length is 2 (covers 2/10 remaining instructions).
Rule #18: OPCODE & $15e = $05a -> length is 1 (covers 2/8 remaining instructions).
Rule #19: OPCODE & $187 = $102 -> length is 1 (covers 2/6 remaining instructions).
Rule #20: OPCODE & $1b6 = $002 -> length is 1 (covers 2/4 remaining instructions).
Rule #21: OPCODE & $0d8 = $0c0 -> length is 2 (covers 1/3 remaining instructions).
Rule #22: OPCODE & $10c = $104 -> length is 2 (covers 1/2 remaining instructions).
Rule #23: OPCODE & $16b = $062 -> length is 2 (covers 1/1 remaining instructions).
Rule #24: OPCODE & $1fb = $058 -> length is 1 (covers 1/0 remaining instructions).
Verifying that 25 rules correctly classify lengths of all instructions.
Generating instruction_lengths.vhdl
Macbook: grep -i bytes4 instruction_lengths.vhdl
Macbook:
```
A prompt with no returned results.  The Ophis link matches the commit short descriptor Ophis @ 98ffd2a.

Your link doesn't provide a reproducer.

gardners, 2017-11-29:
I have simplified the structure, moving almost all type definitions into a new types.vhdl file, which is used via use work.types.all where required.  This specifically includes the definition of bytes4.

I am still seeing the same problem: 

Pauls-MacBook-Pro-3:newcpu gardners$ make
/usr/local/bin/ghdl -i cpu_test.vhdl debugtools.vhdl alu.vhdl types.vhdl instructions.vhdl instruction_lengths.vhdl addressing_modes.vhdl instruction_equations.vhdl extra_instruction_equations.vhdl ram0.vhdl ram1.vhdl ram2.vhdl ram3.vhdl address_translator.vhdl gs4502b.vhdl memory_controller.vhdl gs4502b_core.vhdl gs4502b_instruction_prefetch.vhdl gs4502b_stage_decode.vhdl gs4502b_stage_validate.vhdl gs4502b_stage_execute.vhdl disassemble.vhdl visualise.vhdl
/usr/local/bin/ghdl -m cpu_test

******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.35-dev (tarball) [Dunoon edition]
Compiled with GNAT Version: GPL 2016 (20160515-49)
Target: x86_64-apple-darwin14.5.0
In directory: /Users/gardners/Projects/c64accel/git/newcpu/
Command line:
/usr/local/bin/ghdl -m cpu_test
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : iirs.adb:138
Load address: 0x101a8e000
Call stack traceback locations:
0x101da4561 0x101b5889e 0x101bfadf3 0x101bfae30 0x101d3a583 0x101d406bf 0x101d3c415 0x101d406e2 0x101d572e4 0x101d58033 0x101d2a6f9 0x101d2c332 0x101d2c198 0x101d2c4c9 0x101d2d11e 0x101d2d182 0x101d2d76f 0x101d2dab5 0x101d6127e 0x101d68484 0x101bf72f4 0x101bf7b1f 0x101c3fd74 0x101c3f9c6 0x101c3f59c 0x101c3f393 0x101c3fdf6 0x101c3f9c6 0x101c3f59c 0x101c3f393 0x101c3fdf6 0x101c3f9c6 0x101c3f59c 0x101c3f393 0x101c40eda 0x101c4b8d3 0x101d1d267 0x101d1951d 0x101b8165d 0x101a907dc
******************************************************************
make: *** [cpu_test] Error 2
Pauls-MacBook-Pro-3:newcpu gardners$

Is there any way to get GHDL to provide a more meaningful error message in cases such as this?

gardners, 2017-11-29:
So, I have been working to try to resolve this through refactoring on my side, and have now hit what looks to be the core of the problem: Some definitions are not being recognised when invoked via use work.something.all or work.something.specific.  This is with commit b8c72628e8d85600dd5cff692789468e310b09a8 on github.com/gardners/gs4502b.
```

paul@F96NG92-L:~/Projects/mega65/gs4502b$ make
gcc -g -Wall -o makeram makeram.c
./makeram mega65mem.bin
/usr/local/bin/ghdl -a debugtools.vhdl types.vhdl instruction_equations.vhdl extra_instruction_equations.vhdl instruction_types.vhdl
/usr/local/bin/ghdl -i cpu_test.vhdl debugtools.vhdl alu.vhdl instruction_types.vhdl types.vhdl instructions.vhdl instruction_lengths.vhdl addressing_modes.vhdl instruction_equations.vhdl extra_instruction_equations.vhdl ram0.vhdl ram1.vhdl ram2.vhdl ram3.vhdl address_translator.vhdl gs4502b.vhdl memory_controller.vhdl gs4502b_core.vhdl gs4502b_instruction_prefetch.vhdl gs4502b_stage_decode.vhdl gs4502b_stage_validate.vhdl gs4502b_stage_execute.vhdl disassemble.vhdl visualise.vhdl
/usr/local/bin/ghdl -m cpu_test
gs4502b_core.vhdl:159:33:error: no declaration for "translated_address"
gs4502b_core.vhdl:174:31:error: no declaration for "translated_address"
gs4502b_core.vhdl:184:45:error: no declaration for "translated_address"
gs4502b_instruction_prefetch.vhdl:67:29:error: no declaration for "translated_address"
gs4502b_instruction_prefetch.vhdl:81:34:error: no declaration for "translated_address"
Makefile:34: die Regel für Ziel „cpu_test“ scheiterte
make: *** [cpu_test] Fehler

```

Note that translated_address is defined in types.vhdl, and seems to be able to be used for signals, but not for entities in the port map.

If I use a freshly compiled ghdl_llvm, I see much the same:

```
paul@F96NG92-L:~/Projects/mega65/gs4502b$ ghdl/ghdl_llvm -a debugtools.vhdl types.vhdl instruction_equations.vhdl extra_instruction_equations.vhdl instruction_types.vhdl instruction_lengths.vhdl visualise.vhdl gs4502b_instruction_prefetch.vhdl
gs4502b_instruction_prefetch.vhdl:67:29:error: no declaration for "translated_address"
gs4502b_instruction_prefetch.vhdl:81:34:error: no declaration for "translated_address"
ghdl/ghdl_llvm:error: compilation error
paul@F96NG92-L:~/Projects/mega65/gs4502b$ grep translated_address types.vhdl
  subtype translated_address is unsigned(31 downto 0);
    address : translated_address;
    translated : translated_address;
    translated : translated_address;

```

Have I made some classic blunder, or is there something going astray?

gardners, 2017-11-29:
Stand by: I have spotted some stupidities on my part in this, due to accidentally leaving some type defs in both of the split type definition packages.

diogratia, 2017-11-29:
>Is there any way to get GHDL to provide a more meaningful error message in cases such as this?

Sure, if the problem can be reproduced. The stack trace is mercurial (only useful when running in a debugger).

Run from fresh downloads (with your latest changes for types.vhdl):

```
...

./makeram mega65mem.bin
/usr/local/bin/ghdl -i cpu_test.vhdl debugtools.vhdl alu.vhdl types.vhdl instructions.vhdl instruction_lengths.vhdl addressing_modes.vhdl instruction_equations.vhdl extra_instruction_equations.vhdl ram0.vhdl ram1.vhdl ram2.vhdl ram3.vhdl address_translator.vhdl gs4502b.vhdl memory_controller.vhdl gs4502b_core.vhdl gs4502b_instruction_prefetch.vhdl gs4502b_stage_decode.vhdl gs4502b_stage_validate.vhdl gs4502b_stage_execute.vhdl disassemble.vhdl visualise.vhdl
/usr/local/bin/ghdl -m cpu_test

******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.35-dev (tarball) [Dunoon edition]
Compiled with GNAT Version: GPL 2015 (20150428-49)
Target: x86_64-apple-darwin13.4.0
In directory: /Users/david_koontz/Desktop/gs4502b-ghdl-issue-471/
Command line:
/usr/local/bin/ghdl -m cpu_test
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
Exception name: SYSTEM.ASSERTIONS.ASSERT_FAILURE
Message: configuration.adb:64
Load address: 0x1004e0000
Call stack traceback locations:
0x100674dc1 0x1005c0a09 0x1005c0b3c 0x1005c0b3c 0x1005c0b3c 0x1005c0b3c 0x1005c2537 0x1005cd203 0x1005ecb8e 0x1005dd8cc 0x1004ef97c 0x1004e2002
******************************************************************
make: *** [cpu_test] Error 2
david_koontz@Macbook: make
/usr/local/bin/ghdl -i cpu_test.vhdl debugtools.vhdl alu.vhdl types.vhdl instructions.vhdl instruction_lengths.vhdl addressing_modes.vhdl instruction_equations.vhdl extra_instruction_equations.vhdl ram0.vhdl ram1.vhdl ram2.vhdl ram3.vhdl address_translator.vhdl gs4502b.vhdl memory_controller.vhdl gs4502b_core.vhdl gs4502b_instruction_prefetch.vhdl gs4502b_stage_decode.vhdl gs4502b_stage_validate.vhdl gs4502b_stage_execute.vhdl disassemble.vhdl visualise.vhdl
/usr/local/bin/ghdl -m cpu_test

******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.35-dev (tarball) [Dunoon edition]
Compiled with GNAT Version: GPL 2015 (20150428-49)
Target: x86_64-apple-darwin13.4.0
In directory: /Users/david_koontz/Desktop/gs4502b-ghdl-issue-471/
Command line:
/usr/local/bin/ghdl -m cpu_test
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
Exception name: SYSTEM.ASSERTIONS.ASSERT_FAILURE
Message: configuration.adb:64
Load address: 0x103833000
Call stack traceback locations:
0x1039c7dc1 0x103913a09 0x103913b3c 0x103913b3c 0x103913b3c 0x103913b3c 0x103915537 0x103920203 0x10393fb8e 0x1039308cc 0x10384297c 0x103835002
******************************************************************
Macbook: ghdl -m cpu_test

******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.35-dev (tarball) [Dunoon edition]
Compiled with GNAT Version: GPL 2015 (20150428-49)
Target: x86_64-apple-darwin13.4.0
In directory: /Users/diogratia/Desktop/gs4502b-ghdl-issue-471/
Command line:
ghdl -m cpu_test
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
Exception name: SYSTEM.ASSERTIONS.ASSERT_FAILURE
Message: configuration.adb:64
Load address: 0x102429000
Call stack traceback locations:
0x1025bddc1 0x102509a09 0x102509b3c 0x102509b3c 0x102509b3c 0x102509b3c 0x10250b537 0x102516203 0x102535b8e 0x1025268cc 0x10243897c 0x10242b002
******************************************************************
Macbook:
```
And that error message is the same (with different stack traces, OS X's version of ASLR).

It's also a different file and line number.

I also noticed s use clause circular dependency between types.vhdl and instruction_equations.vhdl. There may be more.

gardners, 2017-11-29:
okay, after splitting types.vhdl into types.vhdl (with no external dependencies) and instruction_types.vhdl (which depends on the auto-generated instruction set description vhdl files), and correcting my previously mentioned stupid errors, it now works.

Thus, what I believe was happening is that types.vhdl and the auto-generated vhdl files had circular dependencies, and ghdl wasn't reporting this.  So perhaps the bug should be renamed to "ghdl gives misleading error with certain types of circular dependencies".

tgingold, 2017-11-29:
> On 29 Nov 2017, at 03:54, Dio Gratia <notifications@github.com> wrote:
> 
> Is there any way to get GHDL to provide a more meaningful error message in cases such as this?
> 
> Sure, if the problem can be reproduced. The stack trace is mercurial (only useful when running in a debugger).
> 
> Run from fresh downloads (with your latest changes for types.vhdl):

I cannot reproduce that crash.  What is the commit id of this latest change ?

gardners, 2017-11-29:
Let me see if I can produce a minimal crashing sample...

diogratia, 2017-11-29:
From his link:
https://github.com/gardners/gs4502b/tree/ghdl-issue-471
He's using ghdl built from https://github.com/tgingold/ghdl/tree/7eb4e6f322074102bc1b038fdf8c5bea1beb88ff 
which has a commit date of April 4, 2017 2:48 PM  (NZST)

gardners, 2017-11-29:
In terms of a minimum example that gives a mis-leading error message:

```
paul@F96NG92-L:~/Projects/mega65/mega65-core$ more src/vhdl/{a,b}.vhdl | more
::::::::::::::
src/vhdl/a.vhdl
::::::::::::::
use work.b.all;

package a is

  type type_a is record
    something : type_b;
  end record;

end package;
    
::::::::::::::
src/vhdl/b.vhdl
::::::::::::::
use work.a.all;

package b is

  type type_b is record
    something : type_a;
  end record;

end package;
    
paul@F96NG92-L:~/Projects/mega65/mega65-core$ ghdl -a src/vhdl/a.vhdl
src/vhdl/a.vhdl:6:17:error: type "type_b" is not visible here

```

The error should ideally be something like "circular depdency between a and b: type_a depends on type_b which depends on type_a"

commit d265cf41 (2017-12-01): Improve error message in case of circular dependency.

Fix #471
closed 2017-12-01 by tgingold