ghdl/ghdl #473: Unconstrained arrays; crash on elaboration.

NJDFan, 2017-11-29:
-v - v says it's translating (with code generation) architecture "tb" of "tbmem"

mcode generator, Linux, a fresh pull today (commit f16a286f2400fc3910f0eb3efcfbff1d9e558cba).  If I explicitly constrain the unsigned as (15 downto 0) and take same out of the ``myram :=`` line it works correctly.

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity tbmem is
end entity tbmem;

architecture TB of tbmem is

begin

    DM: process
        type t_ram is array(natural range <>) of unsigned;
        type p_ram is access t_ram;
        variable myram : p_ram;
    begin
        myram := new t_ram(0 to 31)(15 downto 0);
        for i in myram'range loop
            myram(i) := TO_UNSIGNED(i, 16);
        end loop;
        for i in myram'range loop
            report integer'image(i) & ": " & TO_HSTRING(myram(i));
        end loop;
        wait;
    end process DM;
    
end architecture TB;
```

```
******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.35-dev (2017-03-01-437-gf16a286f) [Dunoon edition]
Compiled with GNAT Version: 6.3.0
Target: x86_64-linux-gnu
In directory: /home/rgaddi/work/amr/
Command line:
ghdl -e --std=08 -v -v tbmem
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans.adb:1621
Call stack traceback locations:
0x5579098431a1 0x55790989205b 0x557909892f88 0x55790989499f 0x557909894985 0x55790981ac3b 0x55790981c533 0x557909817ec2 0x5579098659b3 0x557909874e6f 0x557909874f83 0x557909854571 0x557909858e4c 0x5579098a5745 0x5579098463dd 0x5579098a3b22 0x557909850f25 0x5579098a86bf 0x5579098b2be7 0x5579098af8e6 0x55790970f0be 0x55790961bf08 0x7fb653e2b3ef 0x55790961ad08 0xfffffffffffffffe
******************************************************************
```

tgingold, 2017-12-01:
Array of unbounded arrays are not yet supported.
(This is a duplicate of a frequently requested feature!).

NJDFan, 2017-12-01:
@tgingold I can imagine implementation of the feature is pretty complicated.  Would it be possible, as an interim step, to get a specific error that "Arrays of unbounded arrays are not yet supported" at either analysis or elaboration time rather than just the generic traceback?  That way at least it's a "This is a known thing, I don't need to report it" rather than "Behold this bug that I have found!".

tgingold, 2017-12-01:
> On 01 Dec 2017, at 07:26, NJDFan <notifications@github.com> wrote:
> 
> @tgingold I can imagine implementation of the feature is pretty complicated. Would it be possible, as an interim step, to get a specific error that "Arrays of unbounded arrays are not yet supported" at either analysis or elaboration time rather than just the generic traceback? That way at least it's a "This is a known thing, I don't need to report it" rather than "Behold this bug that I have found!".

Well, I prefer to implement features rather than marking them not implemented!

tgingold, 2018-11-22:
Now working.

closed 2018-11-22 by tgingold