ghdl/ghdl #477: Continuous Integration, releases and ready-to-use rolling images

1138-4EB, 2017-12-07:
THIS IS HEAVY WIP
The content of this comment and the referenced docker images are subject to change in the following hours. Some might be missing now. Check the latest succesful build in branch `builders` at https://travis-ci.org/1138-4EB/ghdl to better understand the explanations below. E.g. https://travis-ci.org/1138-4EB/ghdl/builds/311648067
Meanwhile, you might want to suscribe to the thread and wait for replies. I will remove this note and reply in a new comment when it is considered stable.

---

For a quick reference about the objective of this issue, see section `Use cases for the actually available images` below.

# Context

Some months ago I contributed to this repo introducing the usage of 
[Docker](https://www.docker.com/) in [travis-ci](https://travis-ci.org/tgingold/ghdl) (#272, #270). At that time, docker images were defined in a external repository ([1138-4EB/ghdl-tools](https://github.com/1138-4EB/ghdl-tools)). Therefore, the generation/updating and distribution was not in sync with this repo.

This summer @tgingold refactored the build system, introducing the following changes:

- For each docker image a file was created in `dist/linux/docker/build-<distribution>-<compiler>`.
- Some images where dropped and others added, resulting in the following list: `stretch-mcode`, `ubuntu14-mcode`, `ubuntu14-llvm-3.8` and `fedora26-mcode` (4).
- A variant to `stretch-mcode` was added: `stretch-mcode-gpl` (4+1).
- Instead of using a single job in travis-ci with multiple parallel containers, a different job was used for each image.
- An additional job was added for `macosx-mcode` (4+1+1).
- Logs were significantly improved, using travis folds and timing them.

This issue follows the path. On the one hand, the process is optimized. On the other hand, intermediate artifacts are published. Last, release distribution is simplified.

# Multistage builds of docker images

Thanks to [multistage builds](https://docs.docker.com/engine/userguide/eng-image/multistage-build/) introduced in latest docker releases, there is no need to have separate `build-ubuntu14+mcode` and `build-ubuntu14+llvm-3.8` files. Especially, because the latter must include all the dependencies already included in the first. A single file can be used instead:

``` Dockerfile
# [build] Ubuntu 14

FROM ubuntu:trusty AS mcode

RUN apt-get update -qq \
  && apt-get -y install --no-install-recommends \
            gnat gcc make zlib1g-dev \
  && apt-get autoclean && apt-get clean && apt-get autoremove

FROM mcode AS llvm-3.8

RUN apt-get update -qq \
  && apt-get -y install --no-install-recommends \
            llvm-3.8-dev clang-3.8 \
  && apt-get autoclean && apt-get clean && apt-get autoremove
```

Then, both images are built one after the other:

``` bash
docker build -t ghdl/build:ubuntu-mcode --target mcode - < Dockerfile
docker build -t ghdl/build:ubuntu-llvm-3.8 --target llvm-3.8 - < Dockerfile
```

This ensures that the second build uses the cache from the first, saving the time required to fetch the sources again.

# Separate build and run images

Build/dev dependencies are not the same as runtime dependencies. Right now a single image is built for each platform and compiler, and it is used to build and test. However, I think it would be "cleaner" to use separate images for the build and test procedures. Following with the example above, these would be the images with runtime dependencies only:

``` Dockerfile
# [run] Ubuntu 14

FROM ubuntu:trusty AS mcode

RUN apt-get update -qq \
  && apt-get -y install --no-install-recommends \
            libgnat-4.6 gcc libc6-dev \
  && apt-get autoclean && apt-get clean && apt-get autoremove

FROM mcode AS llvm-3.8
[... 346 lines cut ...]
&& cd build-mcode \
&& gnatdoc -Pghdl"
```

[image]

@tgingold, shall I open a different issue to analyze this?

# Project organization

In order to 'protect' the main repo, I suggest creating an organization. I see that [github.com/ghdl](https://github.com/ghdl) is a user with no history at all. Does it belong to any of you? If not, we might consider either writting to GitHub or using any other (say `ghdl-org`).

An organization would provide a specific repo for GitHub Pages: `ghdl.github.io`. Then, `github.com/ghdl/ghdl` can be an alias to `github.com/tgingold/ghdl`, or the other way round. I think that sources for the site can still be kept in the main repo.

# Pending issues

- [ ] filter stages 0 and 2
- [ ] add license, copying, etc. to the images? where should it be located?
- [ ] No tagging implemented yet. It is a rolling approach.
- [ ] I don't know how to limit the matrix programatically.
- [ ] Really skip an stage/job.
- [ ] Difference between mcode and mcode-gpl.
- [ ] macosx cannot deploy images, can it?
- [ ] Cross-compilation ARM, there ready-to-use Rpi3 docker-based SD imgs
- [ ] Fix `GHDL 0.35-dev (tarball) [Dunoon edition]`.

Paebbels, 2017-12-09:
I don't like splitting up the documentation. For now, you should finish our documention rework started almost a year ago.

1138-4EB, 2017-12-09:
> [@Paebbels] I don't like splitting up the documentation.

I think it is getting too long...

> [@tgingold 2017-02-14](https://github.com/tgingold/ghdl/issues/280#issuecomment-279595802)]
> The doc targets the end user. It is ok to have a chapter on building/installing ghdl (as well as a chapter on the internals), but if that gets too long, I think we'd better to have two doc.

---

> For now, you should finish our documention rework started almost a year ago.

It is my bad, I thought it was partially merged. Indeed, it was almost ready, so I rebased and cleaned it up. See #482.

1138-4EB, 2017-12-13:
I think it is ready to be reviewed now. At least [Use cases for the actually available images](#cases), [Let's have some fun!](#fun) and [Extending base images](#extend).

@flip111 @Nic30 the second demo is an example of how to get "parsing" info from GHDL. In the screen recording the VCD is used only, but you can see that many `disp_tree*.txt` files are created, along with `adder.html` and `adder.xml`. Note that you can mount the local dir in the container. This one liner will do it:

```
$(commnad -v winpty) docker run --rm -it -v "/$(pwd):/src" -w "//src" ghdl/ext:vunit bash -c " \
mkdir -pv elide \
&& curl -L https://github.com/1138-4EB/elide/archive/develop.tar.gz | tar xz -C elide --strip-components=1 \
&& ./elide/examples/full_adder/test.sh"
```

@Paebbels , please, let me know if any of the images fits your needs to run GHDL in CI environments.

Paebbels, 2017-12-17:
@1138-4EB I'm currently out of sync in many of my projects. I try to get them in sync in the Christmas holidays.

1138-4EB, 2018-05-27:
I moved some of the content from this issue to the new ghdl/docker repo. See: [README.md](https://github.com/ghdl/docker/blob/master/README.md), [USE_CASES.md](https://github.com/ghdl/docker/blob/master/USE_CASES.md) and ghdl/docker#2. Some other content was moved to [1138-4EB/ghdl-io](https://github.com/1138-4EB/ghdl-io) (see 1138-4EB/ghdl-io#1).

closed 2021-09-07 by tgingold