ghdl/ghdl #478: GHDL crash on missing ampersand

kenkendk, 2017-12-07:
Crash report is:
```
EX_ALUControl.vhdl:82:141:error: missing ";" at end of statement
parenthesis_name_to_procedure_call: cannot handle IIR_KIND_STRING_LITERAL8 (EX_ALUControl.vhdl:82:141)

******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.34 (tarball) [Dunoon edition]
Compiled with GNAT Version: GPL 2015 (20150428-49)
Target: x86_64-apple-darwin13.4.0
In directory: ~/vhdl/
Command line:
/usr/local/bin/ghdl1-llvm --std=93c --workdir=work -P/usr/local/lib/ghdl/v93/std/ -P/usr/local/lib/ghdl/v93/synopsys/ -c -o work/EX_ALUControl.o EX_ALUControl.vhdl
Exception TYPES.INTERNAL_ERROR raised
Exception information:
Exception name: TYPES.INTERNAL_ERROR
Message: errorout.adb:246
Load address: 0x10ca31000
Call stack traceback locations:
0x10ca95f36 0x10cc41f81 0x10cc4248a 0x10cc42bc1 0x10cc43a1a 0x10cc4603a 0x10cc467cc 0x10cc4849d 0x10cc48620 0x10cae5efc 0x10cae60d7 0x10cbeb9c6 0x10cbec53a 0x10ca79cc3 0x10ca34218
******************************************************************
ghdl:error: compilation error
make: *** [work/EX_ALUControl.o] Error 1
```

The offending line is missing an ampersand for the last concatenation:
```
report "In EX_ALUControl (RST=" & Std_logic'image(RST) & ", RDY=" & Std_logic'image(RDY) & ", re=" & Std_logic'image(reentry_guard) ")";
```

And of course it should have been:
```
report "In EX_ALUControl (RST=" & Std_logic'image(RST) & ", RDY=" & Std_logic'image(RDY) & ", re=" & Std_logic'image(reentry_guard) & ")";
```

I can provide the entire VHDL source if you need it for reproducing.

diogratia, 2017-12-07:
There's enough information to construct a simple case:

```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity EX_ALUControl is
end entity;

architecture foo of EX_ALUControl is
    signal RST:             std_logic;
    signal RDY:             std_logic;
    signal reentry_guard:   std_logic;
begin

NO_LABEL:
    process (RST, RDY, reentry_guard)
    begin
        report "In EX_ALUControl (RST=" & Std_logic'image(RST) & ", RDY=" & 
        Std_logic'image(RDY) & ", re=" & 
        Std_logic'image(reentry_guard) ")"; -- MISSING AMPERSAND
    end process;
end architecture;
```
This reproduces your error:

```
Macbook: ghdl -a kenkendk.vhdl
kenkendk.vhdl:18:40:error: missing ";" at end of statement
parenthesis_name_to_procedure_call: cannot handle IIR_KIND_STRING_LITERAL8 (kenkendk.vhdl:18:40)

******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.35-dev (tarball) [Dunoon edition]
Compiled with GNAT Version: GPL 2015 (20150428-49)
Target: x86_64-apple-darwin13.4.0
In directory: /Users/diogratia/Desktop/
Command line:
/usr/local/bin/ghdl1-llvm -P/usr/local/lib/ghdl/v93/std/ -P/usr/local/lib/ghdl/v93/ieee/ -c -o kenkendk.o kenkendk.vhdl
Exception TYPES.INTERNAL_ERROR raised
Exception information:
Exception name: TYPES.INTERNAL_ERROR
Message: errorout.adb:246
Load address: 0x101739000
Call stack traceback locations:
0x1017a1a52 0x1019546a9 0x101954bb6 0x101955649 0x10195638e 0x101958c42 0x101959456 0x10195b288 0x10195b40c 0x1017f2f24 0x1017f30ff 0x1018fdaea 0x1018fe65e 0x101785313 0x10173ac8c
******************************************************************
ghdl:error: compilation error
Macbook:
```
Adding the missing ampersand allows the example to successfully analyze.

tgingold, 2017-12-08:
Thanks and thank you for the reproducer.
I will commit a fix soon.

commit 151a1aff (2017-12-08): Avoid crash on string as prefix of procedure call.

Fix #478
closed 2017-12-08 by tgingold