ghdl/ghdl #479: Failed to Biuld with GCC backend

WooWapDaBug, 2017-12-08:
I can build gcc without any errors, but when I include vhdl after using "make copy-sources" as explained in BULD.txt the build crashes reporting building errors such as: 

```
make[3]: Entering directory `/data/users/woowap/build_gcchdl/gcc'
/data/users/woowap/build_gcchdl/./prev-gcc/xg++ -B/data/users/woowap/build_gcchdl/./prev-gcc/ -B/data/users/woowap/gcc-ghdl/x86_64-pc-linux-gnu/bin/ -nostdinc++ -B/data/users/woowap/build_gcchdl/prev-x86_64-pc-linux-gnu/libstdc++-v3/src/.libs -B/data/users/woowap/build_gcchdl/prev-x86_64-pc-linux-gnu/libstdc++-v3/libsupc++/.libs  -I/data/users/woowap/build_gcchdl/prev-x86_64-pc-linux-gnu/libstdc++-v3/include/x86_64-pc-linux-gnu  -I/data/users/woowap/build_gcchdl/prev-x86_64-pc-linux-gnu/libstdc++-v3/include  -I/data/users/woowap/gcc-sources/libstdc++-v3/libsupc++ -L/data/users/woowap/build_gcchdl/prev-x86_64-pc-linux-gnu/libstdc++-v3/src/.libs -L/data/users/woowap/build_gcchdl/prev-x86_64-pc-linux-gnu/libstdc++-v3/libsupc++/.libs -no-pie   -g -O2 -gtoggle -DIN_GCC     -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -static-libstdc++ -static-libgcc  -o cc1 c/c-lang.o c-family/stub-objc.o attribs.o c/c-errors.o c/c-decl.o c/c-typeck.o c/c-convert.o c/c-aux-info.o c/c-objc-common.o c/c-parser.o c/c-array-notation.o c/c-fold.o c/gimple-parser.o c-family/c-common.o c-family/c-cppbuiltin.o c-family/c-dump.o c-family/c-format.o c-family/c-gimplify.o c-family/c-indentation.o c-family/c-lex.o c-family/c-omp.o c-family/c-opts.o c-family/c-pch.o c-family/c-ppoutput.o c-family/c-pragma.o c-family/c-pretty-print.o c-family/c-semantics.o c-family/c-ada-spec.o c-family/c-cilkplus.o c-family/array-notation-common.o c-family/cilk.o c-family/c-ubsan.o c-family/c-attribs.o c-family/c-warn.o i386-c.o glibc-c.o \
	  cc1-checksum.o libbackend.a main.o libcommon-target.a libcommon.a ../libcpp/libcpp.a ../libdecnumber/libdecnumber.a libcommon.a ../libcpp/libcpp.a   ../libbacktrace/.libs/libbacktrace.a ../libiberty/libiberty.a ../libdecnumber/libdecnumber.a   -L/data/users/woowap/build_gcchdl/./gmp/.libs -L/data/users/woowap/build_gcchdl/./mpfr/src/.libs -L/data/users/woowap/build_gcchdl/./mpc/src/.libs -lmpc -lmpfr -lgmp -rdynamic -ldl  -L./../zlib -lz
c/c-decl.o: In function `c_write_global_declarations_1(tree_node*)':
c-decl.c:(.text+0x5aa): undefined reference to `global_options'
c/c-decl.o: In function `decl_jump_unsafe(tree_node*)':
c-decl.c:(.text+0x11cc): undefined reference to `global_options'
c/c-decl.o: In function `warn_if_shadowing(tree_node*)':
c-decl.c:(.text+0x1662): undefined reference to `global_options'
c-decl.c:(.text+0x166d): undefined reference to `global_options'
c-decl.c:(.text+0x1833): undefined reference to `global_options'
c/c-decl.o:c-decl.c:(.text+0x1878): more undefined references to `global_options' follow
dwarf2out.o: In function `adjust_piece_list(rtx_def**, rtx_def**, rtx_def**, long, long, long, rtx_def*)':
dwarf2out.c:(.text+0x423b): undefined reference to `free_EXPR_LIST_node(rtx_def*)'
dwarf2out.o: In function `_ZL25include_pubname_in_outputP3vecI14pubname_struct5va_gc8vl_embedEPS0_.isra.156':
```

and some other undefined references such us:
```
dwarf2out.c:(.text+0x5259): undefined reference to `global_options'
loop-iv.c:(.text+0x2504): undefined reference to `free_EXPR_LIST_list(rtx_expr_list**)'
loop-iv.c:(.text+0x4df4): undefined reference to `alloc_EXPR_LIST(int, rtx_def*, rtx_def*)'
```

I've tried different gcc versions with no success. Any clue what am I missing?
I'm using this:

latest version from ghdl-master from github
gcc 7.2.0 source files with 
gmp-6.1.2
mpc-1.0.3
mpfr-3.16

Thank you for the help!

Brian-Drummond, 2017-12-08:
Which platform?
The failing command suggests it falls over building the C compiler ( -o cc1 ) using ./prev-gcc/xg++.

Which makes me wonder what Configure command you used (configuring gcc, before make) - specifically, you need --disable-bootstrap, and the above suggests a bootstrap build is happening.

WooWapDaBug, 2017-12-14:
I'm trying to build on Redhat 6. I first compiled GCC 7.2, and no problems. 
First I configure ghdl:
cd /build_ghdl
/ghdl-master/configure --prefix=/install_directory --with-gcc=/gcc-sources/ 
&&
make copy-sources

Then I configure gcc for ghdl:
cd /build_gcchdl
/gcc-sources/configure --prefix=/install_directory --enable-languages=c,vhdl
  --disable-bootstrap --disable-lto --disable-multilib
  --disable-libssp --disable-libgomp --disable-libquadmath
&&
 make -j8

WooWapDaBug, 2017-12-14:
When I configure gcc for ghdl I get the following error: 

```
checking for isl 0.15 or later... no
required isl version is 0.15 or later
*** This configuration is not supported in the following subdirectories:
     target-libquadmath gnattools gotools target-libada target-libhsail-rt target-libstdc++-v3 target-libgfortran target-libbacktrace target-libgo target-libffi target-libobjc target-libgomp target-liboffloadmic target-libssp target-libcilkrts target-libitm target-libsanitizer target-libvtv
    (Any other directories should still work fine.)
```

Maybe it has to do something with this.

WooWapDaBug, 2017-12-14:
The first error I posted was when trying to compile with GCC 4.4.7

Now with GCC 7.2 I get this:

```
+===========================GNAT BUG DETECTED==============================+
| 7.2.0 (x86_64-pc-linux-gnu) GCC error:                                   |
| in add_gnat_descriptive_type_attribute, at dwarf2out.c:19614             |
| Error detected around /data/users/a27503408/gcc-source/gcc/vhdl/trans-chap8.adb:1762:8|
| Please submit a bug report; see https://gcc.gnu.org/bugs/ .              |
| Use a subject line meaningful to you and us to track the bug.            |
| Include the entire contents of this bug box in the report.               |
| Include the exact command that you entered.                              |
| Also include sources listed below.                                       |
+==========================================================================+

Please include these source files with error report
Note that list may not be accurate in some cases,
so please double check that the problem can still
be reproduced with the set of files listed.
Consider also -gnatd.n switch (see debug.adb).

/data/users/a27503408/gcc-source/gcc/vhdl/trans-chap8.adb
../../gcc-source/gcc/vhdl/trans-chap8.ads
../../gcc-source/gcc/vhdl/trans.ads
../../gcc-source/gcc/vhdl/ortho_nodes.ads
../../gcc-source/gcc/vhdl/ortho_gcc.ads
../../gcc-source/gcc/vhdl/ortho_ident.ads
../../gcc-source/gcc/vhdl/iirs.ads
../../gcc-source/gcc/vhdl/types.ads
../../gcc-source/gcc/vhdl/tokens.ads
../../gcc-source/gcc/vhdl/nodes.ads
../../gcc-source/gcc/vhdl/lists.ads
../../gcc-source/gcc/vhdl/flists.ads
../../gcc-source/gcc/vhdl/std_names.ads
../../gcc-source/gcc/vhdl/errorout.ads
../../gcc-source/gcc/vhdl/iir_chains.ads
../../gcc-source/gcc/vhdl/iir_chain_handling.ads
../../gcc-source/gcc/vhdl/nodes_meta.ads
../../gcc-source/gcc/vhdl/canon.ads
../../gcc-source/gcc/vhdl/evaluation.ads
../../gcc-source/gcc/vhdl/std_package.ads
../../gcc-source/gcc/vhdl/iirs_utils.ads
../../gcc-source/gcc/vhdl/trans-chap2.ads
../../gcc-source/gcc/vhdl/trans-chap3.ads
../../gcc-source/gcc/vhdl/trans-chap4.ads
../../gcc-source/gcc/vhdl/trans-chap6.ads
../../gcc-source/gcc/vhdl/trans-chap7.ads
../../gcc-source/gcc/vhdl/trans-chap9.ads
../../gcc-source/gcc/vhdl/trans-chap14.ads
../../gcc-source/gcc/vhdl/trans_decls.ads
../../gcc-source/gcc/vhdl/translation.ads
../../gcc-source/gcc/vhdl/trans-helpers2.ads
../../gcc-source/gcc/vhdl/trans-foreach_non_composite.ads
../../gcc-source/gcc/vhdl/trans-foreach_non_composite.adb

raised TYPES.UNRECOVERABLE_ERROR : comperr.adb:406
gnatmake: "/data/users/a27503408/gcc-source/gcc/vhdl/trans-chap8.adb" compilation error
make[2]: *** [ghdl1] Error 4
rm gcc.pod
make[2]: Leaving directory `/data/users/a27503408/build_gccghdl/gcc'
make[1]: *** [all-gcc] Error 2
make[1]: Leaving directory `/data/users/a27503408/build_gccghdl'
make: *** [all] Error 2

```

1138-4EB, 2017-12-14:
@Bamban just report in #490 that it was successfully built on redhat7.4 and gcc 7.2. However, I don't know what does he mean with *I followed the "old" method with tweaks from the "new", such as make vhdllibs*.

I restored the info lost when BUILD.txt was removed. While it is merged, you can find it here:

- http://ghdl-devfork.readthedocs.io/en/doc-gcc/building/gcc/index.html
- http://ghdl-devfork.readthedocs.io/en/doc-gcc/building/gcc/GNULinux-GNAT.html

Let me know if you find anything which should be fixed.

Bamban, 2017-12-14:
Just to make it clear, I only checked out to the commit where BUILD.txt was in the root directory (87a5d4a) and followed the instruction there. It appears that the build method has not changed since, so it worked.

WooWapDaBug, 2017-12-14:
It doesn't work for me. I keep getting the GNAT bug detected error that I pasted before. The output of configure looks like this. I fixed the problem with isl library, I don't know what else I'm missing.

```
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking target system type... x86_64-pc-linux-gnu
checking for a BSD-compatible install... /usr/bin/install -c
checking whether ln works... yes
checking whether ln -s works... yes
checking for a sed that does not truncate output... /bin/sed
checking for gawk... gawk
checking for libatomic support... yes
checking for libcilkrts support... yes
checking for libitm support... yes
checking for libsanitizer support... yes
checking for libvtv support... yes
checking for libmpx support... yes
checking for libhsail-rt support... yes
checking for gcc... gcc
checking for C compiler default output file name... a.out
checking whether the C compiler works... yes
checking whether we are cross compiling... no
checking for suffix of executables...
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking whether g++ accepts -static-libstdc++ -static-libgcc... yes
checking for gnatbind... gnatbind
checking for gnatmake... gnatmake
checking whether compiler driver understands Ada... yes
checking how to compare bootstrapped objects... cmp --ignore-initial=16 $$f1 $$f2
checking for objdir... .libs
checking for the correct version of gmp.h... yes
checking for the correct version of mpfr.h... yes
checking for the correct version of mpc.h... yes
checking for the correct version of the gmp/mpfr/mpc libraries... yes
checking for isl 0.15 or later... yes
The following languages will be built: c,vhdl
*** This configuration is not supported in the following subdirectories:
     target-libquadmath gnattools gotools target-libada target-libhsail-rt target-libstdc++-v3 target-libgfortran target-libbacktrace target-libgo target-libffi target-libobjc target-libgomp target-liboffloadmic target-libssp target-libcilkrts target-libitm target-libsanitizer target-libvtv
    (Any other directories should still work fine.)
checking for default BUILD_CONFIG...
checking for --enable-vtable-verify... no
checking for bison... no
checking for byacc... no
checking for yacc... no
checking for bison... no
checking for gm4... no
checking for gnum4... no
checking for m4... m4
checking for flex... flex
checking for flex... flex
checking for makeinfo... makeinfo
checking for expect... expect
checking for runtest... runtest
checking for ar... ar
checking for as... as
checking for dlltool... no
checking for ld... ld
checking for lipo... no
checking for nm... nm
checking for ranlib... ranlib
checking for strip... strip
checking for windres... no
checking for windmc... no
checking for objcopy... objcopy
checking for objdump... objdump
checking for readelf... readelf
checking for cc... cc
checking for c++... c++
checking for gcc... gcc
checking for gfortran... gfortran
checking for gccgo... no
checking for ar... no
checking for ar... ar
checking for as... no
checking for as... as
checking for dlltool... no
checking for dlltool... no
checking for ld... no
checking for ld... ld
checking for lipo... no
checking for lipo... no
checking for nm... no
checking for nm... nm
checking for objcopy... no
checking for objcopy... objcopy
checking for objdump... no
checking for objdump... objdump
checking for ranlib... no
checking for ranlib... ranlib
checking for readelf... no
checking for readelf... readelf
checking for strip... no
checking for strip... strip
checking for windres... no
checking for windres... no
checking for windmc... no
checking for windmc... no
checking where to find the target ar... host tool
checking where to find the target as... host tool
checking where to find the target cc... just compiled
checking where to find the target c++... host tool
checking where to find the target c++ for libstdc++... host tool
checking where to find the target dlltool... host tool
checking where to find the target gcc... just compiled
checking where to find the target gfortran... host tool
checking where to find the target gccgo... host tool
checking where to find the target ld... host tool
checking where to find the target lipo... host tool
checking where to find the target nm... host tool
checking where to find the target objcopy... host tool
checking where to find the target objdump... host tool
checking where to find the target ranlib... host tool
checking where to find the target readelf... host tool
checking where to find the target strip... host tool
checking where to find the target windres... host tool
checking where to find the target windmc... host tool
checking whether to enable maintainer-specific portions of Makefiles... no
configure: creating ./config.status
config.status: creating Makefile
```

1138-4EB, 2017-12-15:
@WooWapDaBug this worked for me. Hope it can be useful to debug your case.

```
docker run --rm -it fedora:26 bash -c " \
set -e \
&& dnf install --nodocs -y gcc-gnat gcc gcc-c++ zlib-devel lbzip2 flex autogen texinfo dejagnu make \
&& dnf clean all \
&& mkdir gcc-srcs \
&& curl -L https://github.com/gcc-mirror/gcc/archive/gcc-7_2_0-release.tar.gz | tar -xz -C gcc-srcs --strip-components=1 \
&& cd gcc-srcs \
&& ./contrib/download_prerequisites \
&& cd .. \
&& mkdir ghdl \
&& curl -L https://github.com/tgingold/ghdl/archive/v0.35.tar.gz | tar -xz -C ghdl --strip-components=1 \
&& cd ghdl \
&& mkdir build; cd build \
&& ../configure --with-gcc=../../gcc-srcs --prefix="/opt/ghdl" \
&& make copy-sources \
&& mkdir gcc-objs; cd gcc-objs \
&& mkdir /install-ghdl \
&& ../../../gcc-srcs/configure --prefix="/opt/ghdl" --enable-languages=c,vhdl --disable-bootstrap --disable-lto --disable-multilib --disable-libssp --disable-libgomp --disable-libquadmath --enable-shared CFLAGS=-fPIC \
&& make -j\$(nproc) && make install \
&& cd .. \
&& make ghdllib \
&& make install \
&& cd ../testsuite \
&& export GHDL=/opt/ghdl/bin/ghdl \
&& ./testsuite.sh"
```

tgingold, 2017-12-15:
One possible workaround for the crash in the gnat compiler is not to build with -g.
But you'd better to use a compiler that doesn't crash.

WooWapDaBug, 2017-12-15:
True, removing the -g option works. But I'm using gcc 7.2 which gnat compiler do you suggest using?

WooWapDaBug, 2017-12-15:
Wo! Thank you @1138-4EB ! That did the trick. I'm actually not sure what solved it. I also specified the locations of all the dependencies, including zlib. I'm not administrator of this system, so all libraries have not the default paths. Maybe it was messing with an old zlib version. The command that did it is:

../gcc-source/configure --prefix=/home/user/ghdl --enable-languages=c,vhdl --disable-bootstrap --disable-lto --disable-multilib --disable-libssp --disable-libgomp --disable-libquadmath --enable-shared CFLAGS=-fPIC --with-isl=/home/user/isl-0.15 --with-gmp=/home/user/gmp-6.1.2/ --with-mpc=/home/user/mpc-1.0.3/ --with-mpfr=/home/user/mpfr-3.1.6/ --with-gcc=/home/user/gcc-7.2.0 --with-system-zlib=/home/user/zlib-1.2.11

closed 2017-12-15 by WooWapDaBug

1138-4EB, 2017-12-15:
> Wo! Thank you @1138-4EB ! That did the trick. I'm actually not sure what solved it. I added the enable-shared flag and the CFLAG=-fPIC (not sure what these two do)

Indeed, I am not either :D. Just looked for solutions to the errors I had and seemed legit. Anyway, the script I posted was just a quick check to verify the docs. For example, having to use `--with-gmp` and so, which you had to do, is documented already.

@tgingold any clue of why we need to configure with `--enable-shared CFLAG=-fPIC`?

andreygursky, 2017-12-15:
> any clue of why we need to configure with --enable-shared CFLAG=-fPIC?

If you need to create a shared library, the corresponding sources must be compiled with -fPIC. But normally this will be set auto-magically for you where needed.

I build successfully without specifying these explicitly.

1138-4EB, 2017-12-15:
Thanks @andreygursky . Hope you don't mind a couple of related questions:

If I don't get it wrong we specify a PREFIX (say `/opt/ghdl`) and the result we get there is a custom built gcc with ghdl as a frontend. Then, when we talk about 'shared libraries', we are referring to objects which will lie in `/opt/ghdl` and will be used either by our custom gcc or by ghdl itself. There is no other library which is created outside `/opt/ghdl` that needs to be included in the tarball, is it?

Does it make any sense to create a static single-binary version of GHDL with GCC backend?

Which platform did you build it on? I mean the distribution and version. I am currently trying it on Fedora 26 and Debian Stretch. Fedora works pretty well: https://travis-ci.org/1138-4EB/ghdl/jobs/316742904 But Stretch fails: https://travis-ci.org/1138-4EB/ghdl/jobs/316742903 Indeed, the error in Stretch is:

```
analyze hello.vhdl
elaborate and simulate hello
/usr/bin/ld: e~hello.o: relocation R_X86_64_32 against symbol `work__hello__ARCH__behav__RTI' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: /work/install-gcc/lib/ghdl/v93/std/std_standard.o: relocation R_X86_64_32 against `.rodata' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: hello.o: relocation R_X86_64_32 against `.rodata' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: Nonrepresentable section on output
collect2: error: ld returned 1 exit status
```

even though, it is explictly set, as shown above.

Note that exactly the same code is used to build both of them: https://github.com/1138-4EB/ghdl/blob/gcc-in-docker/dist/linux/buildtest.sh#L92-L121

andreygursky, 2017-12-15:
> If I don't get it wrong we specify a PREFIX (say /opt/ghdl) and the result we get there is a custom built gcc with ghdl as a frontend.

I see we build both: gcc (C) and ghdl (VHDL: gcc with custom ghdl frontend).

> Then, when we talk about 'shared libraries', we are referring to objects which will lie in /opt/ghdl and will be used either by our custom gcc or by ghdl itself. There is no other library which is created outside /opt/ghdl that needs to be included in the tarball, is it?

As always, after you invoke ```make install``` all needed stuff lands in the specified prefix and you don't need anything else for the package (which then will likely have some dependencies).

> Does it make any sense to create a static single-binary version of GHDL with GCC backend?

I don't know whether it is possible. And if it would be possible, this would make sense only if object files for all different VHDL standards could be included in the same binary.

> Which platform did you build it on?

Debian testing (Buster).

> But Stretch fails

Maybe verbose flag would help?

andreygursky, 2017-12-15:
> But Stretch fails: https://travis-ci.org/1138-4EB/ghdl/jobs/316742903

I cannot find configure options supplied to gcc.

1138-4EB, 2017-12-15:
> I see we build both: gcc (C) and ghdl (VHDL: gcc with custom ghdl frontend).

Does this mean that we can get rid of gcc and save only ghdl to the tarball? Or does GHDL somehow use this gcc and it won't work with any other 'vanilla' gcc?

> As always, after you invoke make install all needed stuff lands in the specified prefix and you don't need anything else for the package (which then will likely have some dependencies).

While this is true for the sources which are being built, GHDL has additional runtime dependencies, such as libgnat, zlib, gcc and (if LLVM backend is used) libllvm. Therefore, I'd like to guess which which of these can be retrieved through the package manager and which must be included in the tarball. See question about gcc above.

> I don't know whether it is possible. And if it would be possible, this would make sense only if object files for all different VHDL standards could be included in the same binary.

Then, I'll forget about it.

> Debian testing (Buster).
> Maybe verbose flag would help?

I will try.

> I cannot find configure options supplied to gcc.

Line 109: https://github.com/1138-4EB/ghdl/blob/gcc-in-docker/dist/linux/buildtest.sh#L92-L121

andreygursky, 2017-12-15:
> Does this mean that we can get rid of gcc and save only ghdl to the tarball? Or does GHDL somehow use this gcc and it won't work with any other 'vanilla' gcc?

I'm sure gcc stuff can be removed.

> While this is true for the sources which are being built, GHDL has additional runtime dependencies, such as libgnat, zlib, gcc and (if LLVM backend is used) libllvm. Therefore, I'd like to guess [...]

A typical GNU/Linux distribution has corresponding packages. E.g. in Debian: zlib1g, libgnat-7, libgmp10, libmpc3, libmpfr4, libisl15.

> Line 109

OK.

1138-4EB, 2017-12-15:
Thanks a lot for letting me know that gcc can be removed and dependencies can be retrieved from the package manager.

I tried buster, but I get the same result as with stretch. I compared the log output of the debian run with the fedora run. These are the differences:

``` bash
DEBIAN
 /bin/mkdir -p '/work/install-gcc/lib/../lib'
 /bin/bash ./libtool   --mode=install /usr/bin/install -c   libcc1.la '/work/install-gcc/lib/../lib'
libtool: install: /usr/bin/install -c .libs/libcc1.so.0.0.0 /work/install-gcc/lib/../lib/libcc1.so.0.0.0
libtool: install: (cd /work/install-gcc/lib/../lib && { ln -s -f libcc1.so.0.0.0 libcc1.so.0 || { rm -f libcc1.so.0 && ln -s libcc1.so.0.0.0 libcc1.so.0; }; })
libtool: install: (cd /work/install-gcc/lib/../lib && { ln -s -f libcc1.so.0.0.0 libcc1.so || { rm -f libcc1.so && ln -s libcc1.so.0.0.0 libcc1.so; }; })
libtool: install: /usr/bin/install -c .libs/libcc1.lai /work/install-gcc/lib/../lib/libcc1.la
libtool: finish: PATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/sbin" ldconfig -n /work/install-gcc/lib/../lib
...
/work/install-gcc/lib/../lib
...
./lib/
...
./lib/libcc1.so.0.0.0
...
./lib/libcc1.so.0
...
./lib/gcc/x86_64-pc-linux-gnu/7.2.0/include-fixed/x86_64-linux-gnu/
./lib/gcc/x86_64-pc-linux-gnu/7.2.0/include-fixed/linux/
./lib/gcc/x86_64-pc-linux-gnu/7.2.0/include-fixed/linux/a.out.h
...
./lib/gcc/x86_64-pc-linux-gnu/7.2.0/crtprec80.o
...
./lib/libcc1.so
./lib/libcc1.la
```
``` bash
FEDORA
 /usr/bin/mkdir -p '/work/install-gcc/lib/../lib64'
 /bin/sh ./libtool   --mode=install /usr/bin/install -c   libcc1.la '/work/install-gcc/lib/../lib64'
libtool: install: /usr/bin/install -c .libs/libcc1.so.0.0.0 /work/install-gcc/lib/../lib64/libcc1.so.0.0.0
libtool: install: (cd /work/install-gcc/lib/../lib64 && { ln -s -f libcc1.so.0.0.0 libcc1.so.0 || { rm -f libcc1.so.0 && ln -s libcc1.so.0.0.0 libcc1.so.0; }; })
libtool: install: (cd /work/install-gcc/lib/../lib64 && { ln -s -f libcc1.so.0.0.0 libcc1.so || { rm -f libcc1.so && ln -s libcc1.so.0.0.0 libcc1.so; }; })
libtool: install: /usr/bin/install -c .libs/libcc1.lai /work/install-gcc/lib/../lib64/libcc1.la
libtool: finish: PATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/sbin" ldconfig -n /work/install-gcc/lib/../lib64
...
/work/install-gcc/lib/../lib64
...
./share/info/dir
...
./lib64/libcc1.so.0.0.0
...
./lib64/libcc1.so.0
...
./lib64/libcc1.so
./lib64/libcc1.la
./lib/
...
./lib/gcc/x86_64-pc-linux-gnu/7.2.0/include-fixed/root/
...
./lib/gcc/x86_64-pc-linux-gnu/7.2.0/include-fixed/DONE
./lib/gcc/x86_64-pc-linux-gnu/7.2.0/crtprec80.o
...
```

Might the error be related to the location of libcc?

andreygursky, 2017-12-16:
> I tried buster, but I get the same result as with stretch.

Before configuring gcc, make sure how the system gcc is configured: ```gcc -v 2>&1 | grep --color enable-default-pie```.
If ```--enable-default-pie``` is present (which is the case since Stretch), add it to the ghdl's gcc configure either.

1138-4EB, 2017-12-16:
Thanks @andreygursky , it works! https://travis-ci.org/1138-4EB/ghdl/builds/317223690

talonmyburgh, 2020-02-18:
Hi guys, new to this tool but trying to use it since I need it to run VUnit simulations (since XSim is not supported). I'm using PopOS! 18.04 LTs on a Dell Inspiron machine. I cloned the ghdl master branch. The output of running `git rev-parse origin/master` is:

`939221749e7f1228ed959d1007a3fd3e3b4b9146`

I think I'm getting a simular error but I don't see a solution to it in the comments made in this thread. I installed gnat from [adacore](http://libre.adacore.com/) and try doing the example mentioned here: [](https://ghdl.readthedocs.io/en/latest/getting/mcode.html).

When I type `make` I get the following:

`gcc -c -g -DWITH_GNAT_RUN_TIME -o jumps.o /home/talon/Desktop/ghdl/build/../src/grt/config/jumps.c
gcc: error trying to exec 'cc1': execvp: No such file or directory
../src/grt/Makefile.inc:134: recipe for target 'jumps.o' failed
make: *** [jumps.o] Error 1`

I've read that this is something to do with being unable to find gcc-c++ or something so I have run the following command:

`sudo apt-get install --reinstall build-essential`

to fix this but it's done nothing.
Thanks.

eine, 2020-02-18:
@talonmyburgh, is there any special reason for you to use GCC backend instead of mcode or LLVM? See https://ghdl.readthedocs.io/en/latest/getting/

talonmyburgh, 2020-02-18:
No, I thought I was using an mcode backend. At least that's what it says at the top of the document I am trying to follow: [https://ghdl.readthedocs.io/en/latest/getting/mcode.html](https://ghdl.readthedocs.io/en/latest/getting/mcode.html).

As a follow up, I've just installed Ada and I'm not getting it to compile a test adb file. If I run `gprbuild greet.adb` I get the following error:

`using project file /opt/gps/share/gpr/_default.gpr
Compile
   [Ada]          greet.adb
gcc: error trying to exec 'gnat1': execvp: No such file or directory
gprbuild: *** compilation phase failed
`
So this might point out why I'm getting an error?

eine, 2020-02-18:
Then, please, open a new issue to track your case, since it is unrelated to this issue. Meanwhile, you might want to ensure that GCC works on your setup by compiling any simple C program. Next, did you check if package `gnat` is available in the package manager of PopOS?

talonmyburgh, 2020-02-18:
Okay I will open a new issue if I cannot rectify this.
I think what is wrong is that the gnat gcc version is outdated? When I run `which gcc` I get the following:
`/opt/gps/bin/gcc`
where gps was something that was installed when I installed the community ada from [adacore](http://libre.adacore.com/).

Trying to compile a simple c file doesn't work either and I get the same error `gcc hi.c`:
`gcc: error trying to exec 'cc1': execvp: Not a directory`

Typing `gcc -v` gives:

`Using built-in specs.
COLLECT_GCC=gcc
OFFLOAD_TARGET_NAMES=nvptx-none
Target: x86_64-pc-linux-gnu
Configured with: ../src/configure --enable-languages=ada,c,c++ --enable-dual-exceptions --enable-offload-targets=nvptx-none --enable-_cxa_atexit --enable-threads=posix --with-bugurl=URL:mailto:report@adacore.com --disable-nls --without-libiconv-prefix --disable-libstdcxx-pch --disable-libada --enable-checking=release --disable-multilib --with-mpfr=/it/sbx/a2c2/x86_64-linux/mpfr_stable-c/install --with-gmp=/it/sbx/a2c2/x86_64-linux/gmp_stable-c/install --with-mpc=/it/sbx/a2c2/x86_64-linux/mpc_stable-c/install --with-build-time-tools=/it/sbx/a2c2/x86_64-linux/gcc-c/build/buildtools/bin --prefix=/it/sbx/a2c2/x86_64-linux/gcc-c/pkg --build=x86_64-pc-linux-gnu
Thread model: posix
gcc version 8.3.1 20190518 (for GNAT Community 2019 20190517) (GCC) `

Thank you.

eine, 2020-02-18:
Please, do move this last comment to a new issue.

talonmyburgh, 2020-02-18:
Done, see #1141 
Thanks @eine

eheperson, 2022-05-09:
> Wo! Thank you @1138-4EB ! That did the trick. I'm actually not sure what solved it. I added the enable-shared flag and the CFLAG=-fPIC (not sure what these two do) I also specified the locations of all the dependencies, including zlib. I'm not administrator of this system, so not all libraries have the default paths. Maybe it was messing with an old zlib version. The command that did it is the following:
> 
> ```
> 
> ../gcc-source/configure --prefix=/home/user/ghdl --enable-languages=c,vhdl --disable-bootstrap \
> --disable-lto --disable-multilib --disable-libssp --disable-libgomp --disable-libquadmath \
> --enable-shared CFLAGS=-fPIC --with-isl=/home/user/isl-0.15 \ 
> --with-gmp=/home/user/gmp-6.1.2/ --with-mpc=/home/user/mpc-1.0.3/  \
> --with-mpfr=/home/user/mpfr-3.1.6/ --with-gcc=/home/user/gcc-7.2.0 \ 
> --with-system-zlib=/home/user/zlib-1.2.11
> ```

Do i need installed gcc in my system to build gcc :D 👯

umarcor, 2022-05-09:
@eheperson you need a compiler which can build gcc sources. That is typically gcc, but I believe you can use other. See https://stackoverflow.com/questions/9429491/how-are-gcc-and-g-bootstrapped