ghdl/ghdl #480: [build-system] do not require git in the Makefile

1138-4EB, 2017-12-08:
Right now, if the repo is cloned through git, it becomes a 'soft' build dependecency because https://github.com/tgingold/ghdl/blob/master/Makefile.in#L143-L150

`make` finds a folder named `.git` and it assumes that git exists. If it does not, no error is raised, but var `$desc` is empty. This produces the following result when `GHDL --version` is executed:

```
GHDL 0.35-dev () [Dunoon edition]
```

Steps to reproduce:

- (git) Clone tgindold/ghdl.
- Copy the folder to a machine without git installed (say mount the folder in a docker container).
- Build GHDL.

I realized about this in #477, as you can see in this animated gif: https://user-images.githubusercontent.com/6628437/33694969-2e7b7030-dafb-11e7-9eba-fb3abae1a161.gif

The workaround I implemented is:

``` bash
version.tmp: $(srcdir)/src/version.in force
#	Create version.tmp from version.in, using git date/hash
	if [ "`command -v git`" != "" ]; then \
		desc=`cd $(srcdir); git describe --dirty`; \
		$(SED) -e "s/[(].*[)]/($$desc)/" < $< > $@; \
	else \
		if [ "${GIT_DESCRIBE_DIRTY}" != "" ]; then \
			$(SED) -e "s/[(].*[)]/($$GIT_DESCRIBE_DIRTY)/" < $< > $@; \
		else \
			$(CP) $< $@; \
		fi \
	fi
```

Along with changing https://github.com/tgingold/ghdl/blob/master/dist/linux/travis-ci.sh#L78 to:

```
git fetch --unshallow
docker run --rm --tty --volume $(pwd):/work -w "/work" $DOCKER_NAME bash -c "export GIT_DESCRIBE_DIRTY=$(git describe --dirty --always); $BUILD_CMD"
```

Note the need of `git fetch --unshallow`: travis-ci/travis-ci#3412

---

However, I just solved one problem to introduce a different one. Now, any repo which does not contain a `.git` folder will fail if git command is found in the machine. I though about:

- Making the envvar take precedence over checking the command. This would make it necessary to define the envvar when no `.git` folder is present. At the same time, it would provide a mechanism to overwrite the value even when git and .git are found.
- Add two conditions to the first if: `$(command -v git) != "" && -d "$(srcdir)/.git"`. This will make the envvar a fallback.

Yet, because I am not confident at all. I prefer to discuss it first.

1138-4EB, 2017-12-08:
Note that currently travis builds are affected by this issue: https://travis-ci.org/tgingold/ghdl/builds/313308854

BTW, how did you build/deploy [ghdl-v0.34-mcode-ubuntu.tgz](https://github.com/tgingold/ghdl/releases/download/v0.34/ghdl-v0.34-mcode-ubuntu.tgz) in the [latest release](https://github.com/tgingold/ghdl/releases/tag/v0.34)? I can only make it run on `ubuntu:16.04`, which AFAIK has never been done in travis.

Note about the releases page: all the shields are broken (the link points to image itself). Might be related to github/markup#224 codecov/support#218

tgingold, 2017-12-08:
Thank you for the explanation.

But why not simply installing git in docker ?
Or patching the version.in file before running docker ?

Tristan.

1138-4EB, 2017-12-09:
As always, you are so fast!

> But why not simply installing git in docker ?

This is half conceptual/global trend, and half technical:

- Docker containers are expected to be mostly atomic. That is, they should be feature/tool specific and include only the minimum requirements to make it work. The furthest extreme are RancherOS, CoreOS.
- Git is by no means a requirement to either build or run GHDL. Indeed, it's usage in build and test jobs is symbolic: get a text string to uniquely identify the sources that produced the binary.
  - Right now, if anyone asks *what are the minimum dependencies to build GHDL in ubuntu 16.04?*, I can reply [dist/linux/docker/build/ubuntu16#L7](https://github.com/1138-4EB/ghdl/blob/builders/dist/linux/docker/build/ubuntu16#L7). The same holds for "runtime" dependencies; these are the requirements to run the last release tarball on ubuntu ([dist/linux/docker/run/ubuntu16#L7](https://github.com/1138-4EB/ghdl/blob/builders/dist/linux/docker/run/ubuntu16#L7)). I feel that adding git to any of these should require a companion note explaining when git is required and when it is not.
- Git is a heavy dependency, not just a few mb. Therefore, it would require signifficantly more bandwith when images are pushed/pulled. Even with the current implementation, it would be download before each job.
- Since the sources that produce the docker images, i.e. the Dockerfiles in `dist/linux/docker/`, are hosted in a git repo, it is expected that most of the times (if not always), git will be available in the machine before triggering the build. Downloading inside the container is duplicating resource usage. I think we don't have a dependency on a specific git version to properly justify this.
  - wget/curl can be also used, but we can expect anyone doing so not to bother about the version identifier.

> Or patching the version.in file before running docker ?

This's great. I didn't do it because I feel lost in the makefile. Sure, I can just edit `travis-ci.sh`:

```
git fetch --unshallow
desc=`git describe --dirty`;
sed -e "s/[(].*[)]/($$desc)/" src/version.in;
docker run --rm --tty --volume $(pwd):/work -w "/work" $DOCKER_NAME bash -c "$BUILD_CMD"
```

but it would force any user to execute those commands to build GHDL locally and have the version properly patched. I think it is cleaner now, having it written in the makefile. Is it possible to execute something like `make --patch-version` which would just do that? Note that it should be ignored when `make` is executed later.

Paebbels, 2017-12-09:
Why don't you detect is git is installed?

```Bash
which git > /dev/null
$?
```

You can run different code if GHDL is in a Git environment:
* Git is installed
* .git directory is present
-> patch version file

otherwise use a fixed version number (as already encoded in the version file).

1138-4EB, 2017-12-09:
> Why don't you detect is git is installed?

I am doing that. But using `[ "$(command -v git)" != "" ]`, instead. See [#481](https://github.com/tgingold/ghdl/pull/481/files)

> You can run different code if GHDL is in a Git environment

I don't really want to run different code. Indeed I'd like to ensure that exactly the same code is used. The problem is not that Git is not present, but that git and make are available in one environment (host machine) and build dependencies (gcc, gnat, etc.) are available in a different environment (container, virtual machine). Actually, it is almost working by executing `make -f Makefile.in version.tmp` on the host, just before running the container. See changes to `travis-ci.sh` in #481.

The advantages of the proposed patch are:

- The same code is executed.
- There is no change to be done to build GHDL locally.
- An envvar can be defined as a fallback when git or .git are not found.
- If neither git nor the envvar are available, it is left empty, as happens now.

The issue now is that running `make -f Makefile.in version.tmp` tries to execute this `include $(GRTSRCDIR)/Makefile.inc` (which it shouldn't), and it fails (ee https://travis-ci.org/1138-4EB/ghdl/jobs/313993366). Do you know how to have that single line excluded when `make version.tmp` is executed?

tgingold, 2017-12-10:
Now fixed.

closed 2017-12-10 by tgingold

1138-4EB, 2017-12-10:
Thanks! A couple of details:

Is there any reason not to include the condition `"$(command -v git) != ""`, along with checking `.git`?

This command alone, which is a reasonable command for someone willing to build GHDL locally inside a container, will still produce `()` without showing any error:

```
docker run --rm --tty --volume $(pwd):/work -w "/work" $DOCKER_NAME bash -c "./dist/linux/buildtest.sh -b mcode"
```

Also, `unshallow` will fail if already unshalowed. There should be no more than a single execution, but better check:

```
if [ -f $(git rev-parse --git-dir)/shallow ]; then git fetch --unshallow; fi
```

tgingold, 2017-12-10:
Isn't it over-protection as travis always clone with --depth ?

1138-4EB, 2017-12-10:
> [@tgingold](https://github.com/tgingold/ghdl/pull/481#issuecomment-350556608)
> Are you sure about that ? If git is not available (or fail to describe), then nothing is changed in the version file.

You are right. It was my fault when merging. It all the changes but the 'test'. No issue then.

> Isn't it over-protection as travis always clone with --depth ?

- I am using the same `travis-ci.sh` for 'buildtest' and 'deploy to dockerhub'.
- The first time I execute it, only buildtest is run. buildtest is the same as the master, so it does `git --unshallow` as per the last fix.
- The second time, only deploy is run.
- If I don't set it right, the second time buildtest is executed again just before deploy. That is, I buildtest two times and deploy once. This is not logical, but it should not break the build.