ghdl/ghdl #483: Releases not showing distance to master

1138-4EB, 2017-12-09:
In the releases tab, some of them show '...commits to master since this release', but others don't. It is missing in 'v0.34' and  'v0.33', but shown in the others.

Paebbels, 2017-12-09:
This might be related to "real" releases vs. "pre-releases".

1138-4EB, 2017-12-09:
I thought that at first. But see e.g. https://github.com/gohugoio/hugo/releases

Paebbels, 2017-12-09:
Hmmm....
Maybe it's relate to releases created and uploaded from Travis-CI?

1138-4EB, 2017-12-09:
Where both v0.33 and v0.34 created from Travis-CI?

It might be, because travis works on a detached head. That's why I had to use `git fetch --unshallow` [here](https://travis-ci.org/1138-4EB/ghdl/jobs/313993366). Note at line 688, that all the tags are shown, except v0.34 and v0.33. It is my fork, so not all of the tags here exists there.

Can you try adding `git fetch --unshallow` in a branch and pushing it to see what is the result on tgingold/ghdl?

tgingold, 2017-12-10:
Yes, both 0.33 and 0.34 were crated by travis-ci.
The branches were created on by computer which is already unshallowed.
Maybe they are automatically shallowed in github.

1138-4EB, 2017-12-10:
Just a note. After merging the new documentation, the 'latest tag' badge/shield is affected by this issue, both in the home at RTD and in the README here.

1138-4EB, 2017-12-10:
In the end, I think it has nothing to do with travis, but with the fact that tags v0.33 and v0.34 are not antecessors of the master branch. This is because a separate branch was created for each of them, changes where added there, but they were not merged back to master. Indeed, all previous tags (those with format YYYY-MM-DD) are antecessors of master.

tgingold, 2017-12-12:
Not a big deal!

closed 2017-12-12 by tgingold

1138-4EB, 2017-12-12:
> Not a big deal!

Does this mean 'easy to fix' or 'no need to fix'? In the latter, shall we remove the 'latest tag' shields?

tgingold, 2017-12-12:
> On 12 Dec 2017, at 05:40, 1138-4EB <notifications@github.com> wrote:
> 
> Not a big deal!
> 
> Does this mean 'easy to fix' or 'no need to fix'? In the latter, shall we remove the 'latest tag' shields?

No need to fix.

I will remove the 'latest tag` shield in the readme.

1138-4EB, 2017-12-12:
Then, also `|SHIELD:tag|` from https://github.com/tgingold/ghdl/blob/master/doc/index.rst

1138-4EB, 2017-12-14:
Just to note that this is fixed now. Anyway, I should not add the 'latest tag' shields again. It is ok and cleaner now with only the 'commits since' shield pointing to the latest release.

tgingold, 2017-12-14:
Yes, because the release was made with a tag on master (result of your investigation!)