ghdl/ghdl #484: Documentation extraction from sources

1138-4EB, 2017-12-10:
In order to avoid missmatches between the cli (ghdl --help) and the docs at RTD, I think we should try to move the description to the sources, next to where the flags are defined. Indeed, the sources of GHDL have a lot of comments already. If a comment with a given decoration is added for each flag (even if it is empty), it is easier to track missing content in the docs.

I did not find any conclusive reference about using Doxygen with Ada. However, I found [gnatdoc](http://docs.adacore.com/gnatdoc-docs/users_guide/_build/html/gnatdoc.html), which is part of [GNAT Programming Studio](https://www.adacore.com/gnatpro/toolsuite/gps) ([AdaCore/gps](https://github.com/AdaCore/gps)). Sadly, it throws an error when trying to execute it on GHDL:

```
docker run --rm -t ghdl/build:stretch-mcode sh -c "\
  apt-get update \
&& apt-get install -y gnat-gps git \
&& git clone http://github.com/tgingold/ghdl \
&& cd ghdl \
&& ./dist/linux/buildtest.sh -b mcode \
&& cd build-mcode \
&& gnatdoc -Pghdl"
```

[image]

@tgingold, any thoughs?

Paebbels, 2017-12-10:
I would prefer a Sphinx based approach so it integrates into ReadTheDocs.

tgingold, 2017-12-10:
@Paebbels has a very good point.

I am not a big fan of systems like doxygen. IMHO, doc and comments have different purposes.

1138-4EB, 2017-12-10:
@Paebbels what is the Sphinx based alternative? AFAIK, it is a tool meant for Python, so it cannot analyse the codebase of GHDL to get a list of modules, variables, definitions... I think it is useful as (one among many), static site generators (Ruby has Jekyll, Golang has Hugo, and Python has Sphinx). What advantages do you see in using Sphinx over any of the others? Apart from having multiple versions built, which (is a RTD feature I think, and) we already now how to replicate.

On the disadvantages I see weight (hugo is a single 6mb static binary), performance (hugo builds a website in less than 500ms), and difficulty to develop and test locally (hugo has an embedded webserver with livereload in the same 6mb binary).

Anyway, note that Sphinx, Hugo, Jekyll, Pandoc, etc. are all meant to create static web sites and/or PDF documents (among others) from a bunch of md, rst, mmark... files. This issue is not about it, but about how to automate the generation of those md, rst, mmark... sources from GHDL Ada sources. As commented [here](https://github.com/tgingold/ghdl/issues/111#issuecomment-350512465), Doxygen allows to export data in XML format. Then, it is up to us what to do with it. We can generate a website directly, which is the usual use for doxygen. But we can also use it to partially generate the content in `./doc`. Sphinx can still be used to generate the website and PDFs.

@tgingold, Doxygen identifies two types of comments: those that should be integrated in the documentation and those that should only be shown in code listings. There is an explicit differentiation between comments 'to be shown' and 'internal' comments. E.g.:

`doc/using/Simulation.rst#L70-L73` contains

```
.. option:: --vcd-nodate

  Do not write date in VCD file
```

at the same time, `./src/grt/grt-vcd.adb` contains:

```
Put_Line (" --vcd-nodate       do not write date in VCD file");
```

That is the exact same info. Furthermore, these are also written in `grt-vcd.adb`:

```
   --  If TRUE, put $date in vcd file.
   --  Can be set to FALSE to make vcd comparaison easier.
   Flag_Vcd_Date : Boolean := True;
```

```
      if Opt'Length = 12 and then Opt (F + 5 .. F + 11) = "-nodate" then
         Flag_Vcd_Date := False;
         return True;
      end if;
```

My proposal is to remove the block from `Simulation.rst` and add it to `grt-vcd.adb`:

```
      if Opt'Length = 12 and then Opt (F + 5 .. F + 11) = "-nodate" then
        -- @Simulation: `--vcd-nodate` Do not write date in VCD file
         Flag_Vcd_Date := False;
         return True;
      end if;
```

or

```
   -- @Simulation: `--vcd-nodate` Do not write date in VCD file. Can be set to FALSE to make vcd comparaison easier.
   --  If TRUE, put $date in vcd file.
   Flag_Vcd_Date : Boolean := True;
```

This last example is meaningful because it let's us see that there is useful info in the sources which is not shown in the docs. Note that *Can be set to FALSE to make vcd comparaison easier* 'will' be shown in the docs, while `If TRUE, put $date in vcd file` is only found in the sources.

Of course I completely made the syntax up. The possibilities depend on the preprocessor/parser, say Doxygen or gnatdoc. I opened the issue to see what does gnatdoc offer, because using Doxygen for Ada is not straightforward.

---

Nevertheless, I think that using Doxygen/gnatdoc is useful itself for contributors, developers, people willing to learn how GHDL works inside. It is overkill just to document cli options, I agree. But it provides much more: a browsable description of modules, variables, definitons, etc. with syntax highlighting.

Thinking loud: is something as `ghdl --pp-html <ghdl>` possible? I.e. use a built ghdl to run `--pp-html` on the Ada codebase of ghdl.

tgingold, 2017-12-10:
I understand what you mean, but:
* doc extraction doesn't deal well with ordering (in our case, we will have the options in any order)
* there is no universal tool/convention.

I think there are more cons than pros.

1138-4EB, 2017-12-10:
> doc extraction doesn't deal well with ordering (in our case, we will have the options in any order)

I believe this can be handled either with additional decorations in the extracted comments or in the XML to md/rst conversion. Indeed, I find it interesting to be able to decide how to organize the commands in the visualization level.

> there is no universal tool/convention

Yo mean for Ada or in general?

tgingold, 2017-12-10:
Both. There are multiple conventions and therefore you need additional tool knowledge.

closed 2017-12-15 by tgingold

Paebbels, 2017-12-16:
Doxygen & Co. are mapping everything to classes and methods. This methodology is not compatible to Ada and VHDL.

1138-4EB, 2017-12-16:
Seems that's why there is no ADA support in Doxygen. However, gnatdoc, which is part of AdaCore/gps works quite well with GHDL, considering that:

- I had to reemove `iirs_utils.ads` due to the error in the first comment.
- Tristan does any follow any rigid guidelines, so some info is shown coherently and some info is not.

eine, 2020-05-10:
Dockerfile based on Buster:

```dockerfile
FROM ghdl/ghdl:buster-mcode

RUN apt-get update \
  && apt-get install -y gnat-gps git libgtk-3-0 gir1.2-gtk-3.0 iceweasel
```