ghdl/ghdl #486: Shields broken on GitHub releases

1138-4EB, 2017-12-10:
Shields in GitHub releases are shown as links, but they point to the image, instead of triggering the download of the referred asset or showing additional info about the stats.

Paebbels, 2017-12-10:
Can you post a screenshot? I don't see a problem.

1138-4EB, 2017-12-10:
It is not the screenshot. Click on any shields.

Paebbels, 2017-12-10:
I'm sure I tested it ... Older releases are broken too.

Btw. how do you create these fancy screen recordings as gif animations?

1138-4EB, 2017-12-10:
Axaxax! I knew you'd like it. It is a pretty stupid to use but yet powerful app: http://www.screentogif.com/
Sadly, I think it is only available on windows.

1138-4EB, 2017-12-12:
The equivalent shields in the documentation are outdated but properly linked: http://ghdl.readthedocs.io/en/latest/getting/Releases.html

@Paebbels , maybe those are the ones you remember having tested?

I think the fix in markdown is to replace `![<alt](<url_svg>)` with `[[image]](<url_tg>)`, but I cannot see the source of the tables in the releases tab. Can you post the markdown of the three tables and the three single shields here?

1138-4EB, 2017-12-16:
These are the updated tables for v0.34 and v0.35:

```
| OS                    | Backend  | Filesize  | Downloads                                                                                                     |
| --------------------- | -------- | --------- | ------------------------------------------------------------------------------------------------------------- |
| Windows x86  (1)      | mcode    |  6.2 MiB  | [![](https://img.shields.io/github/downloads/tgingold/ghdl/v0.34/ghdl-v0.34-gnatgpl32-mcode.installer.ps1.svg)](https://github.com/tgingold/ghdl/releases/download/v0.34/ghdl-v0.34-gnatgpl32-mcode.installer.ps1) |
| Windows x86           | mcode    |  2.3 MiB  | [![](https://img.shields.io/github/downloads/tgingold/ghdl/v0.34/ghdl-v0.34-gnatgpl32-mcode.zip.svg)](https://github.com/tgingold/ghdl/releases/download/v0.34/ghdl-v0.34-gnatgpl32-mcode.zip)           |
| Fedora                | LLVM     |  6.7 MiB  | [![](https://img.shields.io/github/downloads/tgingold/ghdl/v0.34/ghdl-v0.34-llvm-fedora.tgz.svg)](https://github.com/tgingold/ghdl/releases/download/v0.34/ghdl-v0.34-llvm-fedora.tgz)               |
| Documentation         |          |  0.3 MiB  | [![](https://img.shields.io/github/downloads/tgingold/ghdl/v0.34/ghdl-v0.34-manual.pdf.svg)](https://github.com/tgingold/ghdl/releases/download/v0.34/ghdl-v0.34-manual.pdf)                    |
| Ubuntu                | mcode    |  2.8 MiB  | [![](https://img.shields.io/github/downloads/tgingold/ghdl/v0.34/ghdl-v0.34-mcode-ubuntu.tgz.svg)](https://github.com/tgingold/ghdl/releases/download/v0.34/ghdl-v0.34-mcode-ubuntu.tgz)              |
| Windows x86 (MinGW32) | mcode    |  3.0 MiB  | [![](https://img.shields.io/github/downloads/tgingold/ghdl/v0.34/ghdl-v0.34-mingw32-mcode.zip.svg)](https://github.com/tgingold/ghdl/releases/download/v0.34/ghdl-v0.34-mingw32-mcode.zip)             |
| Windows x86 (MinGW64) | LLVM     | 12.6 MiB  | [![](https://img.shields.io/github/downloads/tgingold/ghdl/v0.34/ghdl-v0.34-mingw64-llvm.zip.svg)](https://github.com/tgingold/ghdl/releases/download/v0.34/ghdl-v0.34-mingw64-llvm.zip)              |
| Mac OS X              | LLVM     | 12.15 MiB | [![](https://img.shields.io/github/downloads/tgingold/ghdl/v0.34/ghdl-v0.34-llvm-darwin13.pkg.svg)](https://github.com/tgingold/ghdl/releases/download/v0.34/ghdl-v0.34-llvm-darwin13.pkg)             |
| Sum:                  |          |           | [![Downloads of GHDL v0.34](https://img.shields.io/github/downloads/tgingold/ghdl/v0.34/total.svg)]()             |
```

```
| OS | Backend | Filesize | Downloads |
| --- | --- | --- | --- |
| Debian 9 (Stretch) GPL | mcode | 2.61 MB | [![](https://img.shields.io/github/downloads/tgingold/ghdl/v0.35/ghdl-gpl-0.35-mcode.tgz.svg)](https://github.com/tgingold/ghdl/releases/download/v0.35/ghdl-gpl-0.35-mcode.tgz) |
| Debian 9 (Stretch) | mcode | 2.93 MB | [![](https://img.shields.io/github/downloads/tgingold/ghdl/v0.35/ghdl-0.35-mcode-stretch.tgz.svg)](https://github.com/tgingold/ghdl/releases/download/v0.35/ghdl-0.35-mcode-stretch.tgz) |
| Ubuntu 14 (Trusty)  | mcode | 3.15 MB | [![](https://img.shields.io/github/downloads/tgingold/ghdl/v0.35/ghdl-0.35-mcode-ubuntu14.tgz.svg)](https://github.com/tgingold/ghdl/releases/download/v0.35/ghdl-0.35-mcode-ubuntu14.tgz) |
| Ubuntu 14 (Trusty) | llvm | 14.11MB | [![](https://img.shields.io/github/downloads/tgingold/ghdl/v0.35/ghdl-0.35-llvm-3.8-ubuntu14.tgz.svg)](https://github.com/tgingold/ghdl/releases/download/v0.35/ghdl-0.35-llvm-3.8-ubuntu14.tgz) |
| Fedora 26 | mcode | 2.83 MB | [![](https://img.shields.io/github/downloads/tgingold/ghdl/v0.35/ghdl-0.35-mcode-fedora26.tgz.svg)](https://github.com/tgingold/ghdl/releases/download/v0.35/ghdl-0.35-mcode-fedora26.tgz) |
| Windows x86 (MinGW32) | mcode | 3.05 MB | [![](https://img.shields.io/github/downloads/tgingold/ghdl/v0.35/ghdl-v0.35-mingw32-mcode.zip.svg)](https://github.com/tgingold/ghdl/releases/download/v0.35/ghdl-v0.35-mingw32-mcode.zip) |
| Windows x86 (MinGW64) | LLVM | 12.7 MB | [![](https://img.shields.io/github/downloads/tgingold/ghdl/v0.35/ghdl-v0.35-mingw64-llvm.zip.svg)](https://github.com/tgingold/ghdl/releases/download/v0.35/ghdl-v0.35-mingw64-llvm.zip) |
| Mac OS X | mcode | 2.26 MB | [![](https://img.shields.io/github/downloads/tgingold/ghdl/v0.35/ghdl-0.35-mcode-macosx.tgz.svg)](https://github.com/tgingold/ghdl/releases/download/v0.35/ghdl-0.35-mcode-macosx.tgz) |
| Sum: | | | [![Downloads of GHDL v0.35](https://img.shields.io/github/downloads/tgingold/ghdl/v0.35/total.svg)]() |
```

Paebbels, 2017-12-17:
I updated the releases texts for v0.34 and v0.35.

closed 2017-12-17 by Paebbels