ghdl/ghdl #490: doc inconsistency (gcc build)

Bamban, 2017-12-14:
Hi,
The documentation on Readthedocs is inconsistent with the procedure to build with gcc backend. See this page:
http://ghdl.readthedocs.io/en/latest/building/gcc/GNULinux-GNAT.html
It is the same content as http://ghdl.readthedocs.io/en/latest/building/mcode/GNULinux-GNAT.html
(I followed the "old" method with tweaks from the "new", such as make vhdllibs, to build with gcc with success on redhat7.4 and gcc 7.2)
One small remark: I found it more readable not to have so much different pages on the doc. Clicking "next" and "previous" is tedious plus I can't use "find". Cutting it in so much little pieces was wanted or is it the default behavior of RTD ?

1138-4EB, 2017-12-14:
I realized an hour ago. I am [working](https://github.com/1138-4EB/ghdl/tree/doc-gcc) on it. You can check the old [BUILD.txt](https://github.com/tgingold/ghdl/blob/ghdl-0.34/BUILD.txt) meanwhile.

1138-4EB, 2017-12-14:
@Bamban, I updated the docs to reflect the content of the removed BUILD.txt. See #491.

> (I followed the "old" method with tweaks from the "new", such as make vhdllibs, to build with gcc with success on redhat7.4 and gcc 7.2)

What do you mean with *'old method with tweaks from the new'*?

> I found it more readable not to have so much different pages on the doc. Clicking "next" and "previous" is tedious plus I can't use "find"

It was so confusing partially because the left menu was broken if you used certain links. It should be fixed [here](http://ghdl-devfork.readthedocs.io/en/doc-gcc/building/gcc/index.html).

> Cutting it in so much little pieces was wanted or is it the default behavior of RTD ?

It was done like that because GCC, LLVM, mcode, GNU/Linux, macOs, Windows GNAT, Windows MinGW is a 12 element matrix. Anyway, with the fixed menus, you should go directly to [building](http://ghdl-devfork.readthedocs.io/en/doc-gcc/building/Building.html) from the README here. Then select your backend, either in the table or in the sidebar. See general comments and, last, take the platform from the list (or expand the sidebar). Do still find it uncomfortable?

Bamban, 2017-12-14:
Nevermind, in fact I followed the recommended method: I took the BUILD.txt from 87a5d4a in the root directory. It appears that it is up to date.

> Do you still find it uncomfortable?
It's only my point of view but one page for building would have made the trick. Anyway, you build it that way and probably some people would find it clearer like that, so that's ok !

Thanks for the quick update btw.

closed 2017-12-14 by Bamban