ghdl/ghdl #492: compile_intel.sh not up to date with version 17

Bamban, 2017-12-14:
Hi,
There is some bugs in the compile_intel.sh vendor compilation script. It bases itself on the altera part of config.sh and not intel (see config.sh and beginning of compile_intel.sh)
Another thing more important: some simulation libraries of 17.0 intelFPGA toolsuite are not compiled and some are absent from the tool. I made a small patch to add two libraries and remove the ones that was missing. It is probably not 100% finished (some file are present in the source directory but never compiled, such as ct1_hip_atoms.vhd for instance, because of compilations errors).

[0001-vendor-library-compilation-script-update-intel.patch.txt](https://github.com/tgingold/ghdl/files/1559308/0001-vendor-library-compilation-script-update-intel.patch.txt)

Paebbels, 2017-12-14:
This patch would break support for older Intel Quartus Prime versions.

We would need to implement a version check and create different file sets for each Quartus version.

Bamban, 2017-12-15:
You are right, it was too straightforward and not generic enough.

Another thing: the directory structure created is not parsable with ghdl -P option according to the doc ([link](http://ghdl.readthedocs.io/en/latest/using/InvokingGHDL.html?highlight=-P#cmdoption-ghdl-p-directory)). All file are created under intel/vXX/files.o.

Paebbels, 2017-12-15:
GHDL can parse this structure unless I made a big mistake.

Bamban, 2017-12-15:
I directly pasted the `intel` directory into my simulation directory. Note that I have the folders `intel/twentynm/v93/twentynm-obj93.cf`. Then I tried the following:
```
ghdl -a --work=work --workdir=work -g --ieee=synopsys -Pintel/twentynm -Pintel/altera_lnsim ./dsp.vhd ./dsp_bench.vhd
./dsp.vhd:5:9:error: cannot find resource library "twentynm"
ghdl:error: compilation error
```
It also fails with the `--std=93` option.
Now, if I write `-Pintel/twentynm/v93` and `-Pintel/altera_lnsim/v93` instead, it works.
(in fact, it almost works, see #493 ;) I should say, the library is found )

Paebbels, 2017-12-15:
Just use:

```bash
ghdl -a ... -Pintel
```

GHDL will automatically pick the library and appropriate version based on `--std=XX`.

------------------------
Btw. you should NEVER use the name `work` as the name of a VHDL library. This is a fault done by other vendors like Xilinx (ISE), Aldec or Mentor! work is a special name in VHDL and points to the current working library; not to a library called `work`.

Bamban, 2017-12-18:
I tried this elegant solution. Unfortunately, it does not work :/
Just for the completeness, here is how I include the libraries in `dsp.vhd`:
```
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

library intel;                                                                                                                                                                                                                                                                
use intel.twentynm.all;
```
Here is the ghdl call and the answer:
```
ghdl -a -Pintel ./dsp.vhd
./dsp.vhd:5:9:error: cannot find resource library "intel"
./dsp.vhd:6:11:error: unit "twentynm" not found in library "intel"
ghdl:error: compilation error
```
Here is my directory tree:
```
./
|- Makefile
|- dsp.vhd
|- dsp_bench.vhd
+- intel
   |- altera/
   |- altera_lnsim/
   |- altera_mf/
   |- cyclone10gx/                
   |- fourteennm/
   |- lpm/
   |- sgate/
   |- stratixv/
   |- stratixv_pcie_hip/          
   +- twentynm/                   
      |- v08/
      +- v93/
         |- twentynm-obj93.cf      
         |- twentynm_atoms.o       
         |- twentynm_components.o  
         |- twentynm_hip_atoms.o   
         |- twentynm_hip_components.o
         |- twentynm_hssi_atoms.o  
         +- twentynm_hssi_components.o 
```
I can only add the output of this command:
```
ghdl --disp-config -Pintel
Paths at configuration:
compiler command: libexec/gcc/x86_64-pc-linux-gnu/7.2.0/ghdl1
assembler command: as
linker command: gcc
default lib prefix: lib/ghdl

command line prefix (--PREFIX): (not set)
environment prefix (GHDL_PREFIX): (not set)
exec prefix (from program name): /home/me/local

library prefix: /home/me/local/lib/ghdl
library directory: /home/me/local/lib/ghdl
compiler path: /home/me/local/libexec/gcc/x86_64-pc-linux-gnu/7.2.0/ghdl1
assembler path: /usr/bin/as
linker path: /usr/bin/gcc

default library paths:
 intel/
 /home/me/local/lib/ghdl/v93/std/
 /home/me/local/lib/ghdl/v93/ieee/
```
What's wrong ?

Paebbels, 2017-12-18:
Your VHDL code should be:

```vhdl
library twentynm;
use     twentynm.components.all;
```

`intel` is no library name. It's a directory holding all Intel (formerly Altera) libraries.

Bamban, 2017-12-19:
That did the trick with a small tweaking. Instead of what you gave, I had to write it like that:
```vhdl
library twentynm;
use twentynm.twentynm_components.all;
```
as it is the name of the package in the `twentynm-obj93.cf` file.

closed 2017-12-19 by Bamban

Paebbels, 2017-12-19:
Oh sorry, I assumed the file name matches the package name :(.