ghdl/ghdl #493: Component twentynm_fp_mac_encrypted is not bound

Bamban, 2017-12-15:
(posting a lot of issues lately, sorry guys!)

I have an issue with the intel 17.0 quartus libraries. I believe that there is a portion of the simulation models that are encrypted (DSP for instance). Yet, I would like to use a model of DSP (which support FP operations on the last intel families) for my simulation.
I ran the script compile-intel.sh located in ghdl/install/path/lib/ghdl/vendors. I modified it according to the patch I mentioned in #492 in order to add twentynm library.
When I try to run ghdl on a simple source code (instantiation of a DSP) it gives me the following output:

```bash
ghdl -e --work=work --workdir=work -g --ieee=synopsys -Ptwentynm -Paltera_lnsim dsp_bench
/opt/intelFPGA_pro/17.0/quartus/eda/sim_lib/twentynm_atoms.vhd:6265:1:warning: component instance "inst" of 'twentynm_fp_mac_encrypted' is not bound [-Wbinding]
/opt/intelFPGA_pro/17.0/quartus/eda/sim_lib/twentynm_atoms.vhd:6216:14:warning: (in default configuration of twentynm_fp_mac(behavior)) [-Wbinding]
```
I attached the instantiation code.
[dsp_bench.vhd.txt](https://github.com/tgingold/ghdl/files/1562669/dsp_bench.vhd.txt)
[dsp.vhd.txt](https://github.com/tgingold/ghdl/files/1562670/dsp.vhd.txt)
(btw, not being able to attach code directly to an issue is annoying)

1138-4EB, 2017-12-15:
> (btw, not being able to attach code directly to an issue is annoying)

You can just create a gist: https://gist.github.com/
They support versioning and comments. Then, you just paste the link here.

Paebbels, 2017-12-15:
It's not possible for GHDL to pre-compile encrypted primitives or IP cores as long as vendors do not provide an encryption key for GHDL or provide access for us to create binary versions of these models.

closed 2017-12-15 by Paebbels

Bamban, 2017-12-18:
That's too bad ... thanks vendors :(