ghdl/ghdl #496: release 0.35 issues

tgingold, 2017-12-17:
* On windows: the zip still have the 'v' from the tag
* On MacOS: ghdl -v displays 0.35-dirty

Paebbels, 2017-12-17:
I change the file name in the released version, but the travis script still needs fixing.

tgingold, 2017-12-17:
It is not the travis one but the appveyor script.

Paebbels, 2017-12-17:
I also noticed that all release files have the backend name before the platform, except for MinGW builds: `ghdl-0.35-mingw32-mcode.zip`, I also suggest to move those parts in the AppVeyor scripts.

tgingold, 2017-12-17:
I think having the platform name before the backend makes more sense.  So only mingw32 is right!

Paebbels, 2017-12-17:
I strongly agree, that platform name before backend name is better, but then you need to change almost all files and scripts...

Normally, when I suggest such cleanups/improvements, people reject it due to more work :).

tgingold, 2017-12-17:
I think this is doable.

andreygursky, 2017-12-17:
> I think having the platform name before the backend makes more sense. So only mingw32 is right!

This might be right, but please consider, that this might look uncommon, since all 3 popular binary package managers deb, rpm and pacman use schema like: "package name [+ favor] + version + architecture (platform) + pkg manager suffix".

1138-4EB, 2017-12-17:
For the GNU/Linux builds I believe that a single line needs to be changed: [travis-ci.sh#L78](https://github.com/tgingold/ghdl/blob/master/dist/linux/travis-ci.sh#L78). For GPL, [travis-ci.sh#L82](https://github.com/tgingold/ghdl/blob/master/dist/linux/travis-ci.sh#L82).

tgingold, 2017-12-18:
@andreygursky I would say that this is slightly different.
Currently when you look at the release page on github, it is hard to see what you can download on your platform.

andreygursky, 2017-12-27:
> Currently when you look at the release page on github, it is hard to see what you can download on your platform.

Yeah, indeed. So this workaround is needed due to missing GitHub feature to let custom separation of different release files.

1138-4EB, 2017-12-27:
@tgingold, @andreygursky, note that one of the motivations behind #506 is to provide a custom 'Download' page, where we can use JavaScript to selectively show/hide release tarballs providing filters such as: platform, backend, release, etc. My approach is to use [jq](https://stedolan.github.io/jq/) to query GitHub API in the last stage in travis. Then reconstruct a JSON object which contains the list of all available tarballs, binary releases and packages. The site would be served with that JSON.

Therefore, I think that it is better to pick the format which less problems will produce for packaging. UX in GitHub releases should not bother us, because it is not meant to be easy.

closed 2018-12-13 by tgingold