ghdl/ghdl #497: Package GHDL for MinGW32/MinGW64

Paebbels, 2017-12-17:
Now we have automatic builds for the MinGW32 (mcode) and MinGW64 (llvm) environments. It would be nice to provide a package for MinGWs package manager pacman.

Here is a description I found: https://github.com/msys2/msys2/wiki/Creating-Packages

If we can provide such a description, it would be the first distro, that could provide GHDL as an official package.

andreygursky, 2017-12-17:
Pacman based ArchLinux has ghdl package since 2.5 years in AUR:
https://aur.archlinux.org/packages/ghdl/

Paebbels, 2017-12-17:
@andreygursky Oh, nice. Who updates that package description to 0.35? Could this person also maintain the MinGW descriptions for GHDL?

andreygursky, 2017-12-17:
I just know about its existence, but sorry nothing more.

Paebbels, 2017-12-17:
I think this [PKGBUILD template](https://github.com/Alexpux/MINGW-packages/blob/master/mingw-w64-PKGBUILD-templates/PKGBUILD.CMake-github-git) could fit our needs. It supports packaging from GitHub.

tgingold, 2017-12-18:
I am not the right assignee for this issue: I don't use windows, so I don't want to maintain the mingw package.

1138-4EB, 2017-12-18:
Seems that the maintainer of the AUR package can be @vicencb , who already contributed a couple of commits to this repo. Might be worth hearing his opinion on how similar/different AUR and MinGW packages are.

Paebbels, 2017-12-18:
@tgingold By assigning you, I meant that we need someone to fork the MinGW packages repo :).

vicencb, 2017-12-18:
Hi,
neither I am a windows user. In fact, I didn't even know the existence of a MinGW repository of PKGBUILDs.
Sorry, but I can not help here.

1138-4EB, 2019-09-03:
FTR, two PKGBUILD files  (one for mcode and another one for llvm) are added in branch `actions` of my fork (see https://github.com/1138-4EB/ghdl/commits/actions). Moreover, AppVeyor (and subdir `dist/windows/appveyor`) are deprecated in favour of GitHub Actions. Putting both modifications together, in the workflow which is executed after each `push` or PR:

- In a MSYS shell, a `ghdl-mcode` package is built for mingw32 and a `ghdl-llvm` package is built for mingw64. Then, each of them is installed.
- In a corresponding shell (MINGW32 or MINGW64), the testsuite is executed.

See example run: https://github.com/1138-4EB/ghdl/runs/210613752

---

I proposed adding 'stable' versions of these PKGBUILD files downstream (see msys2/MINGW-packages#5757).

---

In [1138-4EB/ghdl-packaging](https://github.com/1138-4EB/ghdl-packaging) (ref #901), three versions are provided for each backend:

- `mcode`|`llvm`: stable. Currently, v0.36.
- `mcode-rc`|`llvm-rc`: release candidate. Some commit SHA between 'stable' and 'git'.
- `mcode-git`|`llvm-git`: latest from `master`.

eine, 2020-12-03:
Apart from the PKGBUILD files available in `dist/msys2-mingw` of this repo, GHDL is now upstreamed to MSYS: [mingw-w64-ghdl](https://github.com/msys2/MINGW-packages/tree/master/mingw-w64-ghdl) ([i686](https://packages.msys2.org/package/mingw-w64-i686-ghdl-mcode), [x86_64](https://packages.msys2.org/package/mingw-w64-x86_64-ghdl-llvm)). See msys2/MINGW-packages#5757 and msys2/MINGW-packages#6688. See also [hdl/MINGW-packages](https://github.com/hdl/MINGW-packages).

closed 2020-12-03 by eine

tgingold, 2020-12-04:
Congrats for all the work!