ghdl/ghdl #502: Building GHDL from source with gcc back end fails on debian

arogbonlo, 2017-12-19:
Hi,

Ihave run into the following problem installing ghdl on debian;
After following steps:
```
$ cd <ghdl>
$ mkdir build
$ cd build
$ ../configure --with-gcc=/path/to/gcc/source/dir --prefix=PREFIX
$ make copy-sources
$ mkdir gcc-objs; cd gcc-objs
$ /path/to/gcc/source/dir/configure --prefix=/usr/local --enable-languages=c,vhdl \
--disable-bootstrap --disable-lto --disable-multilib --disable-libssp \
--disable-libgomp --disable-libquadmath
$ make -j2 && make install
$ cd /path/to/ghdl/source/dir
```

`make ghdllib` fails with the following message:

```
root@bar:/opt/ghdl/ghdl_src/ghdl/build# make ghdllib
make -f ../libraries/Makefile.inc LIBSRC_DIR="../libraries" LIBDST_DIR="lib/ghdl" enable_openieee="false" LN="ln -s" CP="cp" MKDIR="mkdir" GHDL="/opt/ghdl/bin/ghdl" GHDL_FLAGS=" " vhdl.libs.all libs.vhdl.standard
make[1]: Entering directory '/opt/ghdl/ghdl_src/ghdl/build'
make[1]: *** No rule to make target '/opt/ghdl/bin/ghdl', needed by 'lib/ghdl/v87/std/std-obj87.cf'.  Stop.
make[1]: Leaving directory '/opt/ghdl/ghdl_src/ghdl/build'
Makefile:236: recipe for target 'libs.vhdl.gcc' failed
make: *** [libs.vhdl.gcc] Error 2
```

Pleas  advice  on how to resolve this.

Best
Xero

tgingold, 2017-12-19:
You haven't used the same prefix to configure gcc and ghdl.

arogbonlo, 2017-12-19:
Thanks, that solved my issue.
My mistake was:
```
$ /path/to/gcc/source/dir/configure --prefix=/opt/gcc --enable-languages=c,vhdl \
--disable-bootstrap --disable-lto --disable-multilib --disable-libssp \
--disable-libgomp --disable-libquadmath
``` 
Whilst I previously I'd done:
```
$ ../configure --with-gcc=/path/to/gcc/source/dir --prefix=/opt/ghdl
```
Thanks Tristan!

closed 2017-12-19 by arogbonlo