ghdl/ghdl #507: GHDL on ArchLinux: AUR and yaourt

1138-4EB, 2017-12-23:
This issue is to track GHDL support on ArchLinux.

> https://wiki.archlinux.org/index.php/Arch_User_Repository
> The [Arch User Repository (AUR)](https://aur.archlinux.org/) is a community-driven repository for Arch users. It contains package descriptions (PKGBUILDs) that allow you to compile a package from source with makepkg and then install it via pacman.

There are currently two competing somehow up-to-date packages with gcc backend. Also, there are two slightly out-of-date packages with LLVM and mcode backends. See [aur.archlinux.org/packages/?K=ghdl](https://aur.archlinux.org/packages/?O=0&SeB=nd&K=ghdl&outdated=&SB=n&SO=a&PP=50&do_Search=Go):

```
$ yaourt -Ss ghdl
aur/ghdl 0.35-1 [installed] (23) (0.64)
    VHDL simulator
aur/ghdl-gcc-git 0.36dev.git20171218-1 (2) (0.68)
    VHDL simulator - GCC flavour
aur/ghdl-llvm-git 0.35dev.git20170819-1 (1) (0.20)
    VHDL simulator - LLVM flavour
aur/ghdl-mcode-git 0.35dev.git20170819-1 (0) (0.00)
    VHDL simulator - mcode backend
```

Paebbels, 2017-12-23:
This relates to [Package GHDL for MinGW32/MinGW64](https://github.com/ghdl/ghdl/issues/497), because MinGW also uses pacman and `PKGBUILD` description files.

go2sh, 2017-12-28:
@1138-4EB Yes, you are right.
But some insides into Arch packaging philosophy:
Every package is build from source of the current stable release , except some none open-source stuff or packages with a "-bin" ending. So for ghdl with gcc backend, the PKGBUILD file downloads gcc and ghdl release tarballs and then builds them. 
Especially inside the AUR, there are some packages with a "-git", "-svn", "-hg" ending, which are build directly from the repository by creating a local copy. There version you see from yaourt isn't actually the version you build. The version you see is the version, when the PKGBUILD file was updated, the version you build is the latest commit of the default (or selected branch) and the version string of the package is generated from the PKGBUILD files `pkgver`. 

So it's totally fine to have those  4 variants to install.

The packaging from a point of an Arch User there is fine. I use it on a regular bases. You could add a llvm and mcode backend package based on the current release, but I don't know if its needed. It also could be nice to add ghdl to the community repository, then a user doesn't have to build ghdl and can download a binary package. But therefore a trusted user must adopt it.

1138-4EB, 2017-12-28:
Thanks a lot @go2sh. I updated the first comment to remove my invalid assumptions (about some packages being out-of-date, because *it's totally fine to have those 4 variants to install*). I also included your explanation there.

> The packaging from a point of an Arch User there is fine. I use it on a regular bases. You could add a llvm and mcode backend package based on the current release, but I don't know if its needed.

If I didn't understand it wrong, the main difference between `aur/ghdl` and `ghdl-gcc-git` is that the first needs to be manually updated for each release, but the second one will always use the latest commit. Does this imply that:

- `git` is a required dependency only for `-git` packages?
- `-git` packages need to be updated less frequently, because even if the name is wrong the built version is the latest?

If the reponse to both questions is yes, I think we can consider non-git packages to be stable releases (therefore adding llvm and mcode variants would make sense). Then, git variants would be development versions, so it is ok for git to be a dependency. Yet, I think it should be specified. Let me explain:

`ghdl-gcc-git` does not make it clear if it is pointing to the `stable` or the `develop` branch of a repo. Indeed, this tricky for GHDL, because until now we have not had any `stable` branch, and develop is named `master`. I think it would be useful to have a `stable` branch (where all the tags lie, see #483). Then, `-git` packages would still point to `master`, but would be known to be explicitly different from hypothetical `stable-git` packages pointing to `stable`. Indeed, `stable-git` are hypothetical, because the compilation result would be the same as `aur/ghdl` (which does not depend on git).

Summarizing, we would have three non-git stable pkgs `aur/ghdl-gcc`, `aur/ghdl-llvm` and `aur/ghdl-mcode`, and three development pkgs `aur/ghdl-gcc-git`, `aur/ghdl-llvm-git` and `aur/ghdl-mcode-git` which include '(latest dev version)' in the description.

@tgingold, what do you think about creating a `stable` or `release` branch here which points to `v0.35`? You just don't touch it for your every work. When you create the next release, you pull to make this branch jump there. The motto is to have a branch with a fixed name which always points to the commit corresponding to the latest release.

> It also could be nice to add ghdl to the community repository, then a user doesn't have to build ghdl and can download a binary package. But therefore a trusted user must adopt it.

Would this be equivalent to us executing `yaourt -S ghdl` in a travis job and pushing the resulting built tarball to GitHub releases (just as we do now for Debian, Ubuntu and Fedora)?

I mean, does the community repository have it's own CI environment which builds the packages to provide binary alternatives? Or is the maintainer responsible of building from sources in it's own machine and then pushing built binaries to the community repository?

In the later, is there any chance of travis being our trusted user? It should be trivial to push the artifacts to a different host as we do now with GitHub releases.

tgingold, 2017-12-30:
Should it be a branch or a tag?

I'm ok with a tag (or a branch) but we then need to document the release process (which should stay as simple as possible)

go2sh, 2018-01-02:
> If I didn't understand it wrong, the main difference between aur/ghdl and ghdl-gcc-git is that the first needs to be manually updated for each release, but the second one will always use the latest [...]

Ok, more general. Arch has a special package convention, which should be followed. There are fixed version packages (stable packages) and development packages (unstable packages) with a suffix pointing to the vcs. There are also other  suffixes like "-bin" for a binary download without building, "-dev(el)" for development snapshots. Some deeper insides in some other topics

Version:
- Stable: Uses the fields from the PKGBUILD file as version
- Unstable:  Uses a function from the PKGBUILD to calculate the version

Note: The AUR website and consequently the AUR helpers (like yaourt) do not evaluate the version function of unstable packages and rely on the same fields as for stable packages. On a general update run, AUR helpers check only the reported version. If it's greater, the package gets an update. (Both stable and unstable packages). To trigger this kind of update, the version fields in the PKGBUILD file must be changed and the PKGBUILD file must be uploaded to the AUR. There is a special update mode for unstable packages in some helpers, where the sources for all unstable packages are checked out and then the version function gets called to check for an update. (This can take some time.^^)

Sources:
- Stable: Release tarball, a (git, svn, etc.) tag
- Unstable: Latest commit of a branch

Note: The packages using a vcs as source, need the vcs tool package as makedepend, since makepkg needs to checkout the source. Besides my first assumption, stable packages, like `ghdl`, can also use vcs as source, but should point to a fixed point like a tag. So the current `ghdl` package with the usage of git is totally fine. (Both as source and as makedep). Furthermore, makedeps are only needed for making a package, the makedeps are not needed for installation.

> If the reponse to both questions is yes, I think we can consider non-git packages to be stable releases (therefore adding llvm and mcode variants would make sense). Then, git variants would be [...]

This makes sense.

> ghdl-gcc-git does not make it clear if it is pointing to the stable or the develop branch of a repo. Indeed, this tricky for GHDL, because until now we have not had any stable branch, and develop is [...]

See my explanations above. Tags can be used as stable points, since they are mostly the same as release tarballs and some projects don't host them. But I think it's just a flavor.

> Summarizing, we would have three non-git stable pkgs aur/ghdl-gcc, aur/ghdl-llvm and aur/ghdl-mcode, and three development pkgs aur/ghdl-gcc-git, aur/ghdl-llvm-git and aur/ghdl-mcode-git which include '(latest dev version)' in the description.

Yes, that would be the ultimate solution, but I don't think the dev version in the description is needed as all Arch users know that convention with the vcs suffixes. The source for the stable versions can be a git tag like it's now.

> Would this be equivalent to us executing yaourt -S ghdl in a travis job and pushing the resulting built tarball to GitHub releases (just as we do now for Debian, Ubuntu and Fedora)?
> I mean, does the [...]

Some more basics: `pacman` is the Arch package manager (not `yaourt`). It is responsible for e.g. dependencies, downloading, signature checking and installing. It's like apt and dpkg in one. `yaourt` is a just a frontend to `pacman` with some special features for AUR support.

Every package is build the same way. (Official and AUR packages)
1) Download PKGBUILD and other packaging sources (icons, install files, desktop files, patches, not the actual source) from any source, e.g. official SVN-Repos, AUR, etc.
2) Run `makepkg`
This can be done by everybody. The installation is also handle through pacman (pacman -U = dpkg -i). An AUR helper combines all those steps (Download, Build, Installation) including dependency resolution for AUR and official packages. Official packages are build the same, but additionally signed and uploaded to the official mirrors.

Sooo, long story short: GHDL could use the AUR packages (the AUR provides packaging tarballs) to create the binary packages and upload the to Github as artifacts or wherever. There is an official Archlinux docker image which could be used (`docker pull archlinux/base`) with trevis.

> In the later, is there any chance of travis being our trusted user? It should be trivial to push the artifacts to a different host as we do now with GitHub releases.

Theoretical yes. A trusted user is just a person with a signed GPG key, which could be uploaded to travis (Maybe?). Practical, maybe not.

go2sh, 2018-01-02:
@tgingold Tags should be fine. Since they are already there. :-)

1138-4EB, 2019-08-27:
Coming from [#901](https://github.com/ghdl/ghdl/issues/901#issuecomment-525473365).

@marzoul, I'm still setting up an `archlinux/base` container to test the current status of your `mcode-git`, `llvm-git` and `gcc-git` packages in AUR. I think I managed to set the environment properly:

```bash
#!/bin/sh

dos2unix PKGBUILD

docker build -t test - <<EOF
FROM archlinux/base

RUN pacman -Sy --noconfirm
RUN pacman -S  --noconfirm grep base-devel

RUN mkdir /wrk \
 && chgrp nobody /wrk \
 && chmod g+ws /wrk \
 && setfacl -m u::rwx,g::rwx /wrk \
 && setfacl -d --set u::rwx,g::rwx,o::- /wrk \
 && echo 'ALL ALL=(ALL) NOPASSWD: /usr/sbin/pacman' >> /etc/sudoers
EOF

winpty docker run --rm -itv /$(pwd)://wrk -w //wrk test bash -c "sudo -u nobody makepkg -sCLfc --noconfirm"
```

but I get an error when it tries to create a copy of the git repo:

```bash
==> Validating source files with md5sums...
    ghdl ... Skipped
==> Removing existing $srcdir/ directory...
==> Extracting sources...
  -> Creating working copy of ghdl git repo...
Cloning into 'ghdl'...
done.
fatal: failed to read object cb66dc7ee55c7138ccc7d1a4ed9f885d72a8a53d: Invalid argument
fatal: the remote end hung up unexpectedly
==> ERROR: Failure while creating working copy of ghdl git repo
    Aborting...
```

I can check that the (bare) repo was properly cloned, and that the sha it is trying to clone now exists. Therefore, I think the issue might be related to the workdir being bind from the Windows 10 host to the container. I will now set a GitHub Actions workflow to run it on a GNU/Linux host. We'll see if this fixes the issue.

---

These are some specific issues with the current PKGBUILD files that I'd like to change/request:

- I think that the URL should be either https://github.com/ghdl/ghdl or http://ghdl.free.fr/, instead of https://github.com/tgingold/ghdl.
- The license is 'GPLv2+', instead of 'GPLv2'. I don't know if the difference is supported.
  - The default build includes non-free libraries. Do you know how can this info be added to the PKGBUILD file?
- Is it possible/reasonable to add the short commit SHA to pkgver instead of the date? For example `v0.36git640.gf977ba0dd5` or `v0.37dev640.gf977ba0dd5`.
- Tests are not executed. I tried to add the testsuite to `check()`, but it fails because it includes expected failures. Anyway, in the CI workflow I think it is better to run the test in a new/clean container.

marzoul, 2019-08-28:
Hi @1138-4EB, thank ou for your observations. I will fix the url and license GPLv2+.

For non-free libs : I'll check again, we should be able to add another license, usually "custom". Other than that there is no particular place to describe the type of free-ness of a package. Experienced users usually disable/enable manually what they want in the PKGBUILD to build their package. If there is a separate command of Makefile recipe in GHDL to build/install "non-free" libs we can make this visible in the PKGBUILD. 

For the version : usually people update packages rarely enough, so the date is a quite human-readable thing while not too long and makes files naturally sorted when listing files. I'll have a look again.

Last time I tried compiling/installing libghdlsynth, it only worked with gcc backend, but that may have been an interfacing issue between chair and keyboard, I'll check again and enable it in all backends where it compiles :-) (it's not that big so it's ok)
makepkg accepts no arguments the kind that could change package config.

BTW, In the discussion there is mention of using yaourt.
I don't use yaourt because it ruins /tmp, does things like compilation with unclear permissions under root, runtime not predictable because of compilation, and all may fail (other tools too) if /tmp is full because objects are too big or because there are too many packages in there.
You may want to check if your failures are related to stuff like that ? I compile packages in my home directory with user permissions, only then use sudo pacman -U to install.

1138-4EB, 2019-08-28:
> For non-free libs : I'll check again, we should be able to add another license, usually "custom". Other than that there is no particular place to describe the type of free-ness of a package. [...]

The point is that we had this 'issue' with Debian when ghdl was removed from official repos because some of the packaged content did not comply with their 'free/open' requirements. Precisely, some of the 'standard' VHDL [libraries](https://github.com/ghdl/ghdl/tree/master/libraries) which are built by default are not free (see comments about ieee, synopsys and mentor in https://ghdl.readthedocs.io/en/latest/licenses.html). That's why [openieee](https://github.com/ghdl/ghdl/tree/master/libraries/openieee) exists, and there is a [`--enable-openieee`](https://github.com/ghdl/ghdl/blob/master/configure#L107) configuration option.

We had some previous discussions about how should this be packaged. On Debian, only variants with openieee might be distributed. Therefore, users need to either install other libraries from sources (I don't know if there is a specific make target for that). Alternatively, we could provide a `ghdl-*-stdlibs` package for each `ghdl-*` package. However, it is not easy to distribute them independently and still ensure that artifacts are compatible.

Conversely, in Fedora all the default libraries are built, so there is no need to deal with this issue. Hence, I believe it is ok if you keep them in AUR packages too. Any kind of reference in the PKGBUILD, such as `['GPLV2+', 'custom']`, will suffice.

> For the version : usually people update packages rarely enough, so the date is a quite human-readable thing while not too long and makes files naturally sorted when listing files. I'll have a look again.

Having files naturally sorted is a fair point which I had not thought about. However, the drawback of having dates only is that it is not possible to uniquely identify the commit, because multiple commits might be pushed the same day. When using this same scripts in CI, it is useful to be able to identify the commit that produced an artifact. Nevertheless, I'm pretty sure that the information is contained in some file with metainfo inside the artifact. Hence, this is not critical. I will also think about including both the date and the SHA.

> Last time I tried compiling/installing libghdlsynth, it only worked with gcc backend, but that may have been an interfacing issue between chair and keyboard, I'll check again and enable it in all backends where it compiles :-) (it's not that big so it's ok)

It might be broken when you tried it. There have been several changes to these features in the last month, including merging the shared libraries. As you say, it is not that big; so, I think the reasons to not enable it by default should only be unmet dependencies (such as minimum version of GCC).

> makepkg accepts no arguments the kind that could change package config.

Well, this should not be a problem for ArchLinux. It might be annoying for msys-mingw, tho.

> BTW, In the discussion there is mention of using yaourt.
> I don't use yaourt because it ruins /tmp, does things like compilation with unclear permissions under root, runtime not predictable because [...]

I'm not a user of ArchLinux myself. I have installed it a couple of times for didactic purposes, and I keep a partition on disk. But I've been unable to set up any of the three latest keyboards I've had. It recognizes `Ctrl`, `Shift` and `Alt` as the same key. It is so annoying to keep fixing it every time.

My interest on having up-to-date packages for ArchLinux is because i) I like the philosophy of the community, ii) they provide an unvaluable source of knowledge through their wiki, iii) PKGBUILD files and pacman are used by MSYS2 too.

So, I get lost when discussing about AUR helpers. It was mentioned in the title just because it was 'the most known helper' when this issue was opened.

> You may want to check if your failures are related to stuff like that ? I compile packages in my home directory with user permissions, 

The point is that I'm doing everything in a docker container, because I don't have a working GNU/Linux host here. Docker containers are executed with 'root' by default, so no other *regular* user exists in the environment. When I found this issue with other projects, I'd just set the workdir to `/tmp/whatever` and start the container with `--user $(id -u)`. However, in this case, `makepkg` uses `sudo` internally, to change the user, and who knows what user-related resources it expects which do not exist.

I followed http://allanmcrae.com/2015/01/replacing-makepkg-asroot/ to configure `/wrk` inside the container as shown in my previous message. Then, I'm running `sudo -u nobody makepkg -sCLfc --noconfirm`. Ideally, I'd like to do `makepkg -sCLfc --noconfirm` and prevent it from using `sudo` at all. `sudo` should not even be installed in the container. Do you know how to achieve this?

> only then use sudo pacman -U to install.

This is desirable, indeed. I don't want to install the package in the same environment, but save it for later.

marzoul, 2019-09-10:
@1138-4EB Install of synthesis lib is now enabled in the 3 Archlinux packages.

Following Archlinux conventions, the string "GPLv2" means GPLv2 or later so I won't modify this part.
[https://wiki.archlinux.org/index.php/PKGBUILD#license](https://wiki.archlinux.org/index.php/PKGBUILD#license)

I'm interested in the `--enable-openieee` configuration option. There is no information about it in [https://ghdl.readthedocs.io/en/latest/licenses.html](https://ghdl.readthedocs.io/en/latest/licenses.html)
If `openieee` is a drop-in replacement and reasonably similar completeness than `ieee` libraries then I would be happy to enable these by default. Is there an `openieee` for VHDL-2002 (--std=02)?

It seems your `makepkg -sCLfc --noconfirm` only uses `sudo` internally because of option `-s` which installs dependencies if needed. Instead you could use a container where you pre-install dependencies once, and use makepkg only for build operations. That would also reveal when packages end up requiring unexpectedly stuff.

1138-4EB, 2019-09-10:
> @1138-4EB Install of synthesis lib is now enabled in the 3 Archlinux packages.

Great!

> Following Archlinux conventions, the string "GPLv2" means GPLv2 or later so I won't modify this part.

Good.

> I'm interested in the `--enable-openieee` configuration option. There is no information about it in https://ghdl.readthedocs.io/en/latest/licenses.html

I don't think this is documented at all.

> If `openieee` is a drop-in replacement and reasonably similar completeness than `ieee` libraries then I would be happy to enable these by default.

I think it is a drop-in replacement, but it is not as complete as `ieee` libraries. See:

https://github.com/ghdl/ghdl/blob/cb3468002497ab2090bef31d559e2eaf4747b90a/configure#L209-L210

https://github.com/ghdl/ghdl/blob/cb3468002497ab2090bef31d559e2eaf4747b90a/dist/travis/build.sh#L52-L55

https://github.com/ghdl/ghdl/blob/cb3468002497ab2090bef31d559e2eaf4747b90a/Makefile.in#L582-L590

It seems that `gna` tests are removed because they may contain non-free sources. However, there is no comment stating that non-supported features are required. Therefore, you might want to build ghdl with `--enable-openieee` and then run the `gna` testsuite. This will let you know how complete it is.

> Is there an `openieee` for VHDL-2002 (--std=02)?

I think that openieee sources are for all the versions of the standard. VHDL-2002 is compatible with VHDL-1993, so `93` is used. See [#908](https://github.com/ghdl/ghdl/issues/908#issuecomment-527699541).

> It seems your `makepkg -sCLfc --noconfirm` only uses `sudo` internally because of option `-s` which installs dependencies if needed. Instead you could use a container where you pre-install [...]

Thanks a lot for the hint! I will try it!

1138-4EB, 2019-09-11:
@marzoul, I had a look at https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=ghdl-mcode-git. I'd say that the last step (`make DESTDIR="${pkgdir}" install.libghdlsynth`) is not required, because libghdlsynth is deprecated (https://github.com/ghdl/ghdl#advanced-users). Now, synthesis features are merged in libghdl. By the same token, I'd add `--enable-libghdl` to the configuration. It is enabled by default, but it doesn't harm.

Regarding the description, since ghdl is not just a simulator anymore, I'd update it something like `open-source analyzer, compiler and simulator for VHDL (mcode backend)`.

marzoul, 2019-10-07:
I followed your advice for `--enable-libghdl`.
The libiberty issue was a mistake, it has been fixed.
I added a lib symlink `libghdl.so` to ease development of "client" applications. Maybe this could/should be done by ghdl install ?
I'll update the title as your suggested one is the description inside the readme of the repo.

eine, 2020-09-15:
> I added a lib symlink `libghdl.so` to ease development of "client" applications. Maybe this could/should be done by ghdl install ?

@marzoul, I missed this comment. Where did you add the symlink to? The shared libs should be found in the expected location. So, if GHDL is installed to `*/bin/ghdl`, the libs should be at `*/lib/libghdl*`.

marzoul, 2020-09-17:
@eine I added it at same place where original *.so is installed.
In Archlinux, in directory /usr/lib : 
```
libghdl-1_0_dev.so
libghdl.so -> libghdl-1_0_dev.so*
```
This is ease linking with libghdl and ease dealing with the version installed.

eine, 2020-09-17:
Oh. Now I understand. Yes, it makes sense. @tgingold, what do you think?

tgingold, 2020-09-18:
I am not sure: as there is no fully backward compatibility, this would be confusing: there is no uniq libghdl.so

If you use the ghdl command to link, it takes care of the complex name.

eine, 2020-09-18:
My understanding is that `libghdl.so` would be a symbolic link to whatever version is installed last. So, users could and should keep retrieving the lib from the command. This would be an addition, not a replacement.

tgingold, 2020-09-18:
Yes, but if you update the library (but keep the old one), all your old programs would break.

Xiretza, 2020-09-18:
Isn't that what [sonames](https://en.wikipedia.org/wiki/Soname) are for? If implemented properly, there'd be a `libghdl.so.1(.x)` with soname `libghdl.so.1`, and if there's ever an ABI change, that gets bumped to `libghdl.so.2`. `libghdl.so` is a symlink to the latest installed version so you can simply link against `-lghdl`, but the thing that gets embedded in the final executable is the versioned soname from the shared object - if you run the executable, the dynamic linker will search for a (e.g.) `libghdl.so.1`.

tgingold, 2020-09-18:
Yes, you're right about the soname.