ghdl/ghdl #509: libghdlsynth.so

RCoeurjoly, 2017-12-28:


Paebbels, 2017-12-28:
Can you add the makefile log either as an edit to your question or as an attachment?

RCoeurjoly, 2017-12-28:
make libghdlsynth.so 
if test -d ./.git \
	   && desc=`cd .; git describe --dirty`; then \
	  sed -e "s/[(].*[)]/($desc)/" < src/version.in > version.tmp; \
	else \
	  cp src/version.in version.tmp; \
	fi
if [ ! -r version.ads ] || ! cmp version.tmp version.ads > /dev/null; then cp version.tmp version.ads; fi
gnatmake -I- -aI. -D pic -z libghdlsynth -o libghdlsynth.so -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -gnatwe -fPIC -gnat12 -aI./src -aI./src/vhdl -aI./src/grt -aI./src/psl -aI./src/vhdl/translate -aI./src/synth -aI./src/ghdldrv -aI./src/vhdl/simulate -bargs -shared -Llibghdlsynth_ -largs -shared  pic/jumps.o  pic/times.o  pic/grt-cbinding.o -Wl,-soname,libghdlsynth.so
gcc-4.6 -c -I. -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -gnatwe -fPIC -gnat12 -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/synth -I./src/ghdldrv -I./src/vhdl/simulate -I- -o /home/rcl/ghdl/pic/netlists.o /home/rcl/ghdl/src/synth/netlists.adb
netlists.ads:248:04: warning: no warning suppressed by this pragma
gnatmake: "/home/rcl/ghdl/src/synth/netlists.adb" compilation error
make: *** [libghdlsynth.so] Error 4

1138-4EB, 2017-12-28:
Hi @RCoeurjoly , on the one hand, did you edit and remove a couple of messages here? Just to know, because I saw them half an hour ago, and it's driving me mad now.

About the issue, are you trying to compile the 'vanilla' GHDL with gcc backend or trying to build tgingold/ghdlsynth-beta?

RCoeurjoly, 2017-12-28:
I just put the makefile log into the issue description instead of on a separate comment. Sorry for the confusion.

I am trying to build tgingold/ghdlsynth-beta by following the How to build as a module steps in the README.

Since the step that I am stuck in concerns GHDL, I figured I put the issue here, but I could put it in ghdlsynth-beta if that's more appropriate.

1138-4EB, 2017-12-28:
> I just put the makefile log into the issue description instead of on a separate comment. Sorry for the confusion.

Thanks you made it clear. It' ok. Me myself edit and move content quite a lot. Just wanted to make sure I didn't 'dream about it'.

> I am trying to build tgingold/ghdlsynth-beta by following the How to build as a module steps in the README.

I personally think this issue fits better in ghdlsynth-beta, because it is in a very early development and not as actively maintained as GHDL itself. This means that ghdlsynth is somehow expected to always fail, except in very narrow contexts. Keeping this type of issues there helps us know which are bugs/fixes for the main project and which are interesting but accesory.

Anyway, this is not a high traffic project. So I'd just suggest that you state clearly in the first message what you are trying to achieve. E.g.:

- I am trying to build ghdlsynth-beta with a prebuilt GHDL (version v0.0, downloaded from GitHub releases) and GCC (version 7_2_0, installed through apt-get).
- I downloaded GHDL sources and built it with gcc backend as explained in XXXX. I downloaded GHDL sources through git clone, and got GCC source with wget/curl from XXXX. Up to here, everything works XXXX. Now, I want to add ghdlsynth-beta.

> Since the step that I am stuck in concerns GHDL, I figured I put the issue here, but I could put it in ghdlsynth-beta if that's more appropriate.

Note that almost every ghdlsynth-related issue concerns GHDL, but most GHDL-related issues do not concern ghdlsynth at all. That is, when you open an issue here, we default to think that it has nothing to do with synthesis. However, if you open an issue in ghdlsynth, we all expect it to be somehow related to GHDL.

Anyway, as said, it's ok. Let's just focus on knowing the complete and exact list of steps that you executed. Imagine you where given a clean Ubuntu14.04. What are the steps until you get the error?

RCoeurjoly, 2017-12-29:
@1138-4EB I updated the first comment to reflect the steps until I get the error.
I will try later to reproduce the error on Trisquel 7 (based on Ubuntu 14.04) and Ubuntu 16.04 machines.

1138-4EB, 2017-12-29:
I reproduced the steps you described with a difference: I built GHDL with gcc backend (I think you did with mcode). I works on Debian Sid: https://travis-ci.org/1138-4EB/ghdl/jobs/322845901. Note that the error is produced after everything is installed, due to a wrong path: https://api.travis-ci.org/v3/job/322845901/log.txt You can see the full procedure here: https://github.com/1138-4EB/ghdl/blob/gcc-in-docker/dist/linux/buildtest.sh

I am not sure, but ghdlsynth might be only available with gcc backend. @tgingold?

tgingold, 2017-12-30:
I think the warning can be avoided to be fatal if you configure ghdl with --disable-werror

You can use any backend for ghdlsynth

RCoeurjoly, 2017-12-30:
In the same machine, I configure GHDL with --disable-werror:
``~/ghdl$ ./configure --disable-werror ``

I now get a different but similar error upon trying to build libghdlsynth.so

``~/ghdl$ make libghdlsynth.so 
if test -d ./.git \
	   && desc=`cd .; git describe --dirty`; then \
	  sed -e "s/[(].*[)]/($desc)/" < src/version.in > version.tmp; \
	else \
	  cp src/version.in version.tmp; \
	fi
if [ ! -r version.ads ] || ! cmp version.tmp version.ads > /dev/null; then cp version.tmp version.ads; fi
gnatmake -I- -aI. -D pic -z libghdlsynth -o libghdlsynth.so -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -fPIC -gnat12 -aI./src -aI./src/vhdl -aI./src/grt -aI./src/psl -aI./src/vhdl/translate -aI./src/synth -aI./src/ghdldrv -aI./src/vhdl/simulate -bargs -shared -Llibghdlsynth_ -largs -shared  pic/jumps.o  pic/times.o  pic/grt-cbinding.o -Wl,-soname,libghdlsynth.so
gcc-4.6 -c -I. -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -fPIC -gnat12 -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/synth -I./src/ghdldrv -I./src/vhdl/simulate -I- -o /home/rcl/ghdl/pic/netlists-dump.o /home/rcl/ghdl/src/synth/netlists-dump.adb
netlists-dump.adb:147:32: missing ".."
netlists-dump.adb:394:35: missing ".."
gnatmake: "/home/rcl/ghdl/src/synth/netlists-dump.adb" compilation error
make: *** [libghdlsynth.so] Error 4``

tgingold, 2017-12-30:
In the latest message, you are using GCC 4.6, which is too old. Can you try again with GCC 7.2?

RCoeurjoly, 2017-12-31:
I get the following error with gcc version 7.2.0 (Ubuntu 7.2.0-1ubuntu1~14.04) 

`~/ghdl$ sudo make libghdlsynth.so 
[sudo] password for rcl: 
if test -d ./.git \
	   && desc=`cd .; git describe --dirty`; then \
	  sed -e "s/[(].*[)]/($desc)/" < src/version.in > version.tmp; \
	else \
	  cp src/version.in version.tmp; \
	fi
if [ ! -r version.ads ] || ! cmp version.tmp version.ads > /dev/null; then cp version.tmp version.ads; fi
gnatmake -I- -aI. -D pic -z libghdlsynth -o libghdlsynth.so -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -fPIC -gnat12 -aI./src -aI./src/vhdl -aI./src/grt -aI./src/psl -aI./src/vhdl/translate -aI./src/synth -aI./src/ghdldrv -aI./src/vhdl/simulate -bargs -shared -Llibghdlsynth_ -largs -shared  pic/jumps.o  pic/times.o  pic/grt-cbinding.o -Wl,-soname,libghdlsynth.so
gcc-4.6 -c -I. -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -fPIC -gnat12 -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/synth -I./src/ghdldrv -I./src/vhdl/simulate -I- -o /home/rcl/ghdl/pic/netlists-dump.o /home/rcl/ghdl/src/synth/netlists-dump.adb
netlists-dump.adb:147:32: missing ".."
netlists-dump.adb:394:35: missing ".."
gnatmake: "/home/rcl/ghdl/src/synth/netlists-dump.adb" compilation error
make: *** [libghdlsynth.so] Error 4
`
I installed all the tools and configured successfully an iCEstick with GCC version 5.4.0, so the GCC version doesn't seem to be an issue.

tgingold, 2018-01-01:
Sorry, but in the log I can read gcc-4.6
So you may have installed gcc 7, but still use gcc 4.6 for Ada.
What is the output of gnatls -v?
Do you have a package for the gnat compiler?

RCoeurjoly, 2018-01-01:
I had the following gnalts version

`~/ghdl$ gnatls -v

GNATLS 4.6
Copyright (C) 1997-2010, Free Software Foundation, Inc.

Source Search Path:
   <Current_Directory>
   /usr/lib/gcc/x86_64-linux-gnu/4.6/adainclude/

Object Search Path:
   <Current_Directory>
   /usr/lib/gcc/x86_64-linux-gnu/4.6/adalib/

Project Search Path:
   <Current_Directory>
   /usr/share/ada/adainclude/`

I installed GNAT 7.2:

`sudo apt-get install gnat-7`

And now I have the iCEstick demo synthesized. That was it.
Thank you very much!

tgingold, 2018-01-01:
Thank you for the update!

closed 2018-01-02 by tgingold