ghdl/ghdl #511: Instructions to fork GHDL

1138-4EB, 2017-12-29:
I realized that some users fork my fork of this repo (1138-4EB/ghdl), instead of ghdl/ghdl. Please, don't do that:

- Most likely you want to fork the official repo where we all contribute to. That has always been tgingold/ghdl, but was recently moved to ghdl/ghdl.
- There is a lot of trash and wip in my fork. I am sure you don't want at least 7 of the 10 branches there.

If you really want to work on some content which is only available in a fork, follow this procedure:

```
# Fork ghdl/ghdl in GitHub
# Clone your fork
git clone https://github.com/<username>/ghdl
cd ghdl
# Add the fork as a new remote
git remote add <forked_repo_name> <forked_repo_url>
# Create a new branch that points to a remote branch in the new remote
git checkout -b <new_branch_name> <forked_repo_name>/<remote_branch_name>
```

For example, if you want to work on branch `gcc-in-docker` from my fork:

```
git clone https://github.com/<username>/ghdl
cd ghdl
git remote add 1138 https://github.com/1138-4EB/ghdl
git checkout -b gcc-in-docker 1138/gcc-in-docker
```

1138-4EB, 2017-12-29:
I must admit this is partially my fault:

I have a modified README in the master branch of my repo to let users know about this. But at some point I configured the repo to use `tgingold-master` as the main branch, which is, indeed, a raw copy of the master branch here. So users visiting my fork might not be really aware. This is aggravated because most of the content I post here relies on resources only available in my fork, yet (such as web page related proposals).

I fixed it, and I added a link to this issue: https://github.com/1138-4EB/ghdl/blob/master/README.md

---

@eggsactly, I realized that you forked my repo an hour ago. You might want to delete it and fork from ghdl/ghdl.

Alternatively, you might remove all the branches except `master`, `tgingold-master` and any other you are really intered in. Also change the settings so that the master branch is the default. Last, the `deploy_key.enc` won't work for your fork. If you want to regenerate it, because you are trying to work on GitHub pages related stuff, let me know.

eggsactly, 2017-12-29:
@1138-4EB I forked your repo to see if I could modify the proposed GHDL web page to look good on mobile. I'll be logged into gitter for the night.

1138-4EB, 2017-12-29:
Oh, that's defined in branch builders right now:

- The content is defined in [builders/ghdl-io/content/_index.md](https://github.com/1138-4EB/ghdl/blob/builders/ghdl-io/content/_index.md).
- [builders/ghdl-io/themes/ghdl-io-theme/layouts](https://github.com/1138-4EB/ghdl/tree/builders/ghdl-io/themes/ghdl-io-theme/layouts) contains the layouts.
  - The landing page, [index.html](https://github.com/1138-4EB/ghdl/blob/builders/ghdl-io/themes/ghdl-io-theme/layouts/index.html), is a instance of [_default/baseof.html](https://github.com/1138-4EB/ghdl/blob/builders/ghdl-io/themes/ghdl-io-theme/layouts/_default/baseof.html).
  - Partial blocks are defined in [partials/](https://github.com/1138-4EB/ghdl/tree/builders/ghdl-io/themes/ghdl-io-theme/layouts/partials).
- CSS sources are saved in [builders/ghdl-io/themes/ghdl-io-theme/static/css](https://github.com/1138-4EB/ghdl/tree/builders/ghdl-io/themes/ghdl-io-theme/static/css)

Have a look at [gohugo.io/templates/introduction](https://gohugo.io/templates/introduction/) and [gohugo.io/templates/partials](https://gohugo.io/templates/partials/).

Note that you can watch it locally and have it updated as you change the sources: https://1138-4eb.github.io/ghdl/about/site/#local Indeed, I suggest you download a ZIP/tarball of my builders branch and work locally. I am going to split web page specific content to a separate branch, so that it is easier for you to contribute.

About the night, what's you time zone?

eggsactly, 2017-12-29:
Thanks for the links, I'm looking over them. I see you created a new folder ghdl-io for the site. I'll download a tarball of your builders branch and figure out how the build system works. My time-zone is MST.

closed 2018-12-12 by tgingold