ghdl/ghdl #514: Document the release process

1138-4EB, 2017-12-30:
Coming from #507. I open this issue to track the creation of a `stable` branch/tag, which will point to the commit that created the latest release.

> [tgingold](https://github.com/ghdl/ghdl/issues/507#issuecomment-354533047)
> Should it be a branch or a tag?

I really don't know which are the implications. I'd say that it is better for it to be a branch. Shall it be a tag, we will have 'double' releases: one with the version name and a second one when the stable tag is updated. A branch would not trigger two releases in travis.

Anyway, let me investigate, and we'll decide later.

> [tgingold](https://github.com/ghdl/ghdl/issues/507#issuecomment-354533047)
> I'm ok with a tag (or a branch) but we then need to document the release process (which should stay as simple as possible)

Agree. I'll tesk it in my repo and I will contribute a 'release script'. I don't expect it to be more than ten steps:

- create version branch from master
- commit to version branch to bump number
- tag commit with version number
- push tag to remote
- merge branch/tag to master
- push master to remote
- pull master to stable
- push stable to remote

Paebbels, 2017-12-30:
Tags are fixed references in Git, whereas branches are sliding references. As the `stable` pointer will be moved over time, I suggest to use a branch. Sliding tags are possible but create a lot of problems.

In the view of Git, `master` represents the *always perfectly working source code*. So `master` should be considered the *stable* version. Why? Because `master` is Git's default branch, that is automatically used for many operations like `git clone`. It is possible to change the default branch on the server, but I'm not sure if all tools support such a change.

I would prefer the following flow using 3 branches:
- **`master`**  
  The *stable* version.
- **`dev` - the *development* version.**  
  All development is collected on `dev`.
  All PRs need to be addressed against `dev`.
- **A version branch like `v0.36`**  
  It's used to freeze development from `dev` and collects remaining changes until `v0.36` can be merged into `master`. Build systems referring to Git branches can either use the version branch (*specific stable version*) or the master branch (*lastest stable version*)

[image]  
(Source: PLC2 - Git training)

GHDL doesn't need to use a `release` branch. It can simply use version branches.

This flow allows hotfixes on 2 positions in the version history tree:
- A version branch can still get bugfixes. E.g. via `git cherry-pick`.
- The `master` branch can be hotfixed too.

GHDL has already version branches, so there is no big change.

1138-4EB, 2017-12-30:
> Tags are fixed references in Git, whereas branches are sliding references. As the stable pointer will be moved over time, I suggest to use a branch. Sliding tags are possible but create a lot of problems.

Tehn, we should use a branch.

> In the view of Git, master represents the always perfectly working source code. So master should be considered the stable version. Why? Because master is Git's default branch, that is automatically [...]

I really don't mind the naming and it makes sense to use `master` as stable and add a `dev`. However, I would change the default branch on this repo, so that `dev` is used. That will allow to keep the current behaviour: git clone without arguments pulls the lastest commit. In other words, we freeze master now, and start using dev as master. Every contributor will need to be aware of the change, but only currently open PRs are affected.

> I would prefer the following flow using 3 branches:

Where are the tags there? If v2.3 and v2.4 are considered to be both, branches and tags, this flow, as shown in the image, would revert us to issue #483. Indeed, v0.35 is the only version properly shown, because branch `ghdl-0.35` has not received any hotfix, yet.

Note that in [A successful Git branching model](http://nvie.com/posts/a-successful-git-branching-model/) (where Time goes from top to down), the latest commit in the release branch is merged to both `master` and `develop`.

Therefore, I suggest duplicating tags, when a branch receives hotfixes. In the image above:

- Tag v2.3 would point to C7 (with commit message *bump version v2.3*).
- Tag v2.4 would point to C12 (with commit message *bump version v2.4*).
- Tag v2.3-hotfix would point to C14 (and it would be a "moving" release because hotfixes force it).
- Tag v2.3-hotfix would point to C15 (and it would be a "moving" release because hotfixes force it).

This would ensure that `master` has a tag corresponding to each release it its history.

However, I don't know if we should "merge" the functionality of master and release in the image, or the functionality of release and dev.

Paebbels, 2017-12-30:
I would not suggest to change the *default* branch. `git clone` should always fetch the latest stable version, but not a development version. I users want to build from a *testing* version, they need to explicitly switch branches. That's also the behavior e.g. of Debian (*stable/stretch* vs. `testing`).

Normally, tags are attached to commits on the `master` branch.

> Note that in A successful Git branching model (where Time goes from top to down), the latest commit in the release branch is merged to both master and develop.

These backport merges are required, because they did development work on the `release` branch. Ideally, it shouldn't contain such work. If the `release` branch got code changes that are needed by `dev`, then a cherry-pick or merge is needed.

> A successful Git branching model: every commit on master is a new release by definition, remember.
Yes, but release preparation commits are not *stable*. There is a time-frame of several days, while releasing, when a user could clone a broken version from `master`.

Paebbels, 2017-12-30:
This his how a flow could be implemented on GitHub:

* commit and push to `dev`
* if development is done, create a branch `release`
* create a merge request to enable code-review and commenting on that release
* do last changes on `release`
* merge `release`into `master`

Advantages:
* development can go on on `dev`
* release dependent changes are not showing up on `dev`
* `master` can always be downloaded and build

If a version branch is needed, we canreplace `release` by the specific version branch like `ghdl-0.36` or rename `release` after it was merged. Then `ghdl-0.36` becomes an abandoned branch.

1138-4EB, 2017-12-30:
> I would not suggest to change the default branch. git clone should always fetch the latest stable version, but not a development version. I users want to build from a testing version, they need to explicitly switch branches. That's also the behavior e.g. of Debian (stable/stretch vs. testing).

I agree. However, in GHDL, which has never had two different branches, we expect any user using git to be searching for the latest version. If they wanted a specific release, they would either download binaries or packages through their package manager. There is not any know issue that forces users to stick with older versions (stable instead of dev), is it? I think that making GitHub point to stable will make sense when there is a GHDL 1.0 and dev contains breaking changes towards 1.1.

> Normally, tags are attached to commits on the master branch.

How does it apply to tags v0.33 and v0.34 here? Should all those be moved to the closest master commit and leave ghdl-* branches without any tag pointing to the head? I.e., if a hotfix is applied to branch ghdl-0.35, do we need to create a new tag?

> Yes, but release preparation commits are not stable. There is a time-frame of several days, while releasing, when a user could clone a broken version from master.

Indeed. But the 'release' branch can be temporal, not permanent. You create branch 'release' by C6 and C7 from C4, in order to prepare the release (while master is pointing to C2). When you are ready, you pull C6 on top of C2 and rebase C7 on top of the new C6, then push master to the remote. The remote jumps directly from C2 to C7, which are both stable. Then you remove 'release' branch. Last, you tag C7, create a branch from it, and push both.

Note that this keeps the advantages you listed, but it avoids having a permanent release. We can just create release-v0.0 branches as if they where any other feature branch. It is a 2.5 branches scheme.

> If a version branch is needed, we canreplace release by the specific version branch like ghdl-0.36 or rename release after it was merged. Then ghdl-0.36 becomes an abandoned branch.

Do we want/need to provide tags/releases for hotfixed branches? Are we ok with not  providing hotfixed binaries and, thus, 'forcing' users to build those from sources?

tgingold, 2018-01-01:
What is currently proposed is somewhat rather complex...

I don't think we need a dev branch.  We use CI, so a merge cannot introduce regression.  Likewise, I also test my direct commits so that there is no regression.

According to a previous discussion, releases have to be done on master to correctly get the number of changes.

So why not simply have a stable branch, possibly without any CI on it ?

1138-4EB, 2018-01-27:
This is a 2-3 branch model which will most of the time be a single branch in practice, as it is now:

[ghdl-branching-model.pdf](https://github.com/ghdl/ghdl/files/1669939/ghdl-branching-model.pdf)

- Even though R0 and R1 are drawn separately, once M0 is done, they are shown as part of column `dev-master`.
- C4 is shown just as an example. It is easy to avoid it so that R2 and R3 are also shown as part of column `dev-master`.
- Everything would be the same as it is now, but I think that this graph somehow formalizes it.

The release procedure would be:

- Create branch to prepare release.
- After preparation commit are done, bump version.
- Then:

```
VERSION="0.36"

git tag v$VERSION
git push origin v$VERSION

git checkout dev
git merge ghdl-$VERSION
git push origin

git checkout master
git reset --hard ghdl-$VERSION
git push origin +master
```

Paebbels, 2018-01-27:
I'm against fast-forward merges for feature branches.

I proposed 2 branches: `master` and `dev`. The `release` branch is optional. You will be working on `dev` most of the time. If you adhere to *Git flow*, you will also have a lot of tool support in Git GUIs and by additional Git commands on console.

I see no real overhead in using 2 or 3 branches compared to one branch. It's Git not SVN. Git is design to give us powerful branches, so let's use them :).

1138-4EB, 2018-01-27:
> I'm against fast-forward merges for feature branches.

Indeed, it is not required at all. I replaced it with a fast-forward.

I believe that the rest of your comments can be directly applied to the proposed model.

go2sh, 2018-01-29:
Personally, I find the git flow a little bit to complicated. I think one integrated release and integration branch with CI, like now, is sufficient. Especially with the current version numbering and development process.

Paebbels, 2018-01-29:
@go2sh  
Currently, we have only one branch. The "Git Flow" as proposed by Vincent Driessen is a recommendation and best practice to show what is possible with Git. That does not mean that we need to use all of it.

1138-4EB, 2018-01-29:
@go2sh I agree about keeping a single main branch (for release and integration). But, at the same time, we need some model (as simple as it can be) to handle release branches, tags and hotfixes. Indeed, the diagram proposed by @Paebbels and my later simplification are meant to take the bare minimum from the "Git Flow".

go2sh, 2018-01-29:
Hm, if you take that diagram: Feature branches and hotfix branches are somewhat there through the Github Pull-Request process. But the important part of this discussion should be, what is the development workflow and not how is it represented in git. 

Currently there is know such thing as a feature freeze before a release, which would be required for a release branch. Also, there is no support for multiple versions and bugfixes aren't backported, which would be required for dedicated hotfixes. This creates to questions:
- Do you (or Tristan ;) ) want some kind of feature freeze?
- Do you (or Tristan ;) ) want to support multiple Versions of GHLD?
- Do you (or Tristan ;) ) want to backport bugfixes to a certain release? (eg. v0.35.1)
I would answer all questions with no, because I like the way how fast new features get shipped together with bugfixes. Maybe, I would create more releases based on a fixed timing to enable distros to update more frequently.

A completely new game comes up, when there is some software depending on ghdl (eg. python binding) and you don't want to brake the api all the time. So in the end I'am not against the git flow it self, but are against changing the underlying workflow.

1138-4EB, 2018-01-29:
Thanks for making those explicit questions. I'd also answer no to all of them, because, as you say, it is great that new features and bugfixes are shipped together and fast. Indeed, right now I don't know why would anyone want to stick with an "old" version.

That's why I'd like to represent the current workflow. I'm sorry if it sounded like I was proposing a different workflow. There would be a single (minor) addition to the workflow: to keep a `stable` branch pointing to the latest release.

- Feature branches and hotfix branches are currently supported through GitHub PRs. That's ok.
- There is no feature freeze at all. Then, most of the commits that would go in "release branches" can be pushed directly to the main development branch. Yet, I think it is worth underlining that every tag should be a parent of the development branch. See #483.
- We support multiple "versions" of GHDL, given that we accept each release to be a version. However, in practice we don't, because when a user has some problem with an "old" version, the response is to update first.
- As a rule of thumb, we do not backport bugfixes. Yet, branch `ghdl-0.34` is 19 commits ahead of the latest commit shared with `master`, and branch `ghdl-0.33` is 13 commits ahead. At some point `ghdl-0.35` might receive some commits. These might not be related to GHDL's codebase, but to auxiliary files, such as those related to CI services (Travis, ReadTheDocs, DockerHub...). Related to #483, shall a "real" bufix/hotfix be added, it is important not to move release tags (as it is currently done with v0.33 and v0.34), but to add a new one if required (v0.35.1 as you suggested).

> Maybe, I would create more releases based on a fixed timing to enable distros to update more frequently.

The main objectives of this issue are:

- Describe what is the current workflow. We "know" it, but it there is no description of the "model".
- Come up with a (simple) script that Tristan can execute to make the release process easy and consistent across releases.
- Create a diagram describing how should modifications be applied to old branches. Not the "Git Flow" but the "GHDL Flow" using git.

Then, it'll be up to him to pick a period, and maybe change the version format.

> A completely new game comes up, when there is some software depending on ghdl (eg. python binding) and you don't want to brake the api all the time. So in the end I'am not against the git flow it self, but are against changing the underlying workflow.

I think we need not to worry about breaking the API for now. GHDL is known to be in continuous development, so the parts that are prone to change are quite known and described in the docs. Then, it all comes to: how can we represent the current workflow from a "git flow" point-of-view?

eine, 2020-01-21:
Replying to https://github.com/ghdl/ghdl/issues/1108#issuecomment-576618593:

@CraigNoble, while I agree with your point, GHDL is, for now, a single man show. Hence, there is no such distinction between stable or less stable features. By the same token, no feature branches are public; all the branches except `master` are releases. In general, all non-stable features are disabled by default until Tristan is ok with enabling them. Hence, although we could make such an statement, it would be difficult to apply it in practice.

Of course, we can pick the idea and propose to extend the current release scheme to 2-4 releases a year. The first one would be the good/strong/proper release, and others would be release candidades ("pre-release" in GitHub). I like this approach, as I think it would make it easier to have consistent pre-releases distributed through repositories. Currently, each distribution picks a different commit for `*-dev`.

eine, 2020-04-11:
FTR, https://github.com/ghdl/ghdl/issues/1108#issuecomment-596267218