ghdl/ghdl #516: Wired error message

Paebbels, 2018-01-05:
This is a VHDL-2017 code snippet and can not be analyzed by GHDL. But it shows a wired second error message due to the first error.

**Reproducer:**
```vhdl
package reflection is
  type ENUMERATION_VALUE_MIRROR_PT is protected
    impure function pos   return INTEGER;
    impure function image return STRING;
  end protected;
  type ENUMERATION_VALUE_MIRROR is access ENUMERATION_VALUE_MIRROR_PT;
	
  type ENUMERATION_SUBTYPE_MIRROR_PT is protected
    impure function enumeration_literal(literal_name : STRING)       return ENUMERATION_VALUE_MIRROR;
  end protected;
end package;
```

**GHDL command:**
```powershell
ghdl.exe -a --std=08 .\ghdl.vhdl
```

**Analyze messages:**
```
.\ghdl.vhdl:6:36: designated type must not be a protected type
.\ghdl.vhdl:9:21: method return type must not be access of file
```

**Questions:**
- Should the second error message appear?
- Should the second message refer to the *protected type* instead of *file*?

diogratia, 2018-01-05:
The first error:

IEEE Std 1076-2008  
**5.4 Access types**  
**5.4.1 General**  
>For each access type, there is a literal null that has a null access value designating no object at all. The null value of an access type is the default initial value of the type. Other values of an access type are obtained by evaluation of a special operation of the type, called an *allocator*. Each such access value designates an object of the subtype defined by the subtype indication of the access type definition. This subtype is called the *designated subtype* and the base type of this subtype is called the *designated type*. The designated type shall not be a file type or a protected type.

**13.5  Order of analysis**
>   If any error is detected while attempting to analyze a design unit, then the attempted analysis is rejected and has no effect whatsoever on the current working library.

Once your beyond the first error what you do is implementation dependent.

Perhaps that error message should have been:  
>"method return type must not be access of file or protected type" 

when specifying -2000, -2002 or -2008 standard compliance? 

Protected types were introduced in the -2000 revision and the corresponding paragraph in IEEE Std 1076-2000 (3.3 ) had "or protected type" added.  

As far as stopping after the first error it depends on whether the earlier declaration resulting in an error is used in a further declaration - you can't tell without encountering it's name during semantic analysis. A heuristic pass during semantic analysis might tell you when to quit fiddling with errors.

An alternative may be to 'null out' declarations that don't complete (mark them as failed). It'd do wonders for 'fall through' error messages. In this case the second error message could be along the lines of  
>"enumeration_value_mirror is not visible here"

Where that could get interesting is when a name is visible through inheritance from outside the current declarative region.  You could offset that with the failed declaration.

Paebbels, 2018-01-05:
In general I like that GHDL is trying to detect as many errors as possible.

If possible, I'd like to see the proposed "not visible here due to earlier errors" message.

tgingold, 2018-01-05:
This is just a typo in the error message; should be : must not be an access *or* a file

tgingold, 2018-01-05:
> .\ghdl.vhdl:6:36: designated type must not be a protected type
> .\ghdl.vhdl:9:21: method return type must not be access of file

Allowing access to protected type is possible, but we need to deal (or not) with race condition:
What does happen if a process is deallocating a protected type while another process is calling one of its method ?

Paebbels, 2018-01-05:
Access types to protected types can only be created for allocated protected types. So you can not create them for declared objects of protected types and use an attribute to create a pointer link in Ada.

With VHDL-2017 we changed the rules for `deallocate` and allowed garbage collection. So `deallocated` does nothing or at maximum it's a hint to the GC to mark the denoted object for the next GC round.

tgingold, 2018-01-05:
> On 05 Jan 2018, at 09:11, Patrick Lehmann <notifications@github.com> wrote:
> 
> Access types to protected types can only be created for allocated protected types. So you can not create them for declared objects of protected types and use an attribute to create a pointer link in Ada.

That doesn't change anything: two processes can refer to the same protected type via the same access.

> With VHDL-2017 we changed the rules for deallocate and allowed garbage collection. So deallocated does nothing or at maximum it's a hint to the GC to mark the denoted object for the next GC round.

I am aware, but features are therefore coupled: it will be more difficult to implement features one by one.

Paebbels, 2018-01-05:
This problem already exists in VHDL-2008:

> **VHDL-2008 LRM - 5.4.3 Allocation and deallocation of objects:**  
> ...
> NOTE—If an access value is copied to a second variable and is then deallocated, the second variable is not set to null and thus references invalid storage.

I can not find a description how access type dereference works.
I believe the LRM describes an error condition somewhere for that case.

tgingold, 2018-01-05:
> On 05 Jan 2018, at 09:41, Patrick Lehmann <notifications@github.com> wrote:
> 
> This problem already exists in VHDL-2008:
> 
> VHDL-2008 LRM - 5.4.3 Allocation and deallocation of objects:
> ...
> NOTE—If an access value is copied to a second variable and is then deallocated, the second variable is not set to null and thus references invalid storage.
> 
> I can not find a description how access type dereference works.
> I believe the LRM describes an error condition somewhere for that case.

I would say this is a different issue, as in vhdl-2008 you cannot share an access between two processes.

commit 2fcecf81 (2018-01-05): Fix typo in error message.

Fix #516
closed 2018-01-05 by tgingold