ghdl/ghdl #519: Error when importing empty vhdl file

FranzForstmayr, 2018-01-16:
Hey,

i just upgraded to ghdl-0.35, thanks for your effort!

I don't know if this is a bug, but if i want to import a empty vhdl file i got following error:
```
C:\Software\ghdl\ghdl-0.35\bin\ghdl.exe -i empty.vhdl
empty.vhdl:1:1: design file is empty (no design unit found)
C:\Software\ghdl\ghdl-0.35\bin\ghdl.exe: importation has failed due to compilation error
```

When i want to try this with ghdl-0.34:
```
C:\Software\ghdl\ghdl-0.34\bin\ghdl.exe -i empty.vhdl
empty.vhdl:1:1: design file is empty (no design unit found)
```
Of course I understand the warning, but it would be nice, when i have the chance to get further.

I know, this is pretty a uncommon way of using ghdl. I triggered it first, when i wanted to import the PoC Library,  where the common.vhdl file has no package defined.  So the vhdl file hasn't to be completely empty.

```
context Common is
	library PoC ;
	use PoC.config.all;
	use PoC.debug.all;
	use PoC.FileIO.all;
	use PoC.math.all;
	use PoC.physical.all;
	use PoC.strings.all;
	use PoC.utils.all;
	use PoC.vectors.all;
end context;
```
So this triggers

Paebbels, 2018-01-16:
The VHDL language does not allow empty files. At least one design unit (entity, architecture, package, package body, context) is required.

closed 2018-01-16 by Paebbels

FranzForstmayr, 2018-01-16:
Sorry, i've no access to the lrm yet. 
But i think you are misunderstanding me. I triggered this error when importing common.vhdl.
The common.vhdl contains a context, so the importing shouldn't fail.

Paebbels, 2018-01-16:
Contexts are a VHDL-2008 feature. You need to enable VHDL-2008 with `--std=08`.