ghdl/ghdl #522: Simulation failure due to "bound check failure"

alidebb, 2018-01-24:
Good afternoon,
I wrote a code that do the shifting of a 32bits input, and within this code I used some concatenation:

LIBRARY IEEE;
USE IEEE.std_logic_1164.ALL;
USE IEEE.numeric_std.ALL;

entity Shifter is
port(shift_lsl :in std_logic;
        shift_lsr :in std_logic;
        shift_asr :in std_logic;
        shift_ror :in std_logic;
        shift_rrx :in std_logic;
        cin       :in std_logic;
        shift_val :in std_logic_vector (4 downto 0);
        din       :in std_logic_vector(31 downto 0);     
        dout      :out std_logic_vector(31 downto 0);
        cout      :out std_logic;
        vdd       :in bit;
        vss       :in bit);
end Shifter;

architecture ArchiShifter of Shifter is
    signal res1, res2, res3, res4, res5, res6, res7 : std_logic_vector(31 downto 0):=x"00000000";
    signal carryOut: std_logic;

    begin
        --LSL & LSR & ASR
        res1 <= din(30 downto 0) & '0'      when shift_val(0)='1'  and             shift_lsl='1'                           else 
                '0' & din(31 downto 1)      when shift_val(0)='1'  and (shift_lsr='1' or (shift_asr='1' and din(31)='0'))  else
                '1' & din(31 downto 1)      when shift_val(0)='1'  and (shift_asr='1' and din(31)='1' )                    else din;
...
end ArchiShifter;

But when I wrote a TestBench, I got the following error:
../../src/ieee/numeric_std-body.v93:2098:7:@0ms:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
./shifter_tb:error: bound check failure at Shifter.vhdl:13

GHDL Version:
GHDL 0.33 (20150921) [Dunoon edition]
 Compiled with GNAT Version: 4.8
 llvm code generator
Written by Tristan Gingold.

When I sent both the code & the testbench to a friend of mine, he told me that it worked perfectly. Can you please help to figure it & solve this issue?

tgingold, 2018-01-24:
Please provide a full reproducer so that the issue could be investigated.

alidebb, 2018-01-24:
This is the TB code:
library ieee;
use ieee.std_logic_1164.all;
USE ieee.numeric_std.all;

entity Shifter_tb is
end Shifter_tb;

architecture simu of Shifter_tb is
        signal shift_lsl, shift_lsr, shift_asr, shift_ror, shift_rrx, cin : std_logic;
        signal shift_val   :std_logic_vector (4 downto 0);
        signal din         :std_logic_vector (31 downto 0);     
        signal dout        :std_logic_vector (31 downto 0);
        signal cout        :std_logic;
        signal vdd         :bit;
        signal vss         :bit;

begin

    L0: entity work.Shifter
	port map (shift_lsl, shift_lsr, shift_asr, shift_ror, shift_rrx, cin, shift_val, din, dout,cout, vdd, vss);
	process
        variable test : std_logic_vector (31 downto 0);
        constant exttest : std_logic_vector (31 downto 0) := X"FFFFFFFF";
	begin
        cin       <= '0';
        shift_val <= "00000";
        din       <= X"0FF00000";
        shift_asr <= '0';
        shift_ror <= '0';
        shift_rrx <= '0';
        shift_lsl <= '0';
        shift_lsr <= '0'; 
       	assert  dout = din report  "Variable Test = Initial Defaut"severity error;
      	wait for 1 ns;

--*************************************ASR**************************************************
            din <= X"A5A50000";
            shift_asr <= '0';
            shift_ror <= '1';
            shift_rrx <= '0';
            shift_lsl <= '0';
            shift_lsr <= '0';

            for shift in 0 to 30 looP
                shift_val <= std_logic_vector(to_unsigned(shift,5 ));
                test := din;
                wait for 1 ns;
                if shift /= 0 then
                    --"asr" équivaut à une division signé par "2^shift"
                    --l'operateur exposant n'étant pas disponible on fais une boucle qui divise par 2^shift à chaque tour de boucle   
                    for i in 0 to (shift-1) loop
                        test := std_logic_vector (shift_right(signed(test),1));
                        assert  dout = test  report "Expected Test Result  = " & integer'image(to_integer(signed(test))) & " || Dout  = " & 
                        integer'image(to_integer(signed(dout))) & " || ASR shift = " & integer'image(to_integer(unsigned(shift_val))) severity error;
                    end loop; 
                else
                    test := std_logic_vector(signed(din));      
                    assert  dout = test report "Expected Test Result  = " & integer'image(to_integer(signed(test))) & 
                    " || Dout = " & integer'image(to_integer(signed(dout))) & " ||ASR shift = " & integer'image(shift) severity error;
                end if;
                wait for 2 ns;  
            end loop;

--Test de l'extention de signe
        din       <= X"80000000";
        shift_val <= "11111";
wait for 1 ns; 
assert  dout (31 downto 0) =  exttest (31 downto 0)  report "Bit de Signe = " & integer'image ( to_integer (unsigned(din(31 downto 31)))) & " || Dout  =" & integer'image(to_integer(unsigned(dout (31 downto (31-(to_integer(unsigned(shift_val)))))))) & " || ASR shift = " & integer'image(to_integer(unsigned(shift_val))) severity error;

[... 109 lines cut ...]
        carryOut <= din(32-to_integer(unsigned(shift_val))) when shift_val /="00000" and shift_lsl='1' else 
                    din(to_integer(unsigned(shift_val))-1)  when shift_val /="00000" and (shift_lsr='1' or shift_asr='1' or shift_ror='1') else
                    din(0) when shift_rrx='1' else cin; 
        --RRX
        res6 <= cin & din(31 downto 1) when shift_rrx='1' else din;

        --ROR
        res7(31 downto (32-to_integer(unsigned(shift_val)))) <= din((to_integer(unsigned(shift_val))-1) downto 0) when shift_ror='1' and shift_val /="00000" 
                                                                else din; 
        res7((31-to_integer(unsigned(shift_val))) downto 0)  <= din(31 downto to_integer(unsigned(shift_val)))    when shift_ror='1' and shift_val /="00000" 
                                                                else din;

        dout <= res5 when shift_lsl='1' or shift_asr='1' or shift_lsr='1' else
                res6 when shift_rrx='1' else
                res7 when shift_ror='1' else
                din;
        cout <= carryOut;

end ArchiShifter;

In the terminal I do the following commands:
ghdl -a Shifter.vhdl
ghdl -a Shifter_tb.vhdl
ghdl -e Shifter_tb
ghdl -r Shifter_tb --vcd=Shifter.vcd

Paebbels, 2018-01-24:
The used GHDL version is very old (). Can you update GHDL to a newer version? The current version is [GHDL v0.35]()

alidebb, 2018-01-24:
Can you tell me the procedure in order to update it?
I already downloaded this file "ghdl-0.35-llvm-3.8-ubuntu14.tgz" but didn't know how to install

Paebbels, 2018-01-24:
You can either compile GHDL from sources e.g. with LLVM backend - that's quite easy on Debian, Ubuntu and Co. Or you download a matching file and extract the content. I prefer to build GHDL.

See this guide: http://ghdl.readthedocs.io/en/latest/building/llvm/GNULinux-GNAT.html

alidebb, 2018-01-25:
I finally succeeded in rebuilding GHDL, I now have this version:
GHDL 0.35 (tarball) [Dunoon edition]
 Compiled with GNAT Version: 4.8
 mcode code generator
Written by Tristan Gingold.

tgingold, 2018-01-25:
See lines 53 and 55: the length of the target is variable, while the length of the expression can be the length of din.

closed 2018-02-06 by tgingold