ghdl/ghdl #524: ASSERT_FAILURE

sebunger, 2018-02-02:
I'm on Ubuntu 17.10 with libgnat-4.6 from trusty and GHDL 0.35 from the tar-ball.

> ******************** GHDL Bug occured ****************************
> Please report this bug on https://github.com/tgingold/ghdl/issues
> GHDL release: 0.35 (v0.35) [Dunoon edition]
> Compiled with GNAT Version: 4.6
> Target: x86_64-pc-linux-gnu
> In directory: /home/seb/Documents/work/SPI2APB/
> Command line:
> ghdl -a --std=08 spi2apb.vhdl
> Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
> Exception information:
> Exception name: SYSTEM.ASSERTIONS.ASSERT_FAILURE
> Message: iirs.adb:5289
> Call stack traceback locations:
> 0x7f4cc1f6b44a 0x4b803b 0x674861 0x64e570 0x64ff2e 0x64fe5d 0x650159 0x650c1e 0x650c82 0x65100a 0x651369 0x67bd5d 0x681be8 0x531a9e 0x641a97 0x63f716 0x4c7951 0x408777 0x7f4cc17741bf 0x407d17 0xfffffffffffffffe
> ******************************************************************

[spi2apb.vhdl.gz](https://github.com/ghdl/ghdl/files/1691070/spi2apb.vhdl.gz)

diogratia, 2018-02-03:
In process sync
```vhdl
    reg_in <= reg_reset when (reset_n => '0') else reg_out;
```
The parentheses aren't needed, but the "=>" is erroneous and should be "=".

That's sufficient to cause the crash.

In process clocking
```vhdl
          if (reg_in.running) then
         -- reg_out.sclk <= "0" when reg_in else "1"; -- string literals 
         reg_out.sclk <= '0' when reg_in.sclk else '1'; --  no ?? for reg_type
```
"0" and "1" are string literals and not of the type of reg_out.sclk (std_logic).
There is no  "??" operator defined for type reg_in (the condition operator returns a boolean)

The element evaluated in the third form  is a guess. reg_in.sclk is type std_logic and the "??" operator is defined for it in package std_logic_1164.

The two above problems are also sufficient to produce the error.

What ghdl is doing wrong is not expressing error messages for them, instead generating the message you see. 

Make the changes and your rtl architecture of spi2apb analyzes.

the "??" operator is defined for type bit in std.standard and for std_ulogic in ieee.std_logic_1164. There are rules for it being implicitly applied. See IEEE Std 1076-2008 9.2.9 Condition operator. Both after the if in an if statement and after the when in a conditional assignment are included.

And the last bit of cheery news. The sequential conditional signal assignment and condition operator might not be widely supported for synthesis.

(GHDL 0.36-dev (v0.35-64-g4d52610))

sebunger, 2018-02-03:
Hi Dio Gratia,

Many thanks for your response. I have also found those issues and got it to
analyse, but having an error message would be better. ;-)

Cheers,
Seb

In process sync

<https://github.com/ghdl/ghdl/issues/524#issuecomment-362783823>, or mute
the thread

commit 2f647d9f (2018-02-06): Improve handling of implicit condition.

Avoid a crash on aggregate (fix #524)
Improve error message if operator is not found
Don't print operator with --reprint.
closed 2018-02-06 by tgingold