ghdl/ghdl #525: GHDL 0.35 regresssion: no errors for illegal attributes

brimdavis, 2018-02-04:
In the code example below. GHDL 0.33 identifies the illegal attributes:
`GHDL 0.33 (20150921) [Dunoon edition]
 Compiled with GNAT Version: GPL 2014 (20140331)
 mcode code generator
`
GHDL 0.35 does not issue any errors:

   `GHDL 0.35 (tarball) [Dunoon edition]
   Compiled with GNAT Version: GPL 2017 (20170515-63)
   mcode code generator

  C:\Projects\vivado_tests\illegal_x_interface>ghdl -s counter_illegal_attribute.vhd

  C:\Projects\vivado_tests\illegal_x_interface>`

The first two errors are from incorrect signal attributes in the architecture using port names, the third error is an attribute on a non-existent signal.

tgingold, 2018-02-05:
Yes, this was done to avoid errors in X...x files.  The errors show if you use --std=93.
I will turn that into a warning.

tgingold, 2018-02-06:
I will always generate an error for non-existing names, and by default simply emit a warning when trying to specify the attribute for a port from the architecture.  So the output should be:
counter_illegal_attribute.vhd:31:38:warning: attribute for port "rst" must be specified in the entity [-Wspecs]
counter_illegal_attribute.vhd:32:38:warning: attribute for port "rst" must be specified in the entity [-Wspecs]
counter_illegal_attribute.vhd:37:38:error: no "no_such" for attribute specification

closed 2018-02-06 by tgingold

diogratia, 2018-02-06:
The point the fellow opening the bug was making is demonstrated in your warnings (and emojii are nothing but annoying, particularly here). The various standards use the word must or shall where both have mandatory meaning.

>... An attribute specification for an attribute of a procedure, a function, a type, a subtype, an object (i.e., a constant, a file, a signal, or a variable), a component, literal, unit name, group, or a labeled entity must appear within the declarative part in which that procedure, function, type, subtype, object,component, literal, unit name, group, or label, respectively, is explicitly or implicitly declared. 

Substituting shall for must in -2008 prior to which the explanation of must had normative meaning:

>Additionally, the word "must" is used to indicate mandatory weight. This word is preferred over the more common "shall," as "must" denotes a different meaning to different readers of this standard.  
  
>>a.  To the developer of tools that process VHDL, "must" denotes a requirement that the standard imposes. The resulting implementation is required to enforce the requirement and to issue an error if the requirement is not met by some VHDL source text.  

>>b.  To the VHDL model developer, "must" denotes that the characteristics of VHDL are natural consequences of the language definition. The model developer is required to adhere to the constraint implied by the characteristic.  

>>c.  To the VHDL model user, "must" denotes that the characteristics of the models are natural consequences of the language definition. The model user can depend on the characteristics of the model implied by its VHDL source text. 

And warnings violate two of these three precepts.

In -2008 the three precepts are removed but we also find in 1.3.4 Front matter, examples, notes, references, and annexes that the prior subclauses are now held [informative and not normative](http://standards.ieee.org/develop/policies/opman/sect6.html#6.4.1):

>Prior to this subclause are several pieces of introductory material; following Clause 24 are some annexes and an index. The front matter, annexes (except Annex B), and index serve to orient and otherwise aid the user of this standard, but are not part of the definition of VHDL; Annex B, however, is normative.

And this has two implications. Removal of the three precepts as normative is accordance with IEEE style manuals See the [2014 IEEE-SA Standards Style Manual](https://development.standards.ieee.org/myproject/Public/mytools/draft/styleman.pdf), 10.1 Normative and informative clauses:  
>**Interspersed normative and informative text is not allowed.** As such, neither clauses nor subclauses shall be  labeled  as  informative.  Contact  IEEE-SA content  publishing  staff  early  in  the  process  if  there  are  questions as to whether material in the draft should be labeled as normative or informative. 

and 

10.2.2 Shall, should, may, and can 

>Shall, should, may, and can are defined in [6.4.7](standards.ieee.org/develop/policies/opman/sect6.html#6.4.7) of the  IEEE-SA Standards Board Operations Manual.  

IEEE-SA Standards Board Operation Manual, 6.4.7 Shall, should, may, and can:

>The word *shall* indicates mandatory requirements strictly to be followed in order to conform to the standard and from which no deviation is permitted (*shall* equals i*s required to*).  
>  
>The word *should* indicates that among several possibilities one is recommended as particularly suitable, without mentioning or excluding others; or that a certain course of action is preferred but not necessarily required (*should* equals *is recommended that*).  
>  
>The word *may* is used to indicate a course of action permissible within the limits of the standard (*may* equals *is permitted to*).  
>  
>The word *can* is used for statements of possibility and capability, whether material, physical, or causal (*can* equals *is able to*).

(This change went into effect after 2005 (the shall specification not evinced in the 2005 style manual) and before publication of of the -2008 standard revision).

You shouldn't be able to interpret an ill-formed description (that does not adhere to a requirement) as anything other than synonymous with the informative description of **error**.

Further it should not result in a design unit being entered into a reference library, -2008 13.5 Order of analysis:

>If any error is detected while attempting to analyze a design unit, then the attempted analysis is rejected and has no effect whatsoever on the current working library.

I'd consider your generation of warnings while producing a library entry/update for a design unit containing a description not meeting a requirement not in adherence with any revision of the standard.

For purposes of looking ahead for multiple errors instead of aborting on encountering such an error it would seem possible to add another error type (there's an implication of syntax errors being separate for the same reason). Any errors of either category being sufficient to stop a design unit from updating or creating a working library entry.

You could also note a dearth of informative illegal examples for such requirements in the VHDL standard.

Effectively the IEEE-SA Standards Board Operation Manual has become another unlisted normative reference (noting 2. Normative references has never contained a reference to IEEE Std 610.12-1990 IEEE Standard Glossary of Software Engineering Terminology or it's later derivatives, See https://standards.ieee.org/findstds/standard/610.12-1990.html and https://standards.ieee.org/findstds/standard/24765-2017.html, the shall substitution for must comes from determining a commonly acceptable format and terminology for international standards, IEEE Std 610.12-1990 can be essential in unambiguously interpreting the VHDL standard whose word usage derives from the same era).

tgingold, 2018-02-07:
@diogratia we all agree with you, but the whole of `-frelaxed-rule` is to bend the rules so that existing design are accepted.  By existing design, I mean libraries provided by vendors that are analyzed without any error messages by the commercial simulators.

When you are using ghdl with `--std=93` or `--std=08` and without `-frelaxed-rules`, LRM must be followed (except maybe when the LRM is self-contradicting or obviously wrong).

brimdavis, 2018-02-07:
@diogratia "And warnings violate two of these three precepts"
I believe Tristan's post about warnings was describing the behavior only when relaxed rules are in effect.

Inspecting commit f262f19  , it appears to me that the new behavior will be:

With default settings:
   - attributes on non-existent signals will be an error
   - misplaced port attributes will be an error

With relaxed rules in effect ( -frelaxed-rules or --std=93c ):
   - attributes on non-existent signals will be an error
   - misplaced port attributes will be an a warning (of type Warnid_Specs, now on by default) 

-Brian

tgingold, 2018-02-08:
That's correct.

commit f262f19d (2018-02-06): Refine error message for attribute specification.

Fix #525