ghdl/ghdl #526: GHDL searches for library 'synopsys'

Paebbels, 2018-02-09:
I think some thing went wrong in one of the last commits. GHDL is searching for library `synopsys`, but this is an IEEE library flavor, not a real library.

```
C:\git\GitHub\Paebbels\PoC [paebbels/master ≡ +0 ~20 -0 | +21 ~38 -2 !]> .\tools\precompile\compile-osvvm.ps1 -GHDL -Debug
Pre-compiling OSVVM's simulation libraries for GHDL
--------------------------------------------------------------------------------
  cd C:\git\GitHub\Paebbels\PoC\temp\precompiled\ghdl
  Invoke-Expression & 'C:\Tools\GHDL\0.36-dev-gnatgpl32-mcode\lib\vendors\compile-osvvm.ps1' -All -Source C:\git\GitHub\Paebbels\PoC\lib\osvvm -Output C:\git\GitHub\Paebbels\PoC\temp\precompiled\ghdl -Verbose:$True -Debug:$True
Vendor directory 'C:\git\GitHub\Paebbels\PoC\temp\precompiled\ghdl' already exists.
Compiling library 'osvvm' ...
  Creating library osvvm ...
    mkdir C:\git\GitHub\Paebbels\PoC\temp\precompiled\ghdl/osvvm/v08
    cd C:\git\GitHub\Paebbels\PoC\temp\precompiled\ghdl/osvvm/v08
  Analyzing package file 'C:\git\GitHub\Paebbels\PoC\lib\osvvm\NamePkg.vhd'
    & 'C:\Tools\GHDL\0.36-dev-gnatgpl32-mcode\bin\ghdl.exe' -a -fexplicit -frelaxed-rules --mb-comments --warn-binding --ieee=synopsys --no-vital-checks --std=08 -PC:\git\GitHub\Paebbels\PoC\temp\precompiled\ghdl --work=osvvm C:\git\GitHub\Paebbels\PoC\lib\osvvm\NamePkg.vhd 2>&1
  C:\Tools\GHDL\0.36-dev-gnatgpl32-mcode\bin\ghdl.exe:warning: library synopsys does not exists for v08
  C:\Tools\GHDL\0.36-dev-gnatgpl32-mcode\bin\ghdl.exe: cannot find "std" library
...
...
...
```

**The important messages from compiling OSVVM:**
>  C:\Tools\GHDL\0.36-dev-gnatgpl32-mcode\bin\ghdl.exe: warning: library synopsys does not exists for v08
>  C:\Tools\GHDL\0.36-dev-gnatgpl32-mcode\bin\ghdl.exe: cannot find "std" library

-----------------------------

**When looking at the compilation log of OSVVM, I see these messages:**
```
Running compile-libraries.ps1 -Compile ...
--------------------------------------------------------------------------------
Compiling VHDL Libraries...
Preparing...

Start compilation...
Compiling libraries for VHDL-87
  Compiling library 'std'...
    Creating directory 'C:\git\GitHub\Paebbels\ghdl\build\mcode\lib\v87\std'.
    file: v87\textio.v87
    file: v87\textio_body.v87
  Compiling library 'ieee'...
    Creating directory 'C:\git\GitHub\Paebbels\ghdl\build\mcode\lib\v87\ieee'.
    file: v87\std_logic_1164.v87
      C:\git\GitHub\Paebbels\ghdl\build\mcode\ghdl.exe:warning: library ieee does not exists for v87
    file: v87\std_logic_1164_body.v87
      C:\git\GitHub\Paebbels\ghdl\build\mcode\ghdl.exe:warning: library ieee does not exists for v87
...
```

**Here is the important message:**
> C:\git\GitHub\Paebbels\ghdl\build\mcode\ghdl.exe: warning: library ieee does not exists for v87

**Questions:**
1. Why is such an important message just a warning?
2. How can IEEE not exist?

Paebbels, 2018-02-09:
Here is the IEEE library compilation log in debug mode:

```
Running compile-libraries.ps1 -Compile ...
--------------------------------------------------------------------------------
Compiling VHDL Libraries...
Preparing...
  Directory 'C:\git\GitHub\Paebbels\ghdl\build\mcode\lib' already exists.
  cd C:\git\GitHub\Paebbels\ghdl\build\mcode\lib
  Cleaning up directory...

Start compilation...
Compiling libraries for VHDL-87
  Compiling library 'std'...
    Creating directory 'C:\git\GitHub\Paebbels\ghdl\build\mcode\lib\v87\std'.
    file: v87\textio.v87
      Patching file for VHDL-87
        Get-Content "C:\git\GitHub\Paebbels\ghdl\libraries\std\textio.vhdl" -Encoding Ascii `
          | Format-VHDLSourceFile -Version "87" `
          | Out-File "textio.v87" -Encoding Ascii
      Analyzing file 'textio.v87'
        C:\git\GitHub\Paebbels\ghdl\build\mcode\ghdl.exe -a -C --bootstrap --std=87 --work=std textio.v87 2>&1
    file: v87\textio_body.v87
      Patching file for VHDL-87
        Get-Content "C:\git\GitHub\Paebbels\ghdl\libraries\std\textio_body.vhdl" -Encoding Ascii `
          | Format-VHDLSourceFile -Version "87" `
          | Out-File "textio_body.v87" -Encoding Ascii
      Analyzing file 'textio_body.v87'
        C:\git\GitHub\Paebbels\ghdl\build\mcode\ghdl.exe -a -C --bootstrap --std=87 --work=std textio_body.v87 2>&1
  Compiling library 'ieee'...
    Creating directory 'C:\git\GitHub\Paebbels\ghdl\build\mcode\lib\v87\ieee'.
    file: v87\std_logic_1164.v87
      Patching file for VHDL-87
        Get-Content "C:\git\GitHub\Paebbels\ghdl\libraries\ieee\std_logic_1164.vhdl" -Encoding Ascii `
          | Format-VHDLSourceFile -Version "87" `
          | Out-File "std_logic_1164.v87" -Encoding Ascii
      Analyzing file 'std_logic_1164.v87'
        C:\git\GitHub\Paebbels\ghdl\build\mcode\ghdl.exe -a -C "-P../std" --std=87 --work=ieee std_logic_1164.v87 2>&1
      C:\git\GitHub\Paebbels\ghdl\build\mcode\ghdl.exe:warning: library ieee does not exists for v87
    file: v87\std_logic_1164_body.v87
      Patching file for VHDL-87
        Get-Content "C:\git\GitHub\Paebbels\ghdl\libraries\ieee\std_logic_1164_body.vhdl" -Encoding Ascii `
          | Format-VHDLSourceFile -Version "87" `
          | Out-File "std_logic_1164_body.v87" -Encoding Ascii
      Analyzing file 'std_logic_1164_body.v87'
        C:\git\GitHub\Paebbels\ghdl\build\mcode\ghdl.exe -a -C "-P../std" --std=87 --work=ieee std_logic_1164_body.v87 2>&1
      C:\git\GitHub\Paebbels\ghdl\build\mcode\ghdl.exe:warning: library ieee does not exists for v87
...
...
```

brimdavis, 2018-02-11:
Looks like compile-libraries.ps1 is out of sync with the library path changes of 7ef79a5 
```
-IEEE87_DIR:=$(LIB87_DIR)/ieee
-SYN87_DIR:=$(LIB87_DIR)/synopsys
+IEEE87_DIR:=$(LIBDST_DIR)/ieee/v87
+SYN87_DIR:=$(LIBDST_DIR)/synopsys/v87
```

closed 2018-02-12 by tgingold