ghdl/ghdl #529: impure function call with process(all) fails

putsplut, 2018-02-17:
With the VHDL-2008 process(all), the code which determines the sensitivity list trips up on calls to impure functions, raising a TYPES.INTERNAL_ERROR : canon.adb:547.  A pure function works, and an impure function with a non "all" sensitivity list works.  I have a small example attached.

[sample_impure.txt](https://github.com/ghdl/ghdl/files/1733956/sample_impure.txt)

Paebbels, 2018-02-18:
What GHDL version are you using?

putsplut, 2018-02-18:
GHDL 0.36-dev (v0.35-109-g900f7cf3) [Dunoon edition]
 Compiled with GNAT Version: 7.2.0
 llvm code generator

The git tag is: 900f7cf3c4bcc331f460c278b09af2efd74ebff7

tgingold, 2018-02-18:
I will push a fix for it shortly; I am not yet sure if it handles all the complex cases.

commit c95389e1 (2018-02-18): sem_subprogram_body: handle all_sensitized_state for all subprograms.

fix #529
closed 2018-02-18 by tgingold