ghdl/ghdl #530: casting of sliced port in a module instantiation fails

putsplut, 2018-02-17:
A small example is attached.  When instantiating a module, it is sometimes convenient to cast portions of an input or output port.  In the example, I cast both directions.  The cast on the output raises an exception TYPES.INTERNAL_ERROR : trans.adb:619.  The input port cast works fine, and a cast on an entire port works fine.
Note: support for casting or slices of ports has spotty results across commercial tools.  The example would work in Riviera-PRO and Modelsim.  Riviera has had issues with a function call being assigned to a sliced input (even when the function is just a shorthand cast).  Vivado has had issues with (I think) the sliced output casts, but Quartus accepts it.

[sample_slice_ports.vhd.txt](https://github.com/ghdl/ghdl/files/1733960/sample_slice_ports.vhd.txt)

putsplut, 2018-02-18:
Following up on this, it appears that casts on the input ports do not raise the error, but also do not function properly.  On a std_logic_vector() cast, I end up with 'U' values instead of the actual data.  This may then be the bigger issue because it compiled but was incorrect in behavior.

Paebbels, 2018-02-18:
What GHDL version are you using?

putsplut, 2018-02-18:
GHDL 0.36-dev (v0.35-109-g900f7cf3) [Dunoon edition]
 Compiled with GNAT Version: 7.2.0
 llvm code generator

I did not write an example submodule architecture as I wanted the smallest example I could produce.  I've extended the example but have not yet reproduced my observed problems on input side casting.  I'll attempt to reproduce that with a small example and will provide that.

putsplut, 2018-02-18:
The input side casting is a different issue.  I will file a new bug report for it.
I'll leave this bug as valid since the output side casting is still an issue and is reproducible with the earlier snippit of code.

commit 74663a24 (2018-02-18): translate: handle anonymous types in formal; handle multiple conversions.

Fix #530
closed 2018-02-18 by tgingold