ghdl/ghdl #532: Another issue with slicing input ports

putsplut, 2018-02-20:
GHDL release: 0.36-dev (v0.35-115-g84f88ca2) [Dunoon edition]

A concatenated signal assigned to a sliced input port causes the following exception.
Example VHDL snippit included.

Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : trans-chap7.adb:480 access check failed

By the way, thanks for the tremendous effort you've put in to this tool.  I was blown away by the 1 day turn-around on the last bugs I found.  Amazing!

[sample_sliced2.vhd.txt](https://github.com/ghdl/ghdl/files/1739006/sample_sliced2.vhd.txt)
```vhdl
library ieee;
   use ieee.std_logic_1164.all;

entity sliced_port is
port (
   clk   : in  std_logic;
   reset : in  std_logic;
   arg_a : in  std_logic_vector(15 downto 0)
);
end sliced_port;

library ieee;
   use ieee.std_logic_1164.all;

entity submodule is
port (
   clk : in  std_logic;
   arg : in  std_logic_vector(15 downto 0);
   res : out std_logic_vector(15 downto 0)
);
end submodule;

architecture rtl of sliced_port is
begin
   sub_module : entity work.submodule
   port map (
      clk => clk,
      --arg => x"00" & x"00", -- this works
      arg(15 downto 0) => x"00" & x"00", -- this fails
      res => OPEN
   );
end rtl;

architecture rtl of submodule is
begin
   res <= arg;
end rtl;
```

tgingold, 2018-02-20:
Thank you for your nice comment.  I will try to fix these issues.

The complexity behind the associations is surprisingly high (conversions, individual associations, ...), and very few people use that!

tgingold, 2018-02-20:
Thank you also for your bug reports, that's very useful!

commit 472552e5 (2018-02-20): translation: for port assiation by expression: handle formal before actual.

Fix #532
closed 2018-02-20 by tgingold