ghdl/ghdl #533: Trying a new ghdl invalidates timestamps of xilinx vendor libraries

svenn71, 2018-02-20:
When building and installing a new ghdl from github, the std_logic_1164 gets rebuilt.
When trying to simulate a project with xilinx primitives from unisim, I get message that std_logic_1164 is newer than the vendor libraries, and process stops.

How can I test a new ghdl build without the time consuming recompile of all xilinx libraries.
When there has been no logic change to any of the provided libraries and everything is a matter of timestamps, I guess I should put back the timestamp on the libraries which came out of the ghdl make/install procedure, but where do I do that?

tgingold, 2018-02-21:
The short answer is no, you cannot really do that.

However, if you are using the mcode backend, recompiling the xilinx libraries is quick.  So I suppose you don't use mcode.
If you are using the llvm or gcc backend, then it is slow and painful. However, the new ghdl may fix some issues in the vendor libraries, or may generate code that isn't compatible with the previous installation.  Hence I recommend to recompile.  If you feel lucky, you can simply rebuild ghdl and install only the binaries (ghdl and ghdl1-llvm) overwriting the previous installation.  Look at the install targets in Makefile.  Maybe we should slightly change them to ease this kind of installation.

closed 2018-02-21 by tgingold

Paebbels, 2018-02-21:
The compilation script has different options, e.g. to:
* `SKIP_LARGE_FILES`
* compile only parts of the ISE/Vivado libraries:
  * `COMPILE_UNISIM`
  * `COMPILE_UNIMACRO`
  * `COMPILE_UNIFAST`
  * `COMPILE_SECUREIP`

You can see all options when you cann the precompile script with `--help`.