ghdl/ghdl #535: Problem instantiating generic package within another generic package

kraigher, 2018-02-24:
```vhdl
package pkg1 is
  generic (
    type value_t);
end package;

package pkg2 is
  generic (
    type value_t);

  package pkg1_inst is new work.pkg1 generic map (value_t => value_t);
  use pkg1_inst.all;
end;
```

```
$ ghdl -a --std=08 bug.vhd 
bug.vhd:10:62:error: cannot find a "=" declaration for type "value_t"
ghdl:error: compilation error
```

Paebbels, 2018-02-24:
It sounds like a problem when generic values are chained from generic package to generic package.

I'll classify this as a missing part of one of the lately implemented VHDL-2008 features.
@tgingold Please change the label, if it should already be implemented to bug.

tgingold, 2018-02-25:
Now fixed, but I haven't tested beyond your reproducer!

commit 168286ce (2018-02-25): Also consider interface function for implicit operators association.

Fix #535
closed 2018-02-25 by tgingold