ghdl/ghdl #537: Problem installing GHDL 0.36 DEV

ellorens, 2018-03-02:
Hi there, we are having problems with the  ghdl actualization, we think the installation is correctly done, but when we try to execute the new ghdl version we fail. It is maybe because some issue with the executable? We are checking it and we think it is not being upgraded. 

Here is our console log, thank you in advance for all your work, it's amazing. 

[ghdl_036problem.txt](https://github.com/ghdl/ghdl/files/1774969/ghdl_036problem.txt)

Paebbels, 2018-03-02:
What is the error message you get, when you execute GHDL?

tgingold, 2018-03-02:
How have you configured ghdl ?

It looks like you have configured it for gcc, but you haven't built gcc (the ghdl1 part).

Don't forget the README.md, and the simplest way is to configure for mcode.

ellorens, 2018-03-05:
We had build gcc also, we are using code coverage, that's the reason we need the gcc installation mode, but i'm going to check it again, it's very probably our fault, i'm going to reinstall everything and give you the feedback.

Thank you again.

ellorens, 2018-03-05:
When we try to compile ghdl with gcc backend we get this strange error:

```ellorens@debian~/ghdl $ make ghdllib
make -f ./libraries/Makefile.inc LIBSRC_DIR="./libraries" LIBDST_DIR="lib/ghdl" enable_openieee="false" LN="ln -s" CP="cp" MKDIR="mkdir" GHDL="/usr/local/bin/ghdl" GHDL_FLAGS=" " vhdl.libs.all libs.vhdl.standard
make[1]: Entering directory '/home/ellorens/ghdl'
make[1]: Nothing to be done for 'vhdl.libs.all'.
cd lib/ghdl/std/v93; /usr/local/bin/ghdl --bootstrap-standard  --std=93
/usr/local/bin/ghdl:error: unknown command '--bootstrap-standard', try --help
libraries/Makefile.inc:351: recipe for target 'lib/ghdl/std/v93/std_standard.o' failed
make[1]: *** [lib/ghdl/std/v93/std_standard.o] Error 1
make[1]: Leaving directory '/home/ellorens/ghdl'
Makefile:236: recipe for target 'libs.vhdl.gcc' failed
make: *** [libs.vhdl.gcc] Error 2
```
This is my actual GHDL version

`ellorens@debian~/ghdl $ ghdl --version
GHDL 0.36-dev (tarball) [Dunoon edition]
 Compiled with GNAT Version: 4.9.2
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2015 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
`

ellorens, 2018-03-05:
I have reinstalled all  and now is working, maybe i lost step.

tgingold, 2018-03-05:
> On 05 Mar 2018, at 16:28, ellorens <notifications@github.com> wrote:
> 
> I have reinstalled all and now is working, maybe i lost step.

I think you forget to install the gcc built with ghdl, before installing the libraries.
As a result, make tried to compile the libraries for gcc but using the old mcode-based ghdl.

closed 2018-03-07 by ellorens