ghdl/ghdl #538: GHW format GHLD 0.36-dev

ellorens, 2018-03-05:
Is the GHW format for gtkwave actually working? I have just updated my ghdl and when i try to generate a ghw waveform for GTKWAVE (v3.3.86) i got this error:

`ghw_read_base: error in section HIE             
Error in ghw file '/home/ellorens/git/VHDL/tb/vunit_out/test_output/element_tb.test_0_c440150874ec379b64755bd952/ghdl/wave.ghw'.`

tgingold, 2018-03-05:
Yes, it is supposed to work.  v3.3.86 is recent enough.

Can you file the ghw file ?

ellorens, 2018-03-07:
Here you have, it has to be related to vhdl 2008 and unconstrained arrays, i can compile the .vhd and the testbench, but when i try to simulate the design with ghw format all crashes.

[wave.ghw.zip](https://github.com/ghdl/ghdl/files/1789317/wave.ghw.zip)

ellorens, 2018-03-07:
This is the error gotten when i run the test:
`ghw_read_base: error in section HIE
Error in ghw file '/home/git/tb/vunit_out/test_output/test_0_c440150874ec37692ae4075671e59b64755bd952/ghdl/wave.ghw'.
`
Here is my module, with the arrays unconstrained declared in the types_declaration_pkg.vhd
**Module**:
```entity computacional_element is
  generic(
    size_data : integer := 24;
    size_test : integer := 2;
    size_tf       : integer := 18
  );
  Port (
    data_test : in typea_coretest(0 to size_test)(size_data downto 0);
    data_test_o : out typea_coretest(0 to size_test)(size_data downto 0)
    );
end computacional_element;
```

**types_declaration_pkg.vhd:**

```
package types_declaration_pkg is
  type typea_coretest is array (natural range <>) of std_logic_vector ;   
end types_declaration_pkg;
```

This compiles, but when i try to simulate it, gtkwave dont accept it.

Paebbels, 2018-12-31:
@tgingold is this issue solved or does it need further work?

Xiretza, 2019-02-18:
This is still an issue with f8437ed8 and GTKWave `v3.3.94`.

arcturus140, 2019-03-20:
In my case GTKWave does not crash but it exit in error:
```
GTKWave Analyzer v3.3.66 (w)1999-2015 BSI

get_nbr_elements: unhandled type 31
```

A simple test without GTKWave involved is to use `ghwdump`. It throws the same error:
```
ghwdump top.ghw
get_nbr_elements: unhandled type 31
```

This is related to unconstrained record. GHDL can run the testbench correctly, but does not create a complete GHW dump file. 

## test case

<details>
<summary>MWE used to illustrates the issue: </summary>

    library ieee;                                             
       use ieee.std_logic_1164.all;   
       use ieee.numeric_std.all;      

    entity top is
    end entity ; 

    architecture arch_top of top is

    type tUNCONST is record
       A : std_logic_vector;
    end record;

    signal constrA : tUNCONST(A(3 downto 0));

    begin

    end architecture ;

</details>

1138-4EB, 2019-03-20:
> v3.3.66

Please, try a more recent version. The latest is v3.3.99.

arcturus140, 2019-04-06:
forgot about this issue. Using the latest version solved my problem. thank you

closed 2019-04-07 by Paebbels