ghdl/ghdl #539: Packages with same file name in different subdirectories results in compilation error

Zagitta, 2018-03-07:
I've recently run into an issue where the following folder structure results in a GHDL compilation error:
```
.
├── pkg1
│   └── pkg.vhd
├── pkg2
│   └── pkg.vhd
└── top.vhd

```
The two pkg.vhd files each contain an empty package declaration with different package names and top.vhd contains two use statements for the two packages and an empty top entity.

The ghdl commands issued and output are as following:
```
$ ghdl -i pkg1/pkg.vhd pkg2/pkg.vhd top.vhd 
$ ghdl -m top
analyze pkg1/pkg.vhd
analyze pkg2/pkg.vhd
analyze top.vhd
elaborate top
pkg.o:(.bss+0x0): multiple definition of `work__pkg2__ELABORATED'
pkg.o:(.bss+0x0): first defined here
pkg.o:(.data.rel.ro+0x0): multiple definition of `work__pkg2__RTI'
pkg.o:(.data.rel.ro+0x0): first defined here
pkg.o: In function `work__pkg2__ELAB_SPEC':
pkg.vhd:(.text+0x0): multiple definition of `work__pkg2__ELAB_SPEC'
pkg.o:pkg.vhd:(.text+0x0): first defined here
e~top.o: In function `work__pkg1__ELAB_BODY':
e~top:(.text+0x12): undefined reference to `work__pkg1__ELABORATED'
e~top:(.text+0x1c): undefined reference to `work__pkg1__ELABORATED'
e~top:(.text+0x22): undefined reference to `work__pkg1__ELAB_SPEC'
top.o: In function `work__top__PKG_ELAB':
top.vhd:(.text+0x1a): undefined reference to `work__pkg1__ELABORATED'
collect2: error: ld returned 1 exit status
ghdl:error: compilation error
```
```
$ ghdl -v
GHDL 0.35 (v0.35) [Dunoon edition]
 Compiled with GNAT Version: 7.3.0
 GCC back-end code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2015 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

A full reproduction of the issue can be found here: https://github.com/Zagitta/ghdl-bug
Simply execute `./run.sh`

Mercotui, 2019-05-04:
Hello, I have noticed that this is also the case for files with different extensions, for example I learned the habit of calling my testbenches after the entity they test, with a different extension:
the entity `example.vhd` and its test bench `example.vht`.

Of-course I'll change this habit for now, but might I suggest that you add a hash of the full file path to the generated object files?

Or at least a sanity check on the file names, to clarify the error. As this is my first time using ghdl, I was surprised to see linking errors :D

tgingold, 2019-05-05:
Adding a hash would prevent from moving files.
But adding a sanity check is doable.

Thanks for the suggestion.

closed 2026-08-28 by tgingold