ghdl/ghdl #543: Indexing an array attribute

vicencb, 2018-03-17:
Hello,
the following code raises an internal error in ghdl:
```vhdl
entity ent is end entity;
architecture arch of ent is
  attribute att :integer_vector;
  constant const :integer := 1;
  attribute att of const:constant is (2, 3);
begin
  assert false
  report "const'att(0) is " & integer'image(const'att(0))
  severity note;
end architecture;
```
When compiled with `ghdl -a --std=08 ent.vhd` it reports:
```
build_constant: cannot handle IIR_KIND_AGGREGATE (ent.vhd:5:38)

******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.36-dev (v0.35-133-gcb682caa) [Dunoon edition]
Compiled with GNAT Version: 7.3.0
Target: aarch64-unknown-linux-gnu
In directory: /tmp/
Command line:
/usr/lib/gcc/aarch64-unknown-linux-gnu/7.3.0/ghdl1 --std=08 -P/usr/lib/ghdl/ieee/v08/ -P/usr/lib/ghdl/ -quiet -o ent.s ent.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : errorout.adb:246
******************************************************************

raised TYPES.INTERNAL_ERROR : errorout.adb:246
ghdl:error: compilation error
```
When compiled with `ghdl -a ent.vhd` it reports:
```
ent.vhd:3:18:error: no declaration for "integer_vector"
ent.vhd:3:13:error: type of attribute "att" cannot be error
check_signal_type: cannot handle IIR_KIND_ERROR (*std_standard*:1:1)

******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.36-dev (v0.35-133-gcb682caa) [Dunoon edition]
Compiled with GNAT Version: 7.3.0
Target: aarch64-unknown-linux-gnu
In directory: /tmp/
Command line:
/usr/lib/gcc/aarch64-unknown-linux-gnu/7.3.0/ghdl1 -P/usr/lib/ghdl/ieee/v93/ -P/usr/lib/ghdl/ -quiet -o ent.s ent.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : errorout.adb:246
******************************************************************

raised TYPES.INTERNAL_ERROR : errorout.adb:246
ghdl:error: compilation error
```
The second error message above does not make much sense.

If the code is fixed to work with vhdl pre-08:
```vhdl
entity ent is end entity;
architecture arch of ent is
  type integer_vector is array (natural range<>) of integer;
  attribute att :integer_vector;
  constant const :integer := 1;
  attribute att of const:constant is (2, 3);
begin
  assert false
  report "const'att(0) is " & integer'image(const'att(0))
  severity note;
end architecture;
```
then it works fine with `ghdl -a ent.vhd` but fails in the same way with `ghdl -a --std=08 ent.vhd`.

Regards,
&nbsp;&nbsp;Vicenç.

Paebbels, 2018-03-17:
I suspect attribute support is not fully implemented in GHDL.

`integer_vector` was added in VHDL-2008, thus the error message in VHDL-93 mode. Regardless of the missing `integer_vector` type, it should not crash.

vicencb, 2018-03-17:
The exact same behaviour is obtained if using a record type for the attribute, instead of an array type.

tgingold, 2018-03-17:
Currently evaluation at analyze time of locally static aggregates is not implemented.  I will try to add it.

vicencb, 2018-03-17:
The aim of this bug report is to help improve ghdl.
Currently, I have no requirements for this to work.
So, no need to spend time on it if there are higher priority issues/features.

commit c55c477f (2021-09-18): vhdl-evaluation.adb: handle iir_kind_aggregate in build_constant.  Fix #543
closed 2021-09-18 by tgingold