ghdl/ghdl #545: Accessing intermediate representations?

philtomson, 2018-03-23:
The Yosys Open Synthesis suite is making a lot of progress towards an open source synthesis tool that can target some FPGAs (currently some Xilinx 7 and Lattice ICE FPGAs more info here: https://github.com/YosysHQ/yosys ). However, it currently lacks a VHDL frontend. I'm wondering if it's possible to access various levelsl of intermediate representation in GHDL?  LLVM IR would be too low-level and lacks any kind of structural information. Is there some higher-level IR within GHDL that can be accessed prior to the "lowered" LLVM IR?  What kind of format would it be in if it exists?

It would be great to have a full path from VHDL -> FPGA bitstreams and GHDL probably has the most complete, accurate open source VHDL parser currently available.

tgingold, 2018-03-23:
You should have a look at:  https://github.com/tgingold/ghdlsynth-beta

philtomson, 2018-03-23:
Oh, that looks interesting. Thanks, I'll have a look.

philtomson, 2018-03-23:
I see this requires patching the yosys source. Have you tried passing your changes upstream to yosys?

tgingold, 2018-03-23:
You don't need to patch yosys as you can use it as a yosys module.

philtomson, 2018-03-23:
So it looks like the only patch is applied to the yosys Makefile. Still, it might be nice to have that on the Yosys side?

Also, looks like most of the work is actually being done in ghdl/src/synth.

tgingold, 2018-03-24:
I think the Makefile patch would be submitted once the work is more advanced.

closed 2018-12-12 by tgingold