ghdl/ghdl #546: Crash when given multiple range constraints and initializer

ccorn, 2018-03-23:
The following short `--std=08` testcase (with possibly invalid syntax or semantics, I am just beginning with VHDL) leads to a `TYPE.INTERNAL_ERROR` pointing to `trans-chap3.adb:3094`.

~~~
$ ghdl -c --std=08 test-bug.vhdl -e test_arr

******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.36-dev (v0.35-116-g472552e5-dirty) [Dunoon edition]
Compiled with GNAT Version: 6.4.0
Target: x86_64-pc-cygwin
In directory: /home/Cygwin/Adlershof/CVT-Hardware/ColVisTec/USB development/CPLD/VHDL-examples/
Command line:
/usr/local/bin/ghdl1-llvm --std=08 -P/usr/local/lib/ghdl/ieee/v08/ -P/usr/local/lib/ghdl/ --anaelab test_arr --ghdl-source=test-bug.vhdl -c -o e~test_arr.o e~test_arr
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans-chap3.adb:3094
Call stack traceback locations:
0x1005abcbd 0x10051503d 0x100515508 0x100517667 0x100517728 0x10051a25b 0x100590ea3 0x100590fc8 0x100596f2b 0x10056d4a3 0x1005bfd89 0x100550496 0x1005bde06 0x10055c41a 0x1005c53f5 0x10044a47b 0x100401f5c 0x1800479b0 0x180045651 0x180045702
******************************************************************
ghdl:error: compilation error

Cygwin@Cornelssen-PC ~/Adlershof/CVT-Hardware/ColVisTec/USB development/CPLD/VHDL-examples
$ cat test-bug.vhdl
entity test_arr is
end entity;

architecture behavior of test_arr is
    type T is array(natural range <>) of bit_vector;
    signal a: T(0 to 1)(0 to 1) := ("00", "11");
begin
end architecture;
~~~
If the initialization `:= ("00", "11")` is omitted, compilation succeeds without issuing any diagnostics.

tgingold, 2018-03-24:
Arrays of unbounded arrays is a vhdl 2008 feature that is not yet fully supported by ghdl.

Paebbels, 2018-03-24:
This is related to issues:
* #453 - ghdl-llvm crash when compiling uvvm
* #459 - Constraining partial constrained types
* #473 - Unconstrained arrays; crash on elaboration

commit d22e8789 (2018-11-22): aggregates: WIP for arrays of unbounded elements.

Fix #546
closed 2018-11-22 by tgingold