ghdl/ghdl #547: array of unconstrained records

slaweksiluk, 2018-03-26:
Hello, I haven't found issue for this VHDL 2008 future:
```
[slawek@localhost vhdl-playground]$ ghdl -a --std=08 ghld-bug.vhd 

******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.36-dev (v0.35-126-ge4ce0ea4) [Dunoon edition]
Compiled with GNAT Version: 7.3.1 20180130 (Red Hat 7.3.1-2)
Target: x86_64-redhat-linux
In directory: /home/slawek/vhdl-playground/
Command line:
/home/slawek/install_prefix/libexec/gcc/x86_64-pc-linux-gnu/7.3.0/ghdl1 --std=08 -P/home/slawek/install_prefix/lib/ghdl/ieee/v08/ -P/home/slawek/install_prefix/lib/ghdl/ -quiet -o ghld-bug.s ghld-bug.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans.adb:619
Call stack traceback locations:
0x6bce44 0x6dcd83 0x6de59a 0x708579 0x6e425e 0x6c500b 0x6ba21a 0x6b8ebb 0x5fed98 0xab5e6d 0x5f93b8 0x6047f9 0x71f329 0x5f7ad5 0x7f8b1537cf28 0x5fb7d8 0xfffffffffffffffe
******************************************************************

Execution terminated by unhandled exception
raised TYPES.INTERNAL_ERROR : trans.adb:619
Call stack traceback locations:
0x6bce44 0x6dcd83 0x6de59a 0x708579 0x6e425e 0x6c500b 0x6ba21a 0x6b8ebb 0x5fed98 0xab5e6d 0x5f93b8 0x6047f9 0x71f329 0x5f7ad5 0x7f8b1537cf28 0x5fb7d8 0xfffffffffffffffe
ghdl:error: compilation error
[slawek@localhost vhdl-playground]$ cat ghld-bug.vhd 
library ieee;
use ieee.std_logic_1164.all;

package ghdl_bug is

type uncon_rec_t is record
	slv : std_logic_vector;
end record;
type uncon_rec_arr_t is array (0 to 1) of uncon_rec_t;

end ghdl_bug;
[slawek@localhost vhdl-playground]$ 
```

tgingold, 2018-03-28:
I can reproduce the crash.  I will investigate.

tgingold, 2018-11-12:
Now working.

closed 2018-11-12 by tgingold