ghdl/ghdl #549: GHDL PowerShell Installation fails on Windows 7 Pro 64

ParticleBert, 2018-04-06:
I downloaded the file ghdl-0.35-mcode-windows.installer.ps1 from the website and followed the instructions

```The ghdl-.......installer.ps1 file is a PowerShell script with embedded ZIP compressed and a base64 encoded BLOB (Binary Large OBject) variable. It will extract the BLOB and copy all files to the specified installation directory. Run .\ghdl-.......installer.ps1 -Install C:\Path\To\GHDL```

After running this command in the PowerShell I get an error:

```
PS C:\Users\cwiesner\Downloads> .\ghdl-0.35-mcode-windows.installer.ps1 -Install D:\Programme\GHDL

Sicherheitswarnung
Führen Sie ausschließlich vertrauenswürdige Skripts aus. Skripts aus dem Internet können zwar nützlich sein, stellen jedoch auch eine potenzielle Gefahr für Ihren Computer dar. Möchten Sie
"C:\Users\XXX\Downloads\ghdl-0.35-mcode-windows.installer.ps1" ausführen?
[N] Nicht ausführen  [M] Einmal ausführen  [H] Anhalten  [?] Hilfe (Standard ist "N"): M

Sie müssen auf der rechten Seite des Operators "-" einen Wertausdruck angeben.
Bei C:\Users\XXX\Downloads\ghdl-0.35-mcode-windows.installer.ps1:229 Zeichen:24
+             if ($InstallInPath - <<<< in "m","u","s")
    + CategoryInfo          : ParserError: (:) [], ParseException
    + FullyQualifiedErrorId : ExpectedValueExpression
```

The error translates into *You have to put in a value on the right side of the "-"-Expression*

Paebbels, 2018-04-07:
**For `Sicherheitswarnung`:**  
You need to run the following command in an administrator session of PowerShell:
```powershell
Set-ExecutionPolicy RemoteSigned
```

This allows execution of local (self written) scripts without needing them to signed by a local or company administrator. Only scripts from Internet / intranet need to be signed.

**For line 229:**  
I'll investigate this.

**Work around:**  
Please und the provided zip file for windows.

daesters, 2019-02-07:
Setting the Set-ExecutionPolicy works to execute the script. But then, there **is still an error** (Windows 7, x64)
The error reads like (is the same as the German version above)
```
You must provide a value expression on the right-hand side of the '-' operator.
At C:\Users\user\Downloads\ghdl-0.35-mcode-windows.installer.ps1:229 char:24
+             if ($InstallInPath - <<<< in "m","u","s")
    + CategoryInfo          : ParserError: (:) [], ParseException
    + FullyQualifiedErrorId : ExpectedValueExpression
```

Paebbels, 2019-02-08:
This will be fixed in PR #693.