ghdl/ghdl #550: How to install v0.35 on Ubuntu 16.04-based OS?

dominiksalvet, 2018-04-07:
I am sorry, but I just don't get it. It is possible that I am missing some knowledge about the way of the installation. The way I am trying to reach it is that I downloaded the *ghdl-0.35-llvm-3.8-ubuntu14.tgz* file from the [ghdl 0.35](https://github.com/ghdl/ghdl/releases/tag/v0.35) release, extracted it and recently I have been trying to install it from the extracted content somehow, but I have failed.

**Firstly**, I don't know how to install the ghdl-0.35 to the system the way that it would be possible to execute the `ghdl` command from everywhere. And **secondly**, even when I try to run it locally from the *bin* directory, I am getting the following error:

```
./ghdl: error while loading shared libraries: libgnat-4.6.so.1: cannot open shared object file: No such file or directory
```

I have tried to install the `libgnat-4.9` package, but the error persists. Unfortunately, I am not able to locate package `libgnat-4.6` exactly, when I am trying to install it. The debian package from the release [ghdl 0.33](https://github.com/ghdl/ghdl/releases/tag/v0.33) is preventing me from the installation because of the same reason.

Please, is it possible to provide me step-by-step tutorial how to install the latest stable release, v0.35, on Ubuntu 16.04-based OS (Linux Mint 18.3)? Unfortunately I haven't found anything like that.

Thank you a lot.

Paebbels, 2018-04-07:
Try to build GHDL on your PC manually. Install any LLVM version and follow the build steps from the documentation.

This is easier and faster then solving dependencies of a pre-build file :)

1138-4EB, 2018-04-08:
@dominiksalvet , just for clarification, `ghdl-0.35-llvm-3.8-ubuntu14.tgz` won't work on Ubuntu 16, because it was built with Ubuntu14, which has different library versions. As @Paebbels suggested, it is easier to compile it yourself than trying to revert the libraries in your system and/or installing multiple versions of them.

Furthermore, although it is not active in the CI, GHDL is ready to be tested on Ubuntu16. See https://github.com/ghdl/ghdl/blob/master/dist/linux/docker/build/ubuntu16 This file is meant to build a docker image (a kind of virtual machine) with minimum dependencies to build GHDL. Even if you don't use docker, you can use it to know the exact packages that need to be installed. For LLVM:

```
apt-get install -y gnat gcc make zlib1g-dev llvm-3.9-dev clang-3.9
```

Note: replace `3.9` with `3.8` if you want to keep the already installed packages.

You can follow a similar approach when/if you update to Ubuntu 18: https://github.com/ghdl/ghdl/tree/master/dist/linux/docker/build

dominiksalvet, 2018-04-08:
@1138-4EB I see, thank you for the info.

dominiksalvet, 2018-04-09:
@1138-4EB Please, can you or anyone help me? Now I am dealing with the problem connected with transition to LLVM back-end from mcode. I haven't uninstall ghdl with mcode yet (I don't know how, anyway), but I am trying to install the one with LLVM back-end. I removed `clank`, 'llvm' and 'gnat' that I had previously installed and I installed the ones versions that you suggested. Nevertheless, it doesn't matter which version I use, I always end up with the following error while trying to make with `make` command (after successful `./configure --prefix=/opt --with-llvm-config`):
```
if test -d ./.git \
   && desc=`cd .; git describe --dirty`; then \
  sed -e "s/[(].*[)]/($desc)/" < src/version.in > version.tmp; \
else \
  cp src/version.in version.tmp; \
fi
if [ ! -r version.ads ] || ! cmp version.tmp version.ads > /dev/null; then cp version.tmp version.ads; fi
make -f ./src/ortho/llvm4-nodebug/Makefile \
 ortho_srcdir=./src/ortho ortho_exec=ghdl1-llvm \
 GNAT_FLAGS="-aI./src -aI./src/vhdl -aI./src/grt -aI./src/psl -aI./src/vhdl/translate -aI./src/ghdldrv -aI./src/ortho -aI./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -gnatwe" LDFLAGS="" \
 LLVM_CONFIG="llvm-config-3.9" CXX="clang++" GNATMAKE="gnatmake" \
 all
make[1]: Entering directory '/home/dominik/Downloads/ghdl-0.35'
clang++ -c `llvm-config-3.9 --cxxflags` -o llvm-cbindings.o src/ortho/llvm4-nodebug/llvm-cbindings.cpp
/bin/sh: 1: clang++: not found
src/ortho/llvm4-nodebug/Makefile:18: recipe for target 'llvm-cbindings.o' failed
make[1]: *** [llvm-cbindings.o] Error 127
make[1]: Leaving directory '/home/dominik/Downloads/ghdl-0.35'
Makefile:299: recipe for target 'ghdl1-llvm' failed
make: *** [ghdl1-llvm] Error 2
```

Does anyone know where the problem might be, please?

Paebbels, 2018-04-09:
LLVM ist just a framework. The matching compiler e.g. for C is CLang. Please search your package managing system for a CLang version that matches your LLVM version. You might need to install CLang++ too. It's the C++ compiler for LLVM.

dominiksalvet, 2018-04-09:
@Paebbels Sorry, but I couldn't get out of it what to do. I also tried to install predefined package versions, the same behavior.

1138-4EB, 2018-04-09:
@Paebbels, all the required dependencies should be already installed if he executed `apt-get install -y gnat gcc make zlib1g-dev llvm-3.9-dev clang-3.9`. I think that the problem is related to `--with-llvm-config`: he is not providing any argument, and it should be `--with-llvm-config=llvm-config-3.9`, according to https://github.com/ghdl/ghdl/blob/master/dist/linux/build.sh#L88-L90

@dominiksalvet , try:

```
CXX="clang++-3.9" ./configure --prefix=/opt --with-llvm-config=llvm-config-3.9
make
```

1138-4EB, 2018-04-09:
@dominiksalvet , I edited my previous comment, in order to include how to download the latest version, instead of the 0.35 release, which I think you are/were trying to compile. I used "curl" to do so, but you could use wget or do it manually, as you wish.

Note that, due to the GHDL development flow, there is no meaningful advantage in using the latest release instead of the development branch. Then, it is suggested to always get the latest tarball from the URL I used above. This might change at some point, but I believe that it will be explicitly stated in the docs when done.

dominiksalvet, 2018-04-09:
@1138-4EB You got this, but the problem wasn't related to `--with-llvm-config`, it wouldn't let me go through `./configure ...`, so I edited it later to `--with-llvm-config=llvm-config-3.9`. However, the problem was solved after adding `CXX="clang++-3.9"`. What is it, why did it work? I am really curious. :)

1138-4EB, 2018-04-09:
@dominiksalvet , I am not an expert here, so you might want to take this with a grain of salt.

The default name for the clang compiler is `clang` or `clang++`. And the default name for llvm-config is `llvm-config`. On some distributions, the default/latest installed version takes those names, i.e., there are binaries or symbolic links in the path with those names. However, if you install multiple versions of a library, not all of them can have the exact same path and name, so they must be renamed. Sometimes one of them remains as default, i.e., it is symbolically linked to the 'vanilla' names. Sometimes none of them is allowed to take it.

In this case, you didn't install the default package (`apt-get install clang`), but a specific version (`clang-3.9`); and no symbolic links were created. Then, the system can't find any of the defaults (try `which clang++` or `which llvm-config`). There were three possible solutions:

- Tell you to install `apt-get clang llvm-dev`. Then, `./configure --prefix=/opt --with-llvm-config` would have worked, but you would be using version 3.8 of clang/llvm (which is the default in Ubuntu 16.04).
- Tell you to symbolically link `/usr/bin/clang++` to `/usr/bin/clang++-3.9` (and the same with `llvm-config`). This would be a "permanent" solution for any time `clang++` is required in your system.
- Use environment variables (`CXX`) and command-line arguments to provide the mapping only for the execution of the command(s) of interest now.

I chose the third one for two reasons:

1. It is the same approach that we use in the CI (so I knew it would work straightforward).
2. I saw a proactive attitude on your side, so I thought it wouldn't do any harm to let you know how to compile with different versions of the dependencies and not only with the default given by your distro.

On Ubuntu 16.04 you have these versions of clang available: 3.5, 3.6, 3.7, 3.8, 3.9, 4.0 and 5.0. You can compile GHDL with any of them, just following the same approach. See https://github.com/ghdl/ghdl/blob/master/dist/linux/build.sh#L76-L98 However, this is just an exercise because, up to my knowledge, there is absolutely no difference in the functionality of the produced GHDL.

Hope this helps satisfy your curiosity ;)

dominiksalvet, 2018-04-09:
@1138-4EB Nice, thank you a lot, I think it could help me to know this in the future. :)