ghdl/ghdl #551: GHDL not working in Windows Subsytem for Linux (WSL)

javValverde, 2018-04-09:
I managed to compile ghdl (master) in WSL using gcc4.9 and gcc5. When executing the adder test in cocotb, the following canbe seen:
```
make results.xml
make[1]: Entering directory 'cocotb/examples/adder/tests'
mkdir -p sim_build
cd sim_build && /usr/local/bin/ghdl -a  --work=work testbench/cocotb/examples/adder/tests/../hdl/adder.vhdl && /usr/local/bin/ghdl -e  --work=work adder

Execution terminated by unhandled exception
Exception name: STORAGE_ERROR
Message: stack overflow or erroneous memory access
Call stack traceback locations:
0x7f477bfa54ae 0x7fffe6ea9acb
testbench/cocotb/makefiles/simulators/Makefile.ghdl:60: recipe for target 'analyse' failed
make[1]: *** [analyse] Error 1
make[1]: Leaving directory 'testbench/cocotb/examples/adder/tests'
/testbench/cocotb/examples/adder/tests/../../../makefiles/Makefile.sim:71: recipe for target 'sim' failed
```

WSL would be an easy way of using GHDL in windows.

honzucha, 2018-04-09:
aren't there pre-build versions for GHDL?
I am sure they are. so why to mess with WSL.

Paebbels, 2018-04-09:
There are native (GNAT GPL) and MinGW compiled versions of GHDL for Windows.
No need for WSL.

closed 2018-12-13 by tgingold

bradleyharden, 2019-07-30:
Any chance we could get support for WSL? I haven't tried MinGW, but WSL is pretty seamless. It would be nice to leverage a new feature that Microsoft is actively supporting, rather than a third-party compatibility layer.

tgingold, 2019-07-30:
I could give a look, but WSL is not my first priority :-(

bradleyharden, 2019-07-30:
Fair enough. I thought I would ask. Normally, I work on a native Linux system, so it's not a high priority for me either.

1138-4EB, 2019-07-30:
I would say that GHDL is already supported on WSL. The error reported by @javValverde (which I can reproduce on Debian WSL with mcode backend) is likely to be produced by a not-complete implementation by Microsoft. Some further thoughts:

- There are multiple WSLs: Debian, Ubuntu, OpenSuse, Kali, Fedora, Alpine, etc. If we want to run the testsuite on a regular basis, we should pick one. It'd be a nightmare to support the WSL variant of each distro. I used Debian.
- The same binary that is built for a native Linux system should work in WSL. For example, I can `curl -fSsL https://github.com/ghdl/ghdl/releases/download/v0.36/ghdl-0.36-stretch-mcode.tgz | tar xfz - && cd bin && ./ghdl --version`. It works 'as well as' building GHDL in WSL.
- 'Native Windows binaries', those built with MinGW, should work. This is so because it is easier for Microsoft/WSL to fake a POSIX CLI for a windows binary, than actually faking the environment for a GNU/Linux binary. E.g. the 'golang' binary for Windows works ok from WSL too. However, I tried `./ghdl.exe **--version` and it produces a `./ghdl.exe: Invalid argument` error .
- [WSL2](https://www.youtube.com/watch?v=lwhMThePdIo) is suppossed to be a complete reimplementation, which might already solve the stack overflow issues. @bradleyharden, did you try WSL or WSL2?
- I don't think we can use WSL/WSL2 in AppVeyor, which is the CI service we use to build Windows artifacts. I wonder if Azure Pipelines (i.e. GitHub Actions) support running CI jobs in WSL.
- I don't see any practical advantage in using WSL/WSL2 instead of docker containers; specially because docker-for-win can be used from WSL, and better docker support is expected in WSL2. Unlike WSL, docker containers behave the same on native GNU/Linux systems or on Windows 10. Moreover, with docker it is straighforward to built/test GHDL for foreign architectures, such as armv7 or aarch64 (see https://github.com/dbhi/qus/).

FTR, I use Visual Studio Code and images such as `ghdl/ext:ls-vunit-gtkwave` every day, either on Windows 10 or on Fedora 30.

bradleyharden, 2019-07-30:
Ok, so I think I might be misunderstanding something about "Windows" binaries vs WSL. Maybe @1138-4EB can help.

What I want is to be able to develop using all the same tools and file system features that I would use in a native Linux system. I want to use a Bash shell, change file permissions, use a package manager to download a tool that I need, run that tool, etc. Can you do all that with MinGW? I've never really used Cygwin or anything like it. I can do all that in WSL, so that' what I've been using.

Can you run `ghdl.exe` inside WSL? I didn't think so, but I've never tried. Or are you running `ghdl.exe` in Powershell? Are you switching back and forth between a Bash shell and Powershell when developing vs. running simulations?

By the way, I'm not using WSL2, no. My Windows laptop is provided and maintained by my employer. It was already a somewhat lengthy process to get updated to 1809. I don't know if I can use something as bleeding edge as WSL2. I could look into it though.

I also haven't tried Docker at all. That's my own fault though. I just haven't put in the time to learn how to use it yet, so it's not my go-to solution. Are you using Docker with MinGW, inside WSL, or something else?

It sounds like there's a solution here that will solve my problems. I just don't have enough experience with the tools to see it.

1138-4EB, 2019-07-31:
@bradleyharden, this is me thinking fast and loud, so please take it with a grain of salt.

> What I want is to be able to develop using all the same tools and file system features that I would use in a native Linux system.

I suggest to use OCI containers (i.e. docker containers) to achieve this.

> I want to use a Bash shell, change file permissions, use a package manager to download a tool that I need, run that tool, etc. Can you do all that with MinGW?

I'm not sure about what you mean with MinGW. AFAIK, MinGW is a set of libraries that allows to cross-compile code for Windows from GNU/Linux. It can also be used on Windows in order to build some source code that follows UNIX conventions.

I have never used MinGW as a library, and I have never built GHDL on Windows. @Paebbels might help here.

What I do use regularly is [MSYS2](https://www.msys2.org/). More precisely, MSYS2 MINGW64 Terminal:

> At its core is an independent rewrite of MSYS, based on modern Cygwin (POSIX compatibility layer) and MinGW-w64 with the aim of better interoperability with native Windows software. **It provides a bash shell, Autotools, revision control systems and the like for building native Windows applications using MinGW-w64 toolchains**.
> 
> **It features a package management system** to provide easy installation of packages, Pacman. It brings many powerful features such as dependency resolution and simple complete system upgrades, as well as straight-forward package building.

The package manager, `pacman`, is quite known because it is the default in Arch Linux. You can use it to e.g. `pacman -Syu python gcc`.

Note that all the binaries that you install through this package manager, are Windows binaries (`exe` files). I.e., in `C:\msys64\usr\bin` you will find `ar.exe`, `cpp.exe`, `ld.exe`, `bash.exe`, etc. By the same token, you can execute any binary available on the system, just as you would from the cmd or powershell.

Summarizing, the MSYS shell basically converts paths (`\` to `/`, `C:\` to `/c`, etc.) so you can use GNU/Linux shell scripts to interact with otherwise native windows binaries.

As you see, this alone fullfils all of your requirements. I have been using it for more than five years now, in order to use the same bash scripts on GNU/Linux and on Windows (be it Windows 7 or Windows 10). I use it to interact with applications/packages installed through pacman (which are only available in the mingw64 shell); but I use it to inteact with tools which where installed independently too (e.g. docker-for-win, golang, Vivado, QuestaSim/Precision, etc.). As a matter of fact, I have several script-chains composed of Matlab/Octave -> Bash -> TCL. On Windows, a portable Octave calls MINGW64, which sources Vivado scripts to then execute TCL sequences.

Still, building/installing the tools on each platform is required. Here is where docker comes into the game. See below.

> I've never really used Cygwin or anything like it.

I honestly don't know which are the technical differences between MSYS2 and Cygwin. I can tell you some practical examples, tho:

- On Cygwin there is no CLI package manager I am aware of. The installer is a GUI package manager. So, each time you want to update/add packages, you just run the installer again and select whatever graphically.
  - I prefer having a package manager as in MSYS2.
- Some *important* pieces of software are available on Cygwin but not on MSYS2. For example, Cygwin provides an X server (see [x.cygwin.com](https://x.cygwin.com/)), but there is none on MSYS2.
  - This is more important than it seems, because the X server is precisely the only missing feature on Windows 10 in order to completely ignore the underlying desktop/OS and run any GNU/Linux DE directly.
  - During the last year, I've requested and helped testing support for Windows in [mviereck/x11docker](https://github.com/mviereck/x11docker). Some of the features have been recently split to [mviereck/runx](https://github.com/mviereck/runx). I'd rather not extend much here. Please, see https://github.com/mviereck/x11docker/issues?utf8=%E2%9C%93&q=author%3A1138-4EB+, and feel free to ask any further question that might arise.

> I can do all that in WSL, so that' what I've been using.

As you might imagine, WSL is an afterthought. Microsoft ignored Cygwin/MSYS2 and GNU/Linux OCI containers (aka docker) until they almost lost the wave. They <strike>caught</strike> bought it, and they are now riding it on their own closed source way.

I understand that you might not be allowed to install Cygwin/MSYS2. However, if you can, I see no advantage in using WSL. From an engineering/confidence pov, open source alternatives are more solid.

> Can you run ghdl.exe inside WSL? I didn't think so, but I've never tried.

As I commented above, no you can't. But, yes, you should be able to.

WSL is a single name/entrypoint for multiple features. One of the features is to provide the same behaviour as MSYS2. Hence, you can execute any Windows binary from WSL; you just need to keep in mind that you are using bash/shell paths and not cmd/powershell/windows format.

You can try this easily. Say you have VLC installed on Windows. In WSL, execute `/mnt/c/Program\ Files/VideoLAN/VLC/vlc.exe`.

> Ok, so I think I might be misunderstanding something about "Windows" binaries vs WSL. 

[... cut ...]
I feel this is unnecessarily hard to understand too. But that's probably because different companies are trying to compete for a market share, so technical issues are not as clear as they should be. Fortunately, I've spent some years now *fighting* to get a stable portable development environment across Windows 10, Fedora, Debian and CI services, and I'm quite happy with the result so far. I don't have time to document it as well as I'd like to, but I'll be glad to make you path easier to walk through. Feel free to ask.

bradleyharden, 2019-07-31:
@1138-4EB thank you for the detailed write-up! It is greatly appreciated.

> What I do use regularly is MSYS2. More precisely, MSYS2 MINGW64 Terminal

Right now, I'm using mintty with WSL. It looks like MSYS2 uses mintty as well, so that should be an easy switch.

> The package manager, pacman, is quite known because it is the default in Arch Linux.

I've heard of pacman, but I've never used Arch, so I have no direct experience with it. I've heard that the Arch wiki is excellent though, so that's a plus.

> Note that all the binaries that you install through this package manager, are Windows binaries

I think this is the part I was misunderstanding and still don't completely understand. I guess binaries compiled for MinGW are native Windows binaries that use a lightweight translation layer, whereas the MSYS2 environment uses native Linux binaries. Regardless of the details, I think I can get myself up and running with MSYS2/MinGW.

> rendered on the Windows desktop with Cygwin's X server, or with VcxSrv

I'm familiar with VcsSrv and already have it installed for use with WSL. That should be an easy transition as well.

> If you have never used docker at all, please, let me introduce you:

I've read a little about the concept and I've played around with it in Ubuntu briefly, but I haven't yet taken the time to integrate it into my existing workflow. It's been on my todo list for a while though, so I guess this could provide the impetus to actually follow through.

Luckily, installing software on my laptop is not usually an issue. Upgrading to new builds of Windows is a different story though. If MSYS2 + MinGW + Docker works well though, then I won't need to upgrade at all.

I've already got MSYS2 + MinGW installed, and I'm installing Docker-for-Windows now. Hopefully it all goes smoothly. Thanks again for your help.

1138-4EB, 2019-08-01:
> Right now, I'm using mintty with WSL. It looks like MSYS2 uses mintty as well, so that should be an easy switch.

Honestly, I find absolutely no difference. There might be some, but I have not found any.

> I've heard of pacman, but I've never used Arch, so I have no direct experience with it. I've heard that the Arch wiki is excellent though, so that's a plus.

That's correct. The Arch wiki is an excellent place to learn about modern GNU/Linux distros, even if you are using some other flavour (say Debian/Ubuntu or CentOS/Fedora). Should you have a spare weekend, I suggest to install Arch from scratch, while reading the wiki. This is for learning purposes, tho; it is unrelated to this topic.

Regarding pacman, it is just another package manager. Here you will find equivalent commands for pacman, dnf, apt, zypper and emerge: https://wiki.archlinux.org/index.php/Pacman/Rosetta

> I think this is the part I was misunderstanding and still don't completely understand. I guess binaries compiled for MinGW are native Windows binaries that use a lightweight translation layer, [...]

Unfortunately, I cannot help here. I'd say that both MinGW and MSYS2 use native Windows binaries with a translation layer. Nonetheless, as you said, I think that the details are not so important here.

> I'm familiar with VcxSrv and already have it installed for use with WSL. That should be an easy transition as well.

I used VcxSrv exclusively until a few months ago. For a regular unauthenticated use with Intel (embedded) GPUs, everything works ok. However, if you want to make the X server secure, or if you want to use 3D acceleration with NVIDIA GPUs, VcxSrv crashes. In this context, Cygwin/X works. See [mviereck/x11docker#148](https://github.com/mviereck/x11docker/issues/148#issuecomment-491607833) and mviereck/x11docker#160.