ghdl/ghdl #554: Version detection broken in travis-ci

1138-4EB, 2018-04-13:
Almost a month ago the version detection was broken in travis-ci. The last successful and correct build: [834](https://travis-ci.org/ghdl/ghdl/jobs/354693488) (see line 1518) and the first successful but incorrect one is [836](https://travis-ci.org/ghdl/ghdl/jobs/356988756) (see line 1521). I think that these are the changes to blame: https://github.com/ghdl/ghdl/commit/ffca448651c5b7b0fc5fe6bdbbcf1e0dc13419e1#diff-7e056ddc530791272d363e50c1f25647

Because the version is now set in configure, I think that we need to analyse [travis-ci.sh#L103-L104](https://github.com/ghdl/ghdl/blob/master/dist/linux/travis-ci.sh#L103-L104). The output we get now is:

```
/# git clone https://github.com/ghdl/ghdl
...

/# cd ghdl

/ghdl# cat version.tmp
cat: version.tmp: No such file or directory

/ghdl# cat src/version.in
package Version is
   Ghdl_Ver : constant String := "@VER@";
   Ghdl_Release : constant String :=
      "(tarball) [Dunoon edition]";
end Version;

/ghdl# make -f Makefile.in srcdir=. version.tmp
if ! test -d ./.git \
   || ! desc=`cd .; git describe --dirty`; then \
  desc="tarball"; \
        fi; \
sed -e "s/[(].*[)]/($desc)/" \
               -e "s/@VER@/@ghdl_version@/" < src/version.in > version.tmp; \

/ghdl# cat version.tmp
package Version is
   Ghdl_Ver : constant String := "@ghdl_version@";
   Ghdl_Release : constant String :=
      "(v0.35-150-g60f3bb31) [Dunoon edition]";
end Version;
```

Note that the sequence above is executed just once. Then, inside each container, `configure` is executed and last `make`. As a result, the second time `make` is executed, `Ghdl_Ver` in `src/version.in` contains `@ghdl_version@` instead of `@VER@`, and the replacement with sed does not work.

A possible solution is to remove lines [L98-L104](https://github.com/ghdl/ghdl/blob/master/dist/linux/travis-ci.sh#L103-L104) from `travis-ci.sh`. As a result, we will get `GHDL 0.36-dev (tarball) [Dunoon edition]`. However, I think this is undesired, because the nightly builds and/or docker images will produce binaries that are aparently not different from releases. I believe that `tarball` should only be used for releases, i.e., non `-dev` builds.

1138-4EB, 2018-04-13:
Another option is to replace L120 with `sed -e "s/@ghdl_version@/@VER@/" < version.tmp > src/version.in;`. But, for this to work, the makefile should be modified in order to remove `desc=tarball` and keep the content of `Ghdl_Release` in version.in instead.

tgingold, 2018-04-13:
Thanks.  I think it is fixed.

closed 2018-04-22 by tgingold