ghdl/ghdl #558: Type can not be used as range of an array

Asif71, 2018-04-19:
type      SPIrecGeneric     is      record
    		BIT				: positive;
    end record;	
    
    type     SPI_Rec_input    is   record
		SPI_Data_in 	       :  std_logic_vector (SPIrecGeneric.BIT-1 downto 0);
    end record;					
    
    type    SPI_Rec_Output	is    record
    		SPI_Data_out 		:  std_logic_vector (SPIrecGeneric.BIT-1 downto 0);	
    end record;	

sources_1\new.\Library.vhd:31:93: type "spirecgeneric" cannot be selected by name
sources_1\new.\Library.vhd:43:77: type "spirecgeneric" cannot be selected by name
eval_discrete_type_length: cannot handle IIR_KIND_SIMPLE_NAME (std_logic_1164.v08:72:36)

Paebbels, 2018-04-19:
Why GHDL version are you using?
You code example is not complete. Is this code embedded in a package?

Anyhow, while this is invalid code, it should not crash.

tgingold, 2018-04-21:
I cannot reproduce the crash (I only get the errors).
I will close this issue, feel free to reopen it with a reproducer.

closed 2018-04-21 by tgingold

Asif71, 2018-04-24:
I was using GHDL version v0.36 and this code was embedded in a package. 
I will provide full test case including entity and package in few days.

Asif71, 2018-04-24:
I was using GHDL version v0.36 and this code was embedded in a package.

Paebbels, 2018-04-24:
Please provide the output of `ghdl --version` to verify the exact version (build number / Git hash).