ghdl/ghdl #559: unexpected bound check failure

slaweksiluk, 2018-04-24:
Hello, I have found a bug in bounds checking. I also added commented out lines which remove the bound check error (in tb and dut). Run:
```bash
ghdl -a ghdl_bug.vhd
ghdl -e tb
./tb
ghdl_bug.vhd:24:16:@0ms:(report note): arg'length=2
ghdl_bug.vhd:25:16:@0ms:(report note): v(0)'length=2
./tb:error: bound check failure at ghdl_bug.vhd:28
./tb:error: error during elaboration
```
ghdl just cloned from master:
```
$ ghdl -v
GHDL 0.36-dev (v0.35-160-gcc58394a) [Dunoon edition]
 Compiled with GNAT Version: 7.3.1 20180303 (Red Hat 7.3.1-5)
 GCC back-end code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2015 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
Reproducer:
```vhdl
--
-- dut
--
library IEEE;
use IEEE.STD_LOGIC_1164.ALL;
use IEEE.NUMERIC_STD.ALL;

entity dut is
Generic (
	TARGETS_ADDR : std_logic_vector
);
end dut;
architecture a of dut is

	signal slv : std_logic_vector(1 downto 0);

	type slv_arr_t is array (0 to 1) of std_logic_vector(slv'range);

	function addr_arr_init (
		arg : std_logic_vector
	) return slv_arr_t is
		variable v : slv_arr_t;
	begin
		report "arg'length="&positive'image(arg'length);
		report "v(0)'length="&positive'image(v(0)'length);

		-- Bound check error
		v(0) := arg(1 downto 0);

		-- No bound check error
		--v(0) := arg;
		return v;
	end function;

	constant ADDR_ARR : slv_arr_t := addr_arr_init(TARGETS_ADDR);
begin

end a;

--
-- tb
--
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity tb is
end entity;

architecture bench of tb is
	constant C_SLV : std_logic_vector(1 downto 0) := "00";
begin

	dut : entity work.dut
	generic map	(
		-- Bound check error
		TARGETS_ADDR => "00"
		-- No bound check error
		--TARGETS_ADDR => C_SLV
	);

	stimulus : process
	begin
	report "pass";
	wait;
	end process;

end bench;
```

slaweksiluk, 2018-04-24:
Yet another way to fix bound check error is to constrain generic:
```vhdl
Generic (
	TARGETS_ADDR : std_logic_vector
);
```
Well, can I have unconstrained generic in vhdl at all?

Paebbels, 2018-04-24:
@slaweksiluk  
You can use unconstrained generic values. The size is inferred from the actual in the generic map.

slaweksiluk, 2018-04-24:
@Paebbels  does it mean arrays literals (I don't know valid formal name) like x"00" always have 'to' direction?

Paebbels, 2018-04-24:
The formal type of your generic association element is `std_logic_vector`.  
`std_logic_vector` is declared as a resolved subtype of `std_ulogic_vector`.  
This type in turn is defined as an array type with (unconstrained) index type `natural`.  
`natural` is a subtype of `integer`.  
`integer` is defined with a `to` range.

Thus, bitstring literals are created with implicit `to` range.

If you would create your own integer type with a `downto` range, you could create an array type based on such a "revered integer", that in turn implicitly creates literals with `downto` range :).

slaweksiluk, 2018-04-24:
@Paebbels thank you for that explanation, good to know

CraigNoble, 2018-04-24:
@Paebbels thanks for that bit of information, I knew literals had a to index but never understood why. Every day is a learning day. :)

tgingold, 2018-04-24:
@Paebbels using mismatching bounds is allowed by VHDL.  In that particular case it is difficult to complain during analysis.

slaweksiluk, 2018-04-24:
So this is valid VHDL, but currently it is hard to support it in this tricky test case?

Paebbels, 2018-04-24:
@tgingold  
But it's not a mismatching bound. I see a slice `arg(1 downto 0)`. On an array with `to` direction. A slice must have the same direction as the array is defined.

> **8.5 Slice names**  
> [...]  
> It is an error if the direction of the discrete range is not the same as that of the index range of the array denoted by the prefix of the slice name.  
> [...]

The chain of `"00"` propagation:
1. `TARGETS_ADDR => "00"`  
   creates a `std_logic_vector(0 to 1)`
2. `TARGETS_ADDR : std_logic_vector`  
   gets a `std_logic_vector(0 to 1)` passed
3. `constant ADDR_ARR : slv_arr_t := addr_arr_init(TARGETS_ADDR);`  
   passes a `std_logic_vector(0 to 1)` to `addr_arr_init`
4. `arg : std_logic_vector`  
   gets a `std_logic_vector(0 to 1)` passed
5. `v(0) := arg(1 downto 0);`  
   slices a `std_logic_vector(0 to 1)` with `downto` direction  
   **This must cause an error**

All intermediate declarations (generic constants, parameter constants) are unconstrained, thus the direction is inherited from the actual.

Have I missed something in my reasoning chain?

tgingold, 2018-04-25:
Yes, you're right.  This was allowed only in vhdl 1987.

commit b6d79c23 (2018-04-25): Use specific error message for direction mismatch.

Fix #559
closed 2018-04-25 by tgingold